esphome/feature-requests #940: Xiaomi Smart Scales

nickrout, 2020-10-07:
**Describe the problem you have/What new integration you would like**

Support for Xiaomi Smart Scales via the Xiaomi BLE component
**Please describe your use case for this integration and alternatives you've tried:**

There are a number of these Bathroom Scales sold by Xiaomi. There is a home assistant project https://github.com/lolouk44/xiaomi_mi_scale but I guess that needs the HA machine to be near the scales. 

There is also a bit of reverse engineering info here https://github.com/oliexdev/openScale/wiki/Xiaomi-Bluetooth-Mi-Scale

**Additional context**

I hoped that this might be easy to fit into the existing Xiaomi BLE component. Apparently there is no encryption.

pattyland, 2020-10-20:
Dublicate of #209

nickrout, 2020-11-24:
Ha  ha my scales took this long to come from China, in the meantime a local seller had them on sale for cheaper than aliexpress. 

Anyway, I'll close this and look at #209  and the PR referred to there. Sorry for the duplicate.

closed 2020-11-24 by nickrout