esphome/feature-requests #943: Set stepper position (without moving)

abelmatser, 2020-10-11:
**Describe the problem you have/What new integration you would like**

I would like to be able to set the stepper position to a value. This could make homing, troubleshooting and toggling the position of a cover much easier.

**Please describe your use case for this integration and alternatives you've tried:**

I tried to use stepper.report_position, but that moved my stepper to the location desired to be reported.

**Additional context**

Sometimes my curtain hangs behind something, which causes it to be unable to move. The stepper tries to obtain that location anyway, so it skips steps until it reaches that location. The problem is however, that it did not get to that location, so the next time your cover is opened/closed, it goes the entire wrong way. I want a method to set the position of the stepper after such the cover did not move.

The following snippet should toggle the location of the stepper after it hung, but currently does not work with stepper.report_position.
```
- service: toggle_current_position
  then:
    - stepper.report_position:
        id: ${plug_id}_stepper
        position: !lambda |-
          if (id(${plug_id}_cover).position == id(closed_position)) {
            return 0;
          } else {
            return id(closed_position);
          }
```

swifty99, 2021-03-22:
+1

I will use a motor for infinite rotation in one direrection. I think that is impossible right now due to limited target position.
It would also very helpful for endswitch based homing.

glmnet, 2021-03-22:
Did you try calling set_position after report_position? Thats how docs instructs to do... https://esphome.io/components/stepper/index.html#stepper-report-position-action

tungmeister, 2021-11-24:
I have the same issue with my Venetian blinds, sometimes due to the positioning, they will skip similar to this issue that was mentioned resulting in them being out of alignment. If I could override the open position to introduce an artificial offset this issue would be alleviated.