esphome/feature-requests #948: Request Air Quality (percentage) support for the BME680 sensor

TravellerNL, 2020-10-13:
**Describe the problem you have/What new integration you would like**
Measuring Air Quality is not directly available using ESPhome as it is when the BME680 is connected directly to Home Assistant via GPIO on a Raspberry Pi for instance.. Support for reading Air Quality on a Percentage scale would be highly appreciated as an extra datapoint next to Gas-Resistance, Temperature, Humidity and Pressure.

**Please describe your use case for this integration and alternatives you've tried:**
Home Assistant's website https://www.home-assistant.io/integrations/bme680/ states a clear setup how to integrate the BME680 and get access to the Monitored Conditions:  [“temperature”, “humidity”, “pressure”, "gas(resistance)" and “airquality”]. However, When using the sensor remotely e.g. connected to an ESP8266, ESPhome's implementation of the sensor lacks the airquality property. I'm aware of the fact that you could derive a Lambda from the Gas-resistance in lieu of Relative Humidity externally, however documentation about this method is scarce (and not for everybody) and the output values are not as clear or meaningful as the percentage scale that Home Assistant is using in its examples for the BME680 sensor. There is no reason why ESPhome should not support both methodologies via the ESPhome interface. People who wish to have access to the quick and easy Air Quality Percentage values and People who are more scientific and math inclined and programming savvy to calculate Lambda values could both have their way of successfully implementing the BME680 into their system. There certainly is a demand for this extra functionality!

**Additional context**

TravellerNL, 2020-10-16:
Stumbled on a discussion in the Tasmota realm about the Bosch BME680: https://github.com/arendst/Tasmota/issues/5024
Seemingly, getting the right data, the right amount of data, the right measurement settings and finally calculating the derived values into an Air Quality scale is not as straight forward as thought.

The Integration which is offered as an example for Home Assistant seems to have quite some shortcuts and assumptions.

Anyway, like many other home automation enthousiasts, I will put Air Quality on hold until someone finds a good way to implement this into code.

closed 2020-10-16 by TravellerNL

trvrnrth, 2020-11-26:
The Bosch Sensortec Environment Cluster (BSEC) software library exposes Indoor Air Quality (IAQ) measurements calculated from the BME680 readings. I've got a [PR to add a sensor](https://github.com/esphome/esphome/pull/1313) which integrates the library. If you're interested you can install it as a custom component to give it a whirl (see readme in the [standalone repo](https://github.com/trvrnrth/esphome-bsec-bme680)).

TravellerNL, 2020-11-28:
Hi Trevor,
Very interesting! I’ll certainly look into it again.
Thanks for pointing out,
Wim