esphome/feature-requests #949: Shelly dimmer 2 support

pixelspark, 2020-10-13:
**Describe the problem you have/What new integration you would like**

Add support for the Shelly Dimmer 2. This is an esp8266-based dimmer that is very easy to flash (like the Shelly 1 switches, which work brillantly with ESPHome firmware!). 

In the Dimmer 2, the ESP talks to an STM32 microcontroller on the board that performs the actual dimmers. Others have already reverse-engineered the communications protocol; see e.g. [here](https://github.com/Mollayo/Shelly-Dimmer-2/blob/master/light.cpp). More discussion (involving the Shelly Dimmer *1*) can be found over at the Tasmota project, [here](https://github.com/arendst/Tasmota/issues/6914).

**Please describe your use case for this integration and alternatives you've tried:**

I would like to be able to use the Dimmer 2 like I use the Shelly Switches in ESPHome (e.g. have it function as a light in Home-assistant, for instance).

At this point only the Shelly factory firmware is usable (which works fine but does not integrate with ESPHome, which I use for almost all other equipment). Tasmota support appears to be underway. 

**Additional context**

Thank you very much for making ESPHome! Device management is really convenient and an underestimated aspect of home automation.

craigueh, 2020-10-18:
I would like to see this support added too. I’m waiting on a couple of Dimmer 2 modules to arrive and don’t want to go the Tasmota route. 
Has anyone ever managed to control an stm32 chip with esphome on any other device?

pixelspark, 2020-10-19:
> Has anyone ever managed to control an stm32 chip with esphome on any other device?

The STM32 is 'just' a microcontroller, running a piece of software written by Shelly in this case. The ESP talks to it using a serial protocol. Talking to a peripheral using a serial protocol using the ESP UART functionality has been done before, for instance to read and configure the [MH-Z19 CO2 sensor](https://esphome.io/components/sensor/mhz19.html).

craigueh, 2020-10-21:
If the STM32 is just controlled by UART from the ESP chip then we may be able to do something like we did here for the Sonoff D1 dimmer - https://github.com/esphome/feature-requests/issues/611

JeffResc cleverly wrote a custom light component that controls the D1 dimming chip over UART and it works very well with esphome. Hopefully with some modifications to the code we could re-purpose for the Dimmer 2 module as the concept is pretty much the same.

pixelspark, 2020-11-25:
Looks like Tasmota managed to get it working - should be fairly easy to copy over to ESPHome then? See https://github.com/arendst/Tasmota/issues/6914

kostko, 2020-12-28:
I have work in progress support for the Shelly Dimmer in [my branch](https://github.com/kostko/esphome/tree/kostko/feature/shelly-dimmer). It uses [the open source STM32 firmware](https://github.com/jamesturton/shelly-dimmer-stm32) developed by @jamesturton (the same firmware is also used by Tasmota) which is automatically flashed during component setup. I have only tested it on a single Shelly Dimmer 2 that I ordered recently and it seems to work, I need to do some more testing and cleanup and then I'll probably open a PR against upstream.

Example relevant part of config:
```yaml
logger:
  baud_rate: 0

sensor:

light:
  # Dimmer.
  - platform: shelly_dimmer
    name: ${device_name} Light
    power:
      name: ${device_name} Power
    voltage:
      name: ${device_name} Voltage
    current:
      name: ${device_name} Current
```

pixelspark, 2020-12-28:
Awesome! I am waiting for a new dimmer to be delivered, will test it with this firmware soon.

Do you happen to know if this firmware also includes the 'ant-flicker debounce' feature (not the button debounce)? Without it my lights will flicker every now and then due to some kind of signaling over the power line...

craigueh, 2020-12-28:
Nice work. I look forward to trying it out.

kostko, 2020-12-28:
> Do you happen to know if this firmware also includes the 'ant-flicker debounce' feature (not the button debounce)? Without it my lights will flicker every now and then due to some kind of signaling over the power line...

That's the ripple control signal and yeah you're right, the firmware does not do any filtering at the moment, but [the author said he will work on it](https://github.com/jamesturton/shelly-dimmer-stm32/issues/11#issuecomment-751672160):
> It sounds like I will have to make some changes to how this firmware works to support the dimmer 2 with no neutral so I will add some filtering to cope with the ripple control signals while I'm at it.

curt7000, 2020-12-31:
Looking forward to this!  Willing to test if necessary.

jamesturton, 2020-12-31:
@kostko I have just added a [communication protocol document](https://github.com/jamesturton/shelly-dimmer-stm32/blob/master/COMMUNICATION.md) which might help you when implementing a driver in ESPHome.
If you have any questions about it then feel free to ping me and I will see how I can help out.

kostko, 2021-01-01:
@jamesturton Thanks, I already figured it out from reading the code and I have everything working. Nice work on the firmware btw! I did notice that some of the commands/fields are ignored. E.g. setting brightness via the settings command doesn't seem to do anything and most settings (except leading/trailing edge config) seem to be ignored. Is the idea that support for those will be added later or are the fields there just because the struct is based on the original firmware?

jamesturton, 2021-01-04:
I intend to implement settings as needed, but mainly they are there to maintain compatibility with the original firmware - however I do believe that Shelly have actually changed the protocol that they use now, so I'm not too sure how compatible they still are...

chriscrowe, 2021-03-23:
Any updates on this support? My Shelly Dimmer 2's showed up today and I wrongfully assumed ESPHome already supported them :(

poldim, 2021-04-28:
@kostko  I see the last update was at the end of december.  How is your shelly working?  Any plans of updating & integrating into core?

@jamesturton Any updates from your end?

kostko, 2021-04-28:
The branch I posted has a usable version, although I'm seeing some flickering and haven't had the time yet to debug more. Feel free to test it out.

jamesturton, 2021-04-28:
> 
> 
> @kostko I see the last update was at the end of december. How is your shelly working? Any plans of updating & integrating into core?
> 
> @jamesturton Any updates from your end?

I made a new testing version of the STM32 firmware to try and fix some of the flickering issues which I posted on the thread here: https://github.com/jamesturton/shelly-dimmer-stm32/issues/6. Maybe @kostko could try this out and see if it helps?

rnauber, 2021-04-28:
Hi,
I tested @kostko 's branch with @jamesturton 's firmware (at https://github.com/jamesturton/shelly-dimmer-stm32/commit/36977bdf4cb569a9acb9f676f467af6b70bef6da ). And it worked flawlessly even without a neutral wire! So I think it is ready for a PR (after updating the firmware).

Thank you for your grandiose work @kostko and @jamesturton !
[image]

curt7000, 2021-04-28:
> Hi,
>
> I tested @kostko 's branch with @jamesturton 's firmware (at https://github.com/jamesturton/shelly-dimmer-stm32/commit/36977bdf4cb569a9acb9f676f467af6b70bef6da ). And it worked flawlessly even [...]

This is really awesome!  Just curious, I see  what I believe is a transient voltage spike right at the end, a burst of light, when turning off the bulb and dimming down.  Are you seeing that occur with a higher load?  Can't remember if the Shelly requires a minimum load with their dimmers.  I know this is the case with others.

rnauber, 2021-04-29:
> This is really awesome! Just curious, I see what I believe is a transient voltage spike right at the end, a burst of light, when turning off the bulb and dimming down. Are you seeing that occur with a higher load? 

Well, the load was quite high actually. This is a > 60 W infrared light bulb. But still, the flash is quite noticeable.

craigueh, 2021-04-29:
Sorry if this is a stupid question but how do I use this custom branch in esphome?
I’d like to test it out on my dimmer2.

rnauber, 2021-04-29:
> Sorry if this is a stupid question but how do I use this custom branch in esphome?
> I’d like to test it out on my dimmer2.

That is no stupid question!

You should first checkout the branch to some directory (e.g. `esphome`) and then install it with:
```
git clone ...
cd esphome
sudo pip3 install -e .

```
Then you can call esphome with:
`python3 -m esphome`

craigueh, 2021-04-30:
Can this be done on HassOS?
I recently moved from Ubuntu with HASS on Docker to HassOS and find it very restricted at the command line.

rnauber, 2021-04-30:
I don't know, but I doubt it...

craigueh, 2021-05-04:
Any chance you managed to get this submitted to the latest esphome release today May 4th?

Bascht74, 2021-05-05:
I am interested in using ESPHome with Shelly dimmers as well. Looking forward to it.

AHH4485, 2021-05-12:
I am also interested, id would be nice if Shelly Dimmer support is available in the coming version of ESPHome

craigueh, 2021-06-22:
Do we have any update on when Dimmer 2 support will be added to ESPhome? I was hoping to see it in the official updates by now but not sure if there is still testing or somethign else holding it up. Anything we can do to help get this pushed through?

craigueh, 2021-06-22:
@kostko When I use your branch (https://github.com/kostko/esphome.git) and try to compile I get the following error: 

Failed config
light.shelly_dimmer: [source shelly_dimmer.yaml:32]
 Platform not found: 'light.shelly_dimmer'.

Any ideas or am I using the wrong branch?

jeroenst, 2021-07-10:
> I have work in progress support for the Shelly Dimmer in [my branch](https://github.com/kostko/esphome/tree/kostko/feature/shelly-dimmer). It uses [the open source STM32 [...]

Is it possible to create a pull request for esphome with this code?

rnauber, 2021-07-10:
> Is it possible to create a pull request for esphome with this code?

I think the code is ready, what is missing is:
- [ ] a proper documentation
- [ ] a test case

see [PULL_REQUEST_TEMPLATE](https://github.com/esphome/esphome/blob/7c678659d43a3f9cd8fcb34966e551113fb94a4c/.github/PULL_REQUEST_TEMPLATE.md)

BTW, if you want to quickly test kostko's branch without manually checking it out, just add this to your yaml:

```
external_components:
  - source:
      type: git
      url: https://github.com/kostko/esphome.git
      ref: kostko/feature/shelly-dimmer
    components: [ shelly_dimmer ]
```

So, who wants to go forward with it?

Bascht74, 2021-07-11:
Hi, does somebody have an complete yaml for a shelly dimmer 1 with support of the two buttons?

Here is what I patched together so far...

```
external_components:
  - source:
      type: git
      url: https://github.com/kostko/esphome.git
      ref: kostko/feature/shelly-dimmer
    components: [ shelly_dimmer ]

esphome:
  name: lampe-eg-wohnzimmer-wandlampen
  platform: ESP8266
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "498f5b7c048xxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: "IOT-LOTZ"
  password: "xxxxxxxxxxxxxxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lampe-Eg-Wohnzimmer-Wandlampen"
    password: "xxxxxxxxxxxxxxxxxxxx"

captive_portal:

logger:
  baud_rate: 0

sensor:

light:
  # Dimmer.
  - platform: shelly_dimmer
    name: ${device_name}
    power:
      name: ${device_name}_Power
    voltage:
      name: ${device_name}_Voltage
    current:
      name: ${device_name}_Current
```

But I don't know what to put into the yaml for local button support, leading/trailing edge configuration and probably some calibration (eg. map 0% to 10% power and 99% to 94% power and 100% to 100%)

Seb

craigueh, 2021-07-25:
@ Seb Try increasing your delayed_off: to 100ms. Once I increased mine the buttons work perfectly.

craigueh, 2021-07-25:
Has anyone experienced an issue where the lights state is reported incorrectly? 
With my dimmer sometimes when the lights are turned off they still report as on in HASS but dimmed really low. This is a nuisance because when turning the light back on the switch has to be flicked twice, once to turn off fully then again to turn back on.

Bascht74, 2021-07-26:
> @ Seb Try increasing your delayed_off: to 100ms. Once I increased mine the buttons work perfectly.

Thx. I will try...

malfroid, 2021-08-14:
> I think the code is ready, what is missing is:
>
>     * [ ]  a proper documentation
>
>     * [ ]  a test case
>
>
> see [...]

@kostko would you be interested to proceed with a pull request?

craigueh, 2021-09-09:
@kostko When I try to compile on the latest version of ESPhome I get the following error:

In file included from src/esphome/components/shelly_dimmer/shelly_dimmer.cpp:1:0:
src/esphome/components/shelly_dimmer/shelly_dimmer.h: In member function 'virtual esphome::light::LightTraits esphome::shelly::ShellyDimmer::get_traits()':
src/esphome/components/shelly_dimmer/shelly_dimmer.h:20:12: error: 'class esphome::light::LightTraits' has no member named 'set_supports_brightness'
     traits.set_supports_brightness(true);

Is this something you can fix?

Bascht74, 2021-09-13:
> Is this something you can fix?

I found another solution and used the code of another pull request. 
I updated the code above to use that code: https://github.com/esphome/feature-requests/issues/949#issuecomment-877807666

I changed to:
`external_components:
  - source:
      type: git
      url: https://github.com/jbergler/esphome.src
      ref: shelly-dimmer
    components: [ shelly_dimmer ]
`

truglodite, 2021-10-08:
I still have a few Shelly dimmers that I am looking forward to adding to esphome at some point… besides using mqtt. Any idea how close this is to becoming a pr/merged?

SweAdelpha, 2021-11-14:
> > Is this something you can fix?
> 
> I found another solution and used the code of another pull request. I updated the code above to use that code: [#949 (comment)](https://github.com/esphome/feature-requests/issues/949#issuecomment-877807666)
> 
> I changed to:
> 
> ```
> external_components:
>   - source:
>       type: git
>       url: https://github.com/jbergler/esphome.src
>       ref: shelly-dimmer
>     components: [ shelly_dimmer ]
> ```

This one doesn't seem to work either anymore. Worked great a couple of weeks ago when I flashed another one of my Shelly dimmers.
Is anyone aware of another fork that might work with the Shelly dimmer? I really don't want to use tasmota :)

rnauber, 2021-12-25:
Hi everyone,
As I recently acquired some more shelly dimmers fore some more home automation. Therefore, I brushed up the code of @kostko  and @jamesturton to compile with the current version of esphome. Furthermore, I added the ability to select the [firmware](https://github.com/jamesturton/shelly-dimmer-stm32) of the stm32. 
You can test that with the snippet below. If I find the time, I intend to add a testcase and the documentation to create a proper pull request for shelly dimmer 2 support...

Best,
Richard

```
external_components:
  - source:
      type: git
      url: https://github.com/rnauber/esphome.git
      ref: feature/shelly-dimmer
    refresh: 1min
    components: [ shelly_dimmer ]

light:
  - platform: shelly_dimmer
    name: ${devicename} Light
    id: thislight
    power:
      name: ${devicename} Power
    voltage:
      name: ${devicename} Voltage
    current:
      name: ${devicename} Current
    max_brightness: 500
    firmware: "51.6"
```

ActiveByte, 2021-12-26:
Thanks for your work @rnauber !

DomiStyle, 2021-12-29:
@rnauber Looks good, do you have a complete config for the Shelly Dimmer 2?

My Shelly Dimmer 2 just arrived and I would like to give it a go.

rnauber, 2021-12-29:
> @rnauber Looks good, do you have a complete config for the Shelly Dimmer 2?
> 
> My Shelly Dimmer 2 just arrived and I would like to give it a go.

No, I do not have a full working config yet, especially the temperature measurement is miscalibrated...
However, my current config is the following:

```
substitutions:
  devicename: shellydimmer2
  max_temp: "90.0"

esphome:
  name: $devicename
  platform: ESP8266
  board: esp8285
  
external_components:
  - source:
      type: git
      url: https://github.com/rnauber/esphome.git
      ref: feature/shelly-dimmer
    refresh: 1min
    components: [ shelly_dimmer ]

#<<: !include wifi.yaml

ota:
web_server:
api:

logger:
  baud_rate: 0

light:
  # Dimmer.
  - platform: shelly_dimmer
    name: ${devicename} Light
    id: thislight
    power:
      name: ${devicename} Power
    voltage:
      name: ${devicename} Voltage
    current:
      name: ${devicename} Current
    max_brightness: 500
    firmware: "51.5"
      
     
  - platform: monochromatic
    name: "${devicename} Status LED"
    output: led
    id: led_light
    #internal: True
    
  - platform: binary
    name: "${devicename} RESET SHD"
    output: SHDReset
    
switch:
  - platform: restart
    name: "$devicename Restart"
    
      
sensor:
  - platform: wifi_signal
    name: "$devicename WiFi Signal"
    update_interval: 60s
#    expire_after: 0

  - platform: ntc
    sensor: temp_resistance_reading
    name: ${devicename} Temperature
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    icon: "mdi:thermometer"
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 25°C

  - platform: resistance
    id: temp_resistance_reading
    sensor: temp_analog_reading
    configuration: DOWNSTREAM
    resistor: 32kOhm
  - platform: adc
    id: temp_analog_reading
    pin: A0

  - platform: adc
    pin: VCC
    name: "$devicename VCC"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO12
    name: "$devicename Switch 2"
    filters:
      - delayed_on_off: 100ms
 
  - platform: gpio
    pin:
      number: GPIO14
    name: "$devicename Switch 1"
    filters:
      - delayed_on_off: 100ms     
      
 
output:
  # Register the red LED as a dimmable output ....
  - platform: esp8266_pwm
    id: led
    pin:
      number: GPIO15
      inverted: True
      
  - platform: gpio
    pin: GPIO05
    id: SHDReset

```

DomiStyle, 2021-12-29:
@rnauber Thanks, that should work for now.

DomiStyle, 2022-01-08:
So I got around to installing my dimmer with the @rnauber branch now.

I connected 4 dimmable LED bulbs with a Fibaro Bypass 2 in parallel to the bulbs since they draw almost no power.

Power usage seems to be all over the place, fluctuating between 5 - 60W. Real power draw at max brightness should be at about 16W.

I wasn't quite able to get full brightness out of my bulbs, but that could also be a fault with the bulbs and not with the dimmer. The range I was able to get is 160 - 400. Anything above starts flickering, anything below doesn't light the bulb at all.

rnauber, 2022-01-08:
Dear @DomiStyle ,
my snippet includes:

> max_brightness: 500

which limits the output to 50%. Maybe try removing that line...

Best,
Richard

DomiStyle, 2022-01-08:
I started with `min_brightness: 0` and `max_brightness: 1000` to test the range and 160-400 is what I ended up with. Could be the bulbs though, they are pretty cheap.

Interestingly the bulbs also flicker at 400 brightness when there are 230V on the S1 terminal so I might have to lower it even more.

rnauber, 2022-01-08:
> I started with `min_brightness: 0` and `max_brightness: 1000` to test the range and 160-400 is what I ended up with. Could be the bulbs though, they are pretty cheap.
>
> Interestingly the bulbs also [...]

Hm, I guess the interaction of dimmers and LEDs is hard to get right...
As all the interfacing with mains is handled by the STM co-processor, have a look at the firmware and its issues there: https://github.com/jamesturton/shelly-dimmer-stm32/issues/6 
There are some settings you can play with... 

BTW, I am planning to make "WarmupTime" parameter accessible via esphome too, but it seems to have diverged from the original meaning of that parameter. So a am not sure how to name it...

fyksen, 2022-02-21:
Thanks for all the work. I have connected shelly 2 dimmer to one bulb, and I have trouble getting it to turn completely off. Using the config @rnauber posted earlier.

When I turn off the light entity in home assistant, it still glows. I have tried the 'RESET SHD' mode.
I haven not done a lot with esphome yet, so please excuse my ignorance.

grahampowell, 2022-03-06:
for me i had to change the LED pin for it to work on my shelly dimmer 2 (16 worked for mine)

I used this page as reference
https://templates.blakadder.com/shelly_Dimmer_2.html

output:
  # Register the red LED as a dimmable output ....
  - platform: esp8266_pwm
    id: led
    pin:
      number: GPIO16
      inverted: True

frank8the9tank, 2022-03-16:
Whats the status on this now?
I'm looking to flash the shelly dimmer 2 with esphome so i can additional functionality to it. The main thing that i want to add to it is a rotary knob trough the exposeded GPIO pins, Whould something like this be possible if flashed with esp home? (i have other devices with esp that i have added a rotatry know to, The thing i'm curious about is the exposed GPIO pins if they whould support the functionality that i what. So is can set the Dim level of the lamp with the knob and trough HA. The 2 input pins (the connector one) can still be used for a normal "hotel" switching arangement.

craigueh, 2022-03-16:
I've done this and it works fine with a rotary encoder. Her's my config:

time:
  - platform: homeassistant
    id: ha_time

sun:
  latitude: 53.163972
  longitude: -7.365056

# Basic config
substitutions:
  devicename: living_room_spots
  upper_devicename: Living_Room_Spots
  pin_led1: GPIO15
  pin_switch2n: GPIO14
  pin_switch1n: GPIO12
  pin_temp: A0
  pin_button1: NONE # GPIO for reset button not found out so far
  max_temp: "90.0"

uart:
 tx_pin: 1
 rx_pin: 3
 baud_rate: 115200

external_components:
  - source:
      type: git
      url: https://github.com/rnauber/esphome.git
      ref: feature/shelly-dimmer
    refresh: 1min
    components: [ shelly_dimmer ]

light:
  - platform: shelly_dimmer
    id: living_room_spots
    name: Living Room Spots
    leading_edge: false
#    warmup_brightness: 100
#    warmup_time: 20
    min_brightness: 0
#    max_brightness: 1000
#    gamma_correct: 1
    default_transition_length: 0.1s
    restore_mode: RESTORE_DEFAULT_OFF

sensor:
### Rotary encoder 1
  - platform: rotary_encoder
    name: "Living Room Dimmer Rotary Encoder"
    id: livingroomrotary1
    pin_a: GPIO14
    pin_b: GPIO12
    filters:
      - debounce: 0.1s
      - delta: 1
#      - lambda: |-
#          if (x < 1.0) { x = 1.0; }
#          if (x > 10) { x = 10; }
#          return x;
    resolution: 1
    min_value: 0
    max_value: 10
    on_value:
      then:
        light.turn_on:
          id: living_room_spots
          brightness: !lambda |-
            return id(livingroomrotary1).state * 0.1;

binary_sensor:
### Switch 1
  - platform: gpio
    pin: GPIO0
    id: switch1
    name: "Living Room Dimmer Switch"  
#    internal: true
    filters:
      - invert: 
      - delayed_on_off: 100ms
    on_multi_click:
# Double Click
    - timing:
        - ON for at most 0.3s
        - OFF for at most 0.25s
        - ON for at most 0.3s
        - OFF for at least 0.3s
      then:
        - logger.log: "Double Clicked"
        - homeassistant.service:
            service: light.toggle
            data_template:
              entity_id: light.living_room_lamp
#              brightness_pct: "100"
# Short Click
    - timing:
        - ON for at most 0.3s
        - OFF for at least 0.2s
      then:
       - logger.log: "Short Clicked"
       -   if:
              condition:
                - light.is_off: living_room_spots
              then:
                if:
                  condition: 
                    - lambda: 'return id(ha_time).now().hour < 21;'
                    - lambda: 'return id(ha_time).now().hour > 6;'
                  then: 
                    - logger.log: Turning lights on full
                    - sensor.rotary_encoder.set_value:
                        id: livingroomrotary1
                        value: 10
                    - light.turn_on:
                        id: living_room_spots
                        brightness: 100%
                  else: 
                    - logger.log: Turning lights on dimmed
                    - light.turn_on:
                        id: living_room_spots
                        brightness: 35%
              else:
                - light.turn_off:
                    id: living_room_spots

frank8the9tank, 2022-03-16:
Okay that sound really good! you really made my day!!!
I've never flashed a shelly with esphome, Can you point me in the direct on how to do this?
and for the rotary encoder do you have a photo and/or schematic of what you have made/done.
I'm look to remove all my 40 dimmer in my house and replace then all with shelly/esphome dimmers.
So i'm planning 5 for now, then test them for stablility and then the rest. 
I'm also reading about flickerin in this topic, do you experiance this with your setup?

jaxzin, 2022-03-16:
> I've never flashed a shelly with esphome, Can you point me in the direct on how to do this?

I used these directions to first flash Tasmota to my Shelly Dimmer 2 over the air. Why OTA? Because I had encased mine in epoxy to make it watertight for outdoor use. 
https://github.com/yaourdt/mgos-to-tasmota

I bet there is a direct path using the /ota REST endpoint and an esphome firmware but that escaped me. My two Shelly Dimmer 2 are now running ESPHome beautifully.

frank8the9tank, 2022-03-17:
Do you also have the push function working for the rotary encoder? working trought the header pinout on the back?

designerferro, 2022-03-19:
So... this was woking for me for several months, but,  since v2022.3.0 stopped working, don't know why... 
`external_components:
  - source:
      type: git
      url: https://github.com/rnauber/esphome.git
      ref: feature/shelly-dimmer
    refresh: 1min
    components: [ shelly_dimmer ]
# disable serial, we need this to talk to the co-processor
logger:
  baud_rate: 0

# define a light 
light:
  - platform: shelly_dimmer
    id: ${hostname}_d1
    name: "${hostname} Dimmer 1"
    leading_edge: false
    min_brightness: 0
    max_brightness: 1000
    gamma_correct: 1
    default_transition_length: 1s
    restore_mode: RESTORE_DEFAULT_OFF
    power:
      name: "${hostname} Power"
    voltage:
      name: "${hostname} Voltage"
    current:
      name: "${hostname} Current"`

rnauber, 2022-03-20:
> So... this was woking for me for several months, but, since v2022.3.0 stopped working, don't know why, complaining in the esphome dashboard editor "Platform not found: 'light.shelly_dimmer'."

@designerferro thanks for reporting that, I fixed it...

designerferro, 2022-03-20:
> > So... this was woking for me for several months, but, since v2022.3.0 stopped working, don't know why, complaining in the esphome dashboard editor "Platform not found: 'light.shelly_dimmer'."
> 
> @designerferro thanks for reporting that, I fixed it...
Thanks.
I tried copying esphome/component/shelly-dimmer to my nodes directory, but kept complainng.

rnauber, 2022-03-20:
>  tried copying esphome/component/shelly-dimmer to my nodes directory, but kept complainng.

Hm, you can try adding this instead:
```
external_components:
  - source:
      type: git
      url: https://github.com/rnauber/esphome.git
      ref: feature/shelly-dimmer
    refresh: 1min
    components: [ shelly_dimmer ]
```

rbswift, 2022-03-31:
I'm trying to switch over from Tasmota. Using original Shelly Dimmer v1 on 230V 50Hz and seeing a lot of flickering when dimming. This is my log and config. I'd really appreciate any suggestions for what I'm missing. Thanks!

```
[10:49:56][D][shelly:236]: Sending command: 0x10 (0 bytes) payload 0x
[10:49:56][D][shelly:248]: Command sent, waiting for reply
[10:49:56][D][shelly:358]: Got frame: 0x10
[10:49:56][I][shelly:402]: Got dimmer data:
[10:49:56][I][shelly:403]:   HW version: 0
[10:49:56][I][shelly:404]:   Brightness: 0
[10:49:56][I][shelly:405]:   Fade rate:  0
[10:49:56][I][shelly:406]:   Power:      0.000000 W
[10:49:56][I][shelly:407]:   Voltage:    0.822710 V
[10:49:56][I][shelly:408]:   Current:    0.000000 A
[10:49:56][D][sensor:124]: 'dining Power': Sending state 0.00000 W with 1 decimals of accuracy
[10:49:56][D][sensor:124]: 'dining Voltage': Sending state 0.82271 V with 1 decimals of accuracy
[10:49:56][D][sensor:124]: 'dining Current': Sending state 0.00000 A with 2 decimals of accuracy
[10:50:05][D][sensor:124]: 'dining VCC': Sending state 1.77246 V with 2 decimals of accuracy
```

```
substitutions:
  devicename: dining
  
esphome:
  name: $devicename
  platform: ESP8266
  board: esp8285

external_components:
  - source:
      type: git
      url: https://github.com/rnauber/esphome.git
      ref: feature/shelly-dimmer
    refresh: 1min
    components: [ shelly_dimmer ]

# Disable serial logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:
  password: "x"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

uart:
 tx_pin: 1
 rx_pin: 3
 baud_rate: 115200
 
sensor:
  - platform: wifi_signal
    name: "$devicename WiFi Signal"
    update_interval: 60s

  - platform: ntc
    sensor: temp_resistance_reading
    name: ${devicename} Temperature
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    icon: "mdi:thermometer"
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 25°C

  - platform: resistance
    id: temp_resistance_reading
    sensor: temp_analog_reading
    configuration: DOWNSTREAM
    resistor: 32kOhm
    
  - platform: adc
    id: temp_analog_reading
    pin: A0
    
  - platform: adc
    pin: VCC
    name: "$devicename VCC"
 
light:
  - platform: shelly_dimmer
    name: ${devicename} Light
    id: thislight
    power:
      name: ${devicename} Power
    voltage:
      name: ${devicename} Voltage
    current:
      name: ${devicename} Current
    max_brightness: 500

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO12
    name: "$devicename Switch 1"
    filters:
      - delayed_on_off: 100ms
 
  - platform: gpio
    pin:
      number: GPIO14
    name: "$devicename Switch 2"
    filters:
      - delayed_on_off: 100ms
```

[7 more replies]

closed 2022-04-14 by jesserockz