esphome/feature-requests #952: Modbus RTU sensors

Nicomagno, 2020-10-16:
**Describe the problem you have/What new integration you would like**

I would like Modbus RTU: sensor, binary sensor, switch and register write functions. Similar to home assistants implementation, https://www.home-assistant.io/integrations/sensor.modbus/ (sensor example)

Uart Modbus RTU support is apparently ready for some specific components. 
But I think a higher-level implementation than writing a custom uart sensor but lower than ready-made components with fixed registers, would be really useful and flexible for a lot of users.

**Please describe your use case for this integration and alternatives you've tried:**

the plan is to read data from distant old industrial equipment (mainly Schneider electric soft starters variable frequency drives and energy meters) that work without a PLC but I would benefit from integrating their data into home assistant.

**Additional context**

this could either be using an rs485 adapter but I think esp32 boards are capable of doing this by themselves

nikito7, 2020-11-14:
Obis example

https://www.hjemmeautomasjon.no/forums/topic/5141-han-til-home-assistant-over-esphome/

nikito7, 2021-05-13:
https://github.com/martgras/esphome/tree/modbus_component

nagyrobi, 2022-06-20:
implemented

closed 2022-06-20 by nagyrobi