esphome/feature-requests #953: Use ADC as button input

gitolicious, 2020-10-16:
**Describe the problem you have/What new integration you would like**

[Gosund SP112  v.34](https://templates.blakadder.com/gosund_SP112_v3_4.html) has the physical button attached to ESP8266's ADC pin. This pin can not be configured for buttons on ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**

The button of Gosund SP112  v.34 devices (and others which are using the ADC pin for digital inputs) should be supported.

Tasmota has a [special flag](https://tasmota.github.io/docs/Templates/#flag) for configuring the ADC type where `Button` can be selected to fulfill the requested feature. In ESPHome, I couldn't find something reasonable.

Closest I got to replicating the behavior is the following, but the ADC sensor is flooding the log and responsiveness is limited by `update_interval`:
```
binary_sensor:
  - platform: template
    id: button1
    name: "Gosund Button"
    lambda: return (id(button_adc).state < .5);
    on_press:
      then:
        - switch.toggle: switch1

sensor:
  - platform: adc
    id: button_adc
    pin: A0
    update_interval: .2s
```

**Additional context**

glmnet, 2020-10-17:
Does digitalRead works?

gitolicious, 2020-10-18:
Where would I use that? Could you give a code example?

glmnet, 2020-10-18:
Instead of 
`lambda: return (id(button_adc).state < .5);`

Or in any case analogRead should work

gitolicious, 2020-10-20:
Well, that would only slightly simplify my code but wouldn't solve the original issue of a fixed `update_interval` instead of a check in every loop, right?

glmnet, 2020-10-20:
if it works it will be faster, the `binary_sensor` lambda should run at 60hz I believe (ESPHome task update rate) your bottleneck now is the `sensor: ` you should remove that.

But most importantly is the arduino's digitalRead on the ADC pin posible?, (or in any case you can try analogRead and do the threshold comparison there).

gitolicious, 2020-10-21:
Ok, got your point now. What you suppose is
```yaml
binary_sensor:
- platform: template
  lambda: return digitalRead(A0);
```
This one did not give me any triggers when pushing the button. Then I tried your second suggestion.
```yaml
binary_sensor:
- platform: template
  lambda: return (analogRead(A0) < 0.5);
```
Unfortunately this made the device unusable as it is not booting properly anymore. I see the status LED flashing but it is no longer connecting to WiFi. I have to find a way to recover it somehow... I hope to find a way getting it into safe mode.

As this version of the SP112 is hard to open and debug I flashed the code to a Wemos D1 mini where I can get the debug logs via USB easily. There I can see what is happening. It somehow picks up a few WiFis (only a small subset of what is usually recognized), sometimes it doesn't find any. Eventually it does not connect. I can't really tell who reading an analog value is interfering with WiFi connection, but when flashing it again with `digitalRead` it immediately connects as always.

> [20:59:24]sd[I][logger:166]: Log initialized
[20:59:24][C][status_led:014]: Setting up Status LED...
[20:59:24][C][ota:366]: There have been 2 suspected unsuccessful boot attempts.
[20:59:24][I][app:029]: Running through setup()...
[20:59:24][D][binary_sensor:034]: 'Wemos 1  Status': Sending initial state OFF
[20:59:24][D][text_sensor:015]: 'Wemos 1 ESPHome Version': Sending state '1.16.0-dev Oct 21 2020, 20:52:40'
[20:59:24][C][wifi:033]: Setting up WiFi...
[20:59:24][D][wifi:324]: Starting scan...
[20:59:24][D][sensor:092]: 'Wemos 1 Uptime': Sending state 0.10700 s with 0 decimals of accuracy
[20:59:24][D][binary_sensor:036]: 'Button1': Sending state OFF
[20:59:30][D][wifi:339]: Found networks:
[20:59:30][D][wifi:389]: - 'xxxxx' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[20:59:30][W][wifi:394]: No matching network found!
[20:59:35][D][wifi:324]: Starting scan...
[20:59:41][D][wifi:339]: Found networks:
[20:59:41][D][wifi:341]:   No network found!
[20:59:46][D][wifi:324]: Starting scan...
[20:59:52][D][wifi:339]: Found networks:
[20:59:52][D][wifi:341]:   No network found!
[20:59:57][D][wifi:324]: Starting scan...
[21:00:03][D][wifi:339]: Found networks:
[21:00:03][D][wifi:341]:   No network found!
[21:00:04][D][sensor:092]: 'Wemos 1 Uptime': Sending state 39.70200 s with 0 decimals of accuracy
[21:00:08][D][wifi:324]: Starting scan...
[21:00:14][D][wifi:339]: Found networks:
[21:00:14][D][wifi:341]:   No network found!
[21:00:19][D][wifi:324]: Starting scan...
[21:00:24][I][wifi:116]: Starting fallback AP!
[21:00:24][C][wifi:161]: Setting up AP...
[21:00:24][C][wifi:163]:   AP SSID: 'wemos_1'
[21:00:24][C][wifi:164]:   AP Password: 'xxxxxxxx
[21:00:25][C][wifi:173]:   IP Address: 192.168.4.1
[21:00:25][D][wifi:339]: Found networks:
[21:00:25][D][wifi:389]: - 'xxxxx' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[21:00:26][D][wifi:389]: - 'xxxxx' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[21:00:26][D][wifi:389]: - 'xxxxx' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[21:00:26][D][wifi:389]: - 'xxxxx' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[21:00:26][W][wifi:394]: No matching network found!

glmnet, 2020-10-21:
Seems like ESP8266 doesn't like to analogRead very often!

Check this:

https://arduino.stackexchange.com/questions/19787/esp8266-analog-read-interferes-with-wifi

I'll start thinking in a port expander now, or in any case try switching to another port where you don't need fast updates maybe?

gitolicious, 2020-10-21:
Unfortunately ping usage is not up to me to decide but the engineers at Gosund, or whoever designs their products. Who knows how they came up with the idea of using the analog pin for the button on the SP112.

gitolicious, 2020-10-21:
Oh and thanks for the StackExchange link - sounds like this is the reason for my troubles. Let's see if I find a way to get into safe mode for recovery or I need to open it the hard way over the weekend.

gitolicious, 2020-10-22:
Following [a disassembly video](https://www.youtube.com/watch?v=hfYFB1UENTQ&t=2m06s) I was able to open my malfunctioning SP112. Powering the ESP directly with 3.3V and triggering the reset pin 10 times, I got the device into safe mode where it connected to WiFi successfully. OTA was then possible and it is back to normal operations. 

Coming back to my original issue, I am again stuck with the slight delays due to the fixed `update_interval`. One alternative which makes the code a bit more compact is this:

```yaml
sensor:
- platform: adc
  pin: A0
  id: button_adc
  update_interval: .2s
  on_value_range:
    below: 0.5
    then:
    - switch.toggle: switch1
```

You are losing the nice features of `binary_sensor` like `on_double_click ` though which is specifically useful for the ESP112 to toggle the USB ports.

emptygalaxy, 2021-02-04:
If you're interested in getting the `on_double_click` functionality you can use this to update the state of a `binary_sensor`

```
binary_sensor:
  - platform: template
    name: "Binary Switch"
    id: binary_switch

sensor:
  - platform: adc
    id: button_adc
    pin: A0
    update_interval: .2s
    on_value_range:
      - above: 0.5
        then:
        - binary_sensor.template.publish:
            id: binary_switch
            state: OFF

      - below: 0.5
        then:
        - binary_sensor.template.publish:
            id: binary_switch
            state: ON

```

You can swap the state: ON/OFF in case you want the inverted behavior.

Oderik, 2021-04-12:
The Tasmota firmware supports this device and features an ADC button option. Maybe their implementation helps as a template.

nagyrobi, 2022-07-05:
Implemented: https://esphome.io/components/binary_sensor/analog_threshold.html

closed 2022-07-05 by nagyrobi