esphome/feature-requests #955: Adding support for Micro:bit v2 sensors and actuators

rapport, 2020-10-19:
**Describe the problem you have/What new integration you would like**

The new BBC Micro:bit v2 is currently being rolled out (November 2020 release) and is based around the nRF52 so doesn't have a direct connection to the internet. This is a very important and popular device for young makers and it would be great if we could build an integration for them so the many sensors on the devices and actuators could communicate with over devices over the internet.

**Please describe your use case for this integration and alternatives you've tried:**

If we could provide a low cost bridge for the new Micro:bit device to communicate over MQTT it would make a lot of communication possible that is currently inaccessible to many young makers and would also help lots of other fun areas like gaming etc. I have noted the support for many bluetooth devices on this project, think it fits well and am happy to help with this integration too!

**Additional context**

This new v2 model will be a very popular, it also will be very influential regards helping the community and getting young people involved in home automation! The spec of the device follows, I have some early models for testing and it will be released in November:
https://tech.microbit.org/hardware/

rapport, 2020-10-20:
Aaron Christophel made a really interesting video about changing the firmware on a BLE device so it could work with a ESPHome gateway. I think we would have to do some similar thing with the firmware on the the Micro:bit to get this working but I am sure this will be much easier than want Aaron did.  Aaron kindly shared his video here: https://www.youtube.com/watch?v=NXKzFG61lNs&feature=youtu.be

madb1lly, 2020-10-21:
Hi @rapport,
I think this could equally apply to the v1 Micro:bit, if the comms to the ESP32 is just using BLE.
Cheers 🙂

rapport, 2020-10-22:
I will have a go at this and have joined the ESPHome discord to ask questions where I am getting stuck as I am sure  will need some help! For any ESPHome guys who are not working with the BBC:Microbit and STEM it is still definitely worth looking at this too especially the Micro:bit v2... It is a really low cost SOC package, it has a mems microphone, speaker and AI possibilities so for example you should be able to detect a Glass Break or sound messages the in the home! It also has many other capabilities I can't find easily elsewhere in the ESP home library that could be useful, things like compass, tilt, nice LED matrix, very sensitive touch/wind sensing etc.