esphome/feature-requests #958: IRALAN / CHANCEJIN led chandeliers lights remote control via bluetooth

slydiman, 2020-10-21:
There are a lot of IRALAN / CHANCEJIN led chandeliers lights, which could be controlled by the bluetooth remote controller and Android/iPhone app PC.Lamper, SmartControl or similar. 

It would be very useful to simulate the IRALAN remote control or the app PC.Lamper to control IRLAN lights.

Apps are available here https://new.zscbled.com/app/download
Here is the store https://www.aliexpress.com/item/4000314564312.html

new-kirte, 2021-05-01:
Yes, it would be great

dreamer2, 2021-05-02:
seems it use some ble type commands from app
but in btsnoop_hci.log i dont see nothing similar to commands

golddragon007, 2021-06-11:
I've checked now also with a BLE sniffer, but I was not able to identify the lamp yet. With btsnoop_hci.log I was not able to catch any log that's interesting I think. So... maybe it's some custom radio thing on 2.4g frequency?

new-kirte, 2021-06-12:
@golddragon007, could it be an RF, but on 2.4Ghz frequency? How to control it then?

golddragon007, 2021-06-12:
@new-kirte Yes, I have that feeling it's some kind of RF on the Bluetooth frequency. But I can't check it as I don't have the environment and tools to analyse it. I've also disassembled the remote, but it's a stupid simple thing. My current suspicion is that this sends 2.4g signals changing by time. I.e. the different commands are determined by how long a signal is on/off (but the pairing process is more interesting how it would work).
[image]
[image]

slydiman, 2021-06-12:
Don't forget it can be controlled from a smartphone via standard Bluetooth. Look at Android apps in the first post.

golddragon007, 2021-06-12:
Yes, I know. I also unpacked it and converted the dex file to java, but... It didn't help me too much. I'm kinda lost...

xrays72, 2021-11-27:
Hi. Did anyone manage to figure out the controls?

golddragon007, 2021-11-28:
I had no success as I can't see the messages for some reason. Also, the air is soo polluted with traffic that I can't find any relevant thing. :(

new-kirte, 2021-12-04:
I added Sonoff ZBR3 relays to my ceiling lights, so now I can control them via Zigbee. I can manage all 4 modes (neutral, cold, warm, and night) via switch off/on and if you need dimming, then you can use Paulmann Cephei dimmable relay

xrays72, 2022-05-08:
> I added Sonoff ZBR3 relays to my ceiling lights, so now I can control them via Zigbee. I can manage all 4 modes (neutral, cold, warm, and night) via switch off/on and if you need dimming, then you can use Paulmann Cephei dimmable relay

Can you show this on photo ?

v1ack, 2022-08-27:
hello. I decompiled the app and reached out how to manage my lamp.
I've made a service running on my raspberry and connected to Home assistant. Now I working on native HA integration.
Repo: https://github.com/v1ack/smartLightConnector

But as for me there is an app called GM Smart Light