esphome/feature-requests #963: Add physical resource consumption diagnostic sensors

ptr727, 2020-10-22:
**Describe the problem you have/What new integration you would like**

I am troubleshooting random resets, and I would like to correlate device behavior with device load over time.
I would like a sensor to periodically report resources like; CPU load, task execution, free heap, memory usage, etc.

**Please describe your use case for this integration and alternatives you've tried:**

Monitoring resource consumption over time is a valuable diagnostic for detecting e.g. tight loops, memory leaks, heap fragmentation, etc.

**Additional context**

The capability could be added to the debug component.
Possibly with a filter for the types of resources to monitor, and the frequency of reporting.
The sensor could report the values to HA as a sensor, and I can report on utilization from e.g. Grafana.

Similar in capability to e.g. https://github.com/letscontrolit/ESPEasy/issues/1656#issuecomment-433234295

micronen, 2021-03-05:
I've created a heap sensor, hope it will help debug some of your problems
[Add Heap Sensors - free/max block/fragmentation #1578](https://github.com/esphome/esphome/pull/1578)

davey, 2021-11-10:
I'd be interested in this too, in my case especially the **CPU** usage/load would be interesting.

micronen, 2021-11-14:
> I'd be interested in this too, in my case especially the **CPU** usage/load would be interesting.

I've added free loop time to my branch. I'm still waiting for the previous PR to be accepted.
You can check it out (and some other things I implemented) at https://github.com/micronen/esphome/tree/b8515509ddcd7aab31381f18b0e9650bee852522

I'm using the following config to get all the debug information to HA 
```
debug:
  update_interval: 60s
  device:
    name: $name Device Info
  free:
    name: $name Heap Free
    filters:
      delta: 100
  fragmentation:
    name: $name Heap Fragmentation
    filters:
      delta: 1
  block:
    name: $name Heap Max Block
    filters:
      delta: 100
  loop_time:
    name: $name Loop Time
    filters:
      delta: 1
  loop_delay_time:
    name: $name Loop Delay Time
    filters:
      delta: 1
  loop_high_frequency:
    name: $name Loop High Frequency
  blocking_component:
    name: $name Blocking Component
  blocking_component_time:
    name: $name Blocking Component Time
```

closed 2022-07-06 by nagyrobi