esphome/feature-requests #966: WLED sync for platform rgbww

pashdown, 2020-10-25:
**Describe the problem you have/What new integration you would like**
Enable WLED sync effect for platform RGBWW and single RGB bulbs.

**Please describe your use case for this integration and alternatives you've tried:**
WLED doesn't support PWM for single RGB bulbs, and I'd rather use esphome anyway.  Although WLED is an effect for the neopixelbus, it would be nice to sync my esphome RGB bulbs to the WLED strips in the area.

**Additional context**
Example:
```
output:
  - platform: esp8266_pwm
    pin: GPIO14
    id: output_blue
  - platform: esp8266_pwm
    pin: GPIO12
    id: output_green
  - platform: esp8266_pwm
    pin: GPIO4
    id: output_red
  - platform: esp8266_pwm
    pin: GPIO5
    id: output_coldwhite
  - platform: esp8266_pwm
    pin: GPIO13
    id: output_warmwhite

light:
  - platform: rgbww
    name: ${friendly_name}
    id: ${device_name}
    default_transition_length: 1.5s
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_coldwhite
    warm_white: output_warmwhite
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    restore_mode: RESTORE_DEFAULT_OFF
    color_interlock: true
    effects:
      - wled:
         port: 21324

```
Result:
```
      Unable to find effect with the name 'wled'.
      wled:  [source /config/esphome/include/effects.yaml:269]
        port: 21324
```

Sergey-SRG, 2020-10-26:
WLED support only the addressable led strings!

pashdown, 2020-10-26:
Yes, I understand that.  Is there a reason that a single RGB bulb can't emulate a single LED of that string?

mirko, 2021-02-01:
I'd also fancy this, there's a lot one could do with a simple RGB(W) strip to keep them in syc with the fully-addressable ones.

oxan, 2021-06-04:
This is basically the same as #1107 (as in, a fix for either would most likely fix the other as well).

mrand, 2022-05-01:
Did anyone (@oxan?) figure out how to use the partition functionality to implement WLED sync to a bulb?   #1107 even mentions that it should help, but I'm not seeing how...

WeekendWarrior1, 2022-05-02:
> Did anyone (@oxan?) figure out how to use the partition functionality to implement WLED sync to a bulb? https://github.com/esphome/feature-requests/issues/1107 even mentions that it should help, but I'm not seeing how...

I think it's something like this:
```yaml
light:
  - platform: partition
    name: "Partition Light"
    segments:
      - single_light_id: light1
    effects:
      - wled:
         port: 21324
      
  - platform: rgbww
    name: "Light 1"
    id: light1
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_white_cold
    warm_white: output_white_warm
    cold_white_color_temperature: 5700 K
    warm_white_color_temperature: 3000 K
    restore_mode: RESTORE_DEFAULT_ON
    constant_brightness: true
    color_interlock: true
...
```

Tuckie, 2022-06-25:
were you able to get this work, I still get an error `Unable to find effect with the name 'wled'.`

mrand, 2022-06-25:
> were you able to get this work, I still get an error `Unable to find effect with the name 'wled'.`

Yes and no.  I got it to compile, but it doesn't sync.  Not sure where the fault lies, but I did open issue https://github.com/esphome/issues/issues/3296 on one error.

You're probably missing the "wled:"

```yaml
wled:

light:
  - platform: rgbw
    id: ${devicename}_light
    name: "RGBW 3"
    default_transition_length: 0s
    red: output_red
    green: output_green
    blue: output_blue
    white: cw

  - platform: partition
    name: "RGBW 3 Partition"
    segments:
      - single_light_id: ${devicename}_light
    effects:
      - wled:
```

Drun555, 2022-07-24:
@nagyrobi can we re-open this issue? Offered alternative do not work anyway, and we need to discuss somewhere how badly we need e131/wled effect in our bulbs.

RitwickVerma, 2023-01-04:
@Drun555 Add OpenRGB to that light as well! It'd be super awesome indeed. Adding this comment as a vote towards this feature!

The solutions mentioned https://github.com/esphome/feature-requests/issues/910 do not work for me.

Drun555, 2023-01-09:
> @Drun555 Add OpenRGB to that light as well! It'd be super awesome indeed. Adding this comment as a vote towards this feature!
> 
> The solutions mentioned #910 do not work for me.

For myself, I felt quite satisfied with ESPHome's DDP mode - this way I added my bulbs to the WLED as separate diods.

```
- lambda:
    name: DDP
    update_interval: 0s
    lambda: |-
      // statics in light effects are like globals in Arduino
      static float scaled_r = 0.0;
      static float scaled_g = 0.0;
      static float scaled_b = 0.0;
      static std::unique_ptr<WiFiUDP> ddp_udp;
    
      // allocate and start UDP
      // this is like the start() in an Arduino sketch
      if (!ddp_udp) {
        ddp_udp = make_unique<WiFiUDP>();
        if (!ddp_udp->begin(4048)) {   // always listen on DDP port
        return;
        }
      }
      
      // the rest is like the loop() in an Arduino sketch
      // read UDP payload
      std::vector<uint8_t> payload;
      while (uint16_t packet_size = ddp_udp->parsePacket()) {
        payload.resize(packet_size);
        if (!ddp_udp->read(&payload[0], payload.size())) {
        continue;
        }
      }

      // ignore small payload
      if (payload.size() < 2) {
        return;
      }

      // do the thing
      float r = (float)payload[10]/255.0f;
      float g = (float)payload[11]/255.0f;
      float b = (float)payload[12]/255.0f;
      
      float m = 0.0f;
      if ( (r>=g) && (r>=b) ) { m = r; }
      else if ( g >= b )    { m = g; }
      else              { m = b; }
      
      if (m != 0.0f) {
        scaled_r = r/m;
        scaled_g = g/m;
        scaled_b = b/m;
      } else {
        scaled_r = 0.0f;
        scaled_g = 0.0f;
        scaled_b = 0.0f;
      }
      auto call = id(${device_name}).turn_on();
      call.set_transition_length(0);
      call.set_brightness(m); 
      call.set_color_mode(ColorMode::RGB);
      call.set_rgb(scaled_r, scaled_g, scaled_b);
      call.set_publish(false);
      call.set_save(false);
      call.perform();
```

closed 2025-04-21 by pashdown