esphome/feature-requests #969: Support for Home Assistant's input select integration

rob-deutsch, 2020-10-29:
**Describe the problem you have/What new integration you would like**

Support for [Home Assistant's Input Select](https://www.home-assistant.io/integrations/input_select/) integration.

Not all switches are simple on/off switches. Many switches are multi-gang switches that allow you to select many options.

Some Home Assistant/ESPHome components have this buit in (e.g. climate, fan), but there no way to define a generic one.

**Please describe your use case for this integration and alternatives you've tried:**

I have numerous electronic 'switches' around my home, which don't fit into other Home Assistant integrations, that are essentially enum inputs.

**Additional context**

I would add an ESPHome component myself, but would the API component need to be updated to allow for an input select component?

AAllport, 2020-10-30:
Another use case:
Adding "Modes" to a homemade device.
I have an ESP driven light strip, with a DHT22 and PIR.

At present, there is a script on the device that turns the light on for some a predetermined amount of time when the PIR is triggered.
I'd like to add a "mode" system to the device, to control whether the PIR should trigger the lights and whether it should be a high or low setting.

Currently, the "best" way to achieve this is via some input_selects & automations on the HA side of things.
However, I have a philosophy that my devices should be able to work without a network. Even if that operation is degraded.

It would therefore be great to define a variable on the device that is controlled by an input_select on the HA side of things, allowing the ESP to operate with its last set mode without networking.

Tuckie, 2020-11-07:
I absolutely agree with this one! I think we could also expand upon this with the other HA input types: number, boolean, datetime, text. More dynamic settings are great for quickly changing the behavior of a device without reflashing or stringing up a bunch of HA dependent items.

In my case, I'm using a power meter, and I want to know when a user-defined wattage is reached, so I have created a binary sensor to react quicker and cut down on chatter to HA. Supporting input number would make it easy to create a wattage slider that to change the threshold.

Syndlex, 2020-11-11:
@Tuckie  isn't this possible with user defined services?

Tuckie, 2020-11-11:
Unless there is a piece of this I am missing, you can string it up but it requires an HA input_number, template sensor, and user defined service. It just seems very circuitous and dependent upon HA. I would love to have this stored on the device and simplified from an implementation standpoint as I have a large number of these I need to setup.

Dynamic device settings seems like a very reasonable interface to expose.

Syndlex, 2020-11-11:
@Tuckie  yes you are right. That's a lot of thinks to manage.

Dynamic devices would be cool.

I haven't looked into the HA docs and I don't know if its even possible for an integration to create a input_select. since its only a helper for the user to create user defined stuff.

I know there is the Effect in the Light but this is just exposed from the light service.

AAllport, 2020-11-12:
Worth noting this is possible with MQTT, although that brings a decoupling between the config on HA and the esp home config. Not to mention, adds a dependency on MQTT

ssieb, 2020-11-12:
How is this possible with MQTT that is better than using the homeassistant sensor?

AAllport, 2020-11-15:
> @AAllport, how is it possible to do this with MQTT that is better than using the home assistant sensor?

HA: Input + Automation
ESP: MQTT Subscribe Text

It's a bit janky, but maybe a tad more stable than user-defined services.
Wouldn't for a moment suggest it's a good workaround, but a workaround nonetheless

ssieb, 2020-11-15:
I don't see how that's better than using the homeassistant sensor, but it's still irrelevant to this issue.  The point here is to automatically create an input in HA instead of configuring it there.

jesserockz, 2021-02-16:
HA has recently added the [number](https://www.home-assistant.io/integrations/number/) top level component which allows integrations to add these now, but it requires work on both sides to get this working.

If anyone wants to put their hand up to create this, please do so and comment here.

jesserockz, 2021-06-30:
Oh look, I ended up implementing this =)

closed 2021-07-12 by jesserockz