esphome/feature-requests #970: Decouple api component from async_tcp

rob-deutsch, 2020-10-29:
**Describe the problem you have/What new integration you would like**
Wi-Fi is power hungry. I would like to connect my ESPHome project to Home Assistant via an [RFM69 wireless network](https://www.sparkfun.com/products/12823).

**Please describe your use case for this integration and alternatives you've tried:**

The native API looks to only operate over TCP, which is fine, but given that it's a neat binary protocol it should be able to run over any transport.

**Additional context**

I imagine that this could be tackled by create an appropriate abstract base class for TCP, which will allow for all types of transports for the binary stream.

rob-deutsch, 2020-11-17:
Noting that https://github.com/esphome/esphome/pull/809 is along the lines of what I had in mind.

herostrat, 2021-09-09:
See: https://github.com/esphome/esphome/pull/809
If I understand correctly this should solve this request?

nagyrobi, 2022-07-06:
Assuming PRs above solve the request.

closed 2022-07-06 by nagyrobi