esphome/feature-requests #973: Run custom code before OTA starts

jodur, 2020-10-31:
**Describe the problem you have/What new integration you would like**

I would like to have to have the possibility to run code/lambda functions just before OTA starts. I have build a custom implementation that is using HW interrupts to run a custom code (see: [https://github.com/jodur/ESPEASY_Plugin_ITHO](https://github.com/jodur/ESPEASY_Plugin_ITHO))
When OTA is starting and during OTA a interrupt occurs the OTA fails. It would be nice to have a method to detect that OTA is initiated so i could disable for example the interrupts to insure OTA is not disturbed by the interrupts.

For example you coud add an extra automation  on_ota (in addition to on_boot,on_shutdown) to ESPHOME Core.

In this case it would be easy to autmatically detach the interrupt from GPIO, before OTA is starting.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

pat1300, 2020-10-31:
Did you try to use on_shutdown to put your code?
At https://esphome.io/components/esphome.html#esphome-on-shutdown : it says  code on_shutdown is called when Over-The-Air updates is being applied.

jodur, 2020-10-31:
No, this is to late. It is called after the new binary is downloaded and about to be aplied before reboot. I need to stop the HW interrupts during the upload of the new Firmware. Then interrupts are effecting the OTA upload.

RadekHvizdos, 2020-11-26:
I also think having the start of ota exposed as an event would be useful. Also mentioned in https://github.com/esphome/feature-requests/issues/904

mmakaay, 2021-04-28:
I recently created a pull request that would solve this.
https://github.com/esphome/esphome/pull/1714

closed 2021-07-14 by jodur