esphome/feature-requests #974: PZEM-016 integration

KameDomotics, 2020-10-31:
**Describe the problem you have/What new integration you would like**

Integration of PZEM-016 AC power meter

**Please describe your use case for this integration and alternatives you've tried:**

Would it be possible to integrate PZEM-016 into ESPHome?

I would like to use it to measure (and integrate in HomeAssistant) the electric current and parameters of an electricity meter

**Additional context**

randybb, 2020-10-31:
Just curious. Is there any reason why you need Modbus power meter instead of PZEM-004T V3 that is newer, smaller, with better accuracy, with Modbus RTU over UART (you don't need RS485 transceiver, so it is easier to connect) and being supported?

KameDomotics, 2020-10-31:
> Just curious. Is there any reason why you need Modbus power meter instead of PZEM-004T V3 that is newer, smaller, with better accuracy, with Modbus RTU over UART (you don't need RS485 transceiver, so it is easier to connect) and being supported?

because I already own two PZEM-016.....

randybb, 2020-10-31:
It is exactly as I thought - they still use the same protocol, so one option is to do a small [modification](https://github.com/Gio-dot/PZEM-016-OLED-2-OUT-ESPHome) to get rid of the RS-485 transceiver OR you can connect RS-485 transceiver with a driver (this is important!, because standard transceivers are half duplex, thus have a driver pin for switching direction) to ESP UART port. In both cases the standard [pzemac](https://esphome.io/components/sensor/pzemac.html) platform should work.

For example [here](https://www.aliexpress.com/item/4000974923186.html) you can find some transceiver boards. _1_, _2_ and _4_ are with integrated drivers - basically it does conversion from TTL UART to RS-485 - **this is what you would need**. _3_ is just a bare transceiver - **this is currently not supported.**

KameDomotics, 2020-10-31:
I have already made this change on  PZEM (it was made for an esp32 with Tasmota)

[image]

[image]

Gio-dot, 2020-11-01:
> I have already made this change on PZEM (it was made for an esp32 with Tasmota)
> 
> [image]
> 
> [image]

Ok, now you can use it with esphome with this component https://esphome.io/components/sensor/pzemac.html
There is an issue in that documentation but you can use this setting https://github.com/Gio-dot/PZEM-016-OLED-2-OUT-ESPHome/blob/master/wemos_d1_pzem016_display.yaml
There is already a pull request made by me for correct pzemac documentation https://github.com/esphome/esphome-docs/pull/636

closed 2022-06-29 by nagyrobi