esphome/feature-requests #977: support for Smart Meter ZMAi-90

arko36, 2020-11-02:
**Describe the problem you have/What new integration you would like**

I do have this meter in the main panel and I would like to have the integration using esphome without MQTT
It would be very useful to add this device in the sensors menu of Esphome

**Please describe your use case for this integration and alternatives you've tried:**

I do need MQTT running to get it in home assistant.

**Additional context**

https://kvvhost.ru/2020/04/18/zmai-90-tasmota/
https://www.creationfactory.co/2019/12/reverse-engineering-zmai-90-din-rail.html
[image]

dentra, 2021-04-09:
All upgraded versions of ZMAi-90 supplied with TuyaMCU, so in this case you already have support for it. Please check your device and maybe you get lucky. 

Here is TuyaMCU datapoints for ZMAi-90:
```yaml
substitutions:
  dp_switch: "1"
  dp_total: "17"
  dp_power: "19"
  dp_current: "18"
  dp_voltage: "20"
```

jasimancas, 2021-08-04:
Any news to integrate this device?
I have it currently installed with tasmota+mqtt, if any data is needed I can provide it.

dentra, 2021-08-04:
Here is config for my device https://github.com/dentra/esphome-components/blob/master/zmai90.yaml

If you have other datapoints in your device I can help you integrate it.

jasimancas, 2021-08-04:
I think I don't have the same version, I have the first one, which could be flashed with tuya-convert and I use the serialSend to get the data:
[image]

Is it possible to change tasmota for esphome?

dentra, 2021-08-05:
If it is not Tuya based device it is possible, but you should know the format of received result to transform it to sensors. It can be done with help of custom uart component. Querying is very easy by uart.write.

jasimancas, 2021-08-10:
This:
[image]

Send fe010f080000001c every 5 seconds and I get that long chain, each part of the string is a value given by the meter.

jasimancas, 2021-08-12:
> Try it: https://github.com/dentra/esphome-components/tree/master/components/zmai90v1
> 
> But it I do not have this type of device, so it untested and I need feedback and logs.

And the switch? Its possible?
It looks great!
[image]

dentra, 2021-08-12:
Yes, it possible. But first try to configure it by using existing `ESPHome` components and if it will work I'll modify my component:
```yaml
switch:
  - platform: gpio
    name: "Relay"
    id: relay12
    pin: GPIO12
    inverted: true  # need to be checked
binary_sensor:
  - platform: gpio
    pin: GPIO13
    name: Button
    inverted: true  # need to be checked
    on_click:
    then:
        - switch.toggle: relay12
```

You can set update interval via `update_interval`, default is 60s. https://github.com/dentra/esphome-components/blob/master/components/zmai90v1/zmai90v1.cpp#L67

jasimancas, 2021-08-12:
Installed, but nothing:
Attached the log.
[Log.txt](https://github.com/esphome/feature-requests/files/6977115/Log.txt)

dentra, 2021-08-12:
In your log file I can see only 3 second and it is too small. Could you wait 3 minutes. Do you change `update_interval`?

In `Tasmota` SerialConfig is configured as 8E1. So try to add `parity: EVEN` to `uart`. 

Also I have add logging to send command.

jasimancas, 2021-08-12:
Now yes!! Attached log, but "energy" sensor, 5454460,0 its -> 5454,60 kWh
[Log.txt](https://github.com/esphome/feature-requests/files/6977285/Log.txt)

dentra, 2021-08-12:
That’s right the energy sensor unit of measurement is `Wh` as other ESPHome meters. But I know that Home Assistant has new great "Energy Dashboard" feature and in [demo](https://demo.home-assistant.io/#/energy) all cards show and probably expects data in `kWh`. If so, I could change energy unit of measurement to `kWh` to reduce boilerplate for users. 

The active, reactive and apparent should be in `W`, `VAr` and `VA` respectively.

At this moment you can freely change this behavior by using power of ESPHome - applying filter and changing unit of measurement.

dentra, 2021-08-12:
@jasimancas, did you try to configure switch and binary sensor? I need a feedback to complete the component.

jasimancas, 2021-08-12:
> @jasimancas, did you try to configure switch and binary sensor? I need a feedback to complete the component.

Yes, finally this work at expected:

switch:
  - platform: gpio
    name: Interruptor Contador
    id: relay12
    pin: GPIO12
    
binary_sensor:
  - platform: gpio
    pin: GPIO13
    name: Estado del contador
    on_click:
      then:
        - switch.toggle: relay12

dentra, 2021-08-12:
With no one inverted state?

jasimancas, 2021-08-13:
> With no one inverted state?

Yes:

[image]

dentra, 2021-08-13:
Thanks, I will add this functionality to component

jasimancas, 2021-08-13:
Thanks you very very much.

dentra, 2021-08-16:
Now this component is a switch, the button is bound to it automatically, but this behaviour can be disabled. Also default energy unit of measurement is `kWh`.

To configure you need to move the component from sensor to switch and remove gpio switch and binary sensor from your configuration sensor or follow example to configure it at https://github.com/dentra/esphome-components/tree/master/components/zmai90v1
 
Please leave me feedback about how it works and if all is OK, I'll try to contribute it to ESPHome to share your good experience to others.

jasimancas, 2021-08-17:
Now, i have this error:
[image]

dentra, 2021-08-17:
You did not read my message carefully - now it is a switch, not a sensor.

dentra, 2021-08-17:
You can copy sample config from README.md

jasimancas, 2021-08-17:
Yes, sorry.

It is already installed, but the physical button of the device does not work (before the changes it worked).

dentra, 2021-08-17:
Could you share your config?

jasimancas, 2021-08-17:
external_components:
  - source: github://dentra/esphome-components

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
  parity: EVEN

switch:
  - platform: zmai90v1
    name: Switch
    # main switch pin, GPIOPin, default: GPIO12
    switch_pin: GPIO12
    # button pin, GPIOPin, default: GPIO13
    button_pin: GPIO13
    # button, BinarySensor
    button:
      name: Button
      # automatically bind button to the main switch, boolean, default: true
      # works only with conjunction with button
      bind_to_switch: true
    # energy sensor in kWh, Sensor
    energy:
      name: Energia (Piso)
    voltage:
      id: voltage
      name: Voltaje (Piso)
    current:
      id: current
      name: Consumo actual (Piso)
    active_power:
      name: Energia Activa (Piso)
    reactive_power:
      name: Energia Reactiva (Piso)
    apparent_power:
      name: Energia Aparente (Piso)
    power_factor:
      name: Factor de Potencia (Piso)
    frequency:
      name: Frequencia de energia (Piso)
    update_interval: 5s

dentra, 2021-08-17:
And part of log when you press the button, please

jasimancas, 2021-08-17:
```
[16:09:41][D][sensor:127]: 'Consumo Instantaneo (Piso)': Sending state 1574.65710 W with 1 decimals of accuracy
[16:09:44][D][binary_sensor:036]: 'Button': Sending state OFF
[16:09:44][D][binary_sensor:036]: 'Button': Sending state ON
[16:09:46][D][zmai90v1:091]: Sending request: FE.01.0F.08.00.00.00.1C (8)
```

dentra, 2021-08-17:
Hmm, when you press the button it sending state OFF. It is looks like it is in inverted state.

dentra, 2021-08-17:
And the switch state is correct or inverted too?

jasimancas, 2021-08-17:
It is the main switch of the house, all the electricity in the house goes through it. Right now it is running, everything works.

dentra, 2021-08-17:
I mean state displayed in HA or web

jasimancas, 2021-08-17:
In HA, its ON

dentra, 2021-08-17:
And when you switch it in HA it works as expected?

dentra, 2021-08-17:
Probably I found my mistake. Could you please test it again?
If you want to test it today and make changes to work you need to change refresh time of external_components to something lower than default 1day,  it can be for example 4h:
```yaml
external_components:
  - source: github://dentra/esphome-components
    refresh: 4h
```

jasimancas, 2021-08-17:
Now yes!
```

[21:42:57][D][binary_sensor:036]: 'Button': Sending state ON
[21:42:57][D][binary_sensor:036]: 'Button': Sending state OFF
[21:42:57][D][switch:029]: '' Toggling OFF.
[21:42:57][D][switch:045]: '': Sending state OFF
[21:42:59][D][binary_sensor:036]: 'Button': Sending state ON
[21:42:59][D][binary_sensor:036]: 'Button': Sending state OFF
[21:42:59][D][switch:029]: '' Toggling ON.
[21:42:59][D][switch:045]: '': Sending state ON
```

dentra, 2021-08-17:
You do not set the name to the switch and so it not shown in HA. It is normal if you prefer this way.

The values are same as in default on click handler (50 and 350). Maybe I missed something else, I will try to investigate it. Thanks.

jasimancas, 2021-08-18:
If you need any help, here I am.

dentra, 2021-08-23:
A button is not a switch and not stay ON or OFF continuously it is a binary sensor and shows ON when the button is pressed and OFF when it released. 

As I can see in part of log you shown above, the button click is interpreted normally and main switch is toggling every time you pressed button.

So it seems that all things works correctly.

Liionboy, 2022-02-04:
Hello,

Anyone succesfully put Tasmota on a Zmai 90 with CBU Module?

Thanks

skazo4nik, 2022-06-12:
> Hello,
> 
> Anyone succesfully put Tasmota on a Zmai 90 with CBU Module?
> 
> Thanks

Seems not. I use tuya and local tuya integration for HA.

Here are dp_id's i got from iot.tuya.com

`{ "result": { "category": "zndb", "functions": [ 
{ "code": "switch_prepayment", "dp_id": 12, "type": "Boolean", "values": "{}" }, 
{ "code": "clear_energy", "dp_id": 14, "type": "Boolean", "values": "{}" }, 
{ "code": "charge_energy", "dp_id": 15, "type": "Integer", "values": "{\"unit\":\"kW·h\",\"min\":0,\"max\":50000,\"scale\":0,\"step\":1}" }, 
{ "code": "switch", "dp_id": 16, "type": "Boolean", "values": "{}" }, { "code": "alarm_set_2", "dp_id": 17, "type": "Json", "values": "{}" }, 
{ "code": "event_clear", "dp_id": 20, "type": "Boolean", "values": "{}" } ], "status": [ 
{ "code": "forward_energy_total", "dp_id": 1, "type": "Integer", "values": "{\"unit\":\"kW·h\",\"min\":0,\"max\":99999999,\"scale\":2,\"step\":1}" }, 
{ "code": "phase_a", "dp_id": 6, "type": "Json", "values": "{}" }, 
{ "code": "fault", "dp_id": 10, "type": "Bitmap", "values": "{\"label\":[\"ov_cr\",\"unbalance_alarm\",\"ov_vol\",\"undervoltage_alarm\",\"miss_phase_alarm\",\"outage_alarm\",\"magnetism_alarm\",\"terminal_alarm\",\"cover_alarm\",\"credit_alarm\",\"no_balance_alarm\",\"battery_alarm\",\"meter_hardware_alarm\"]}" }, 
{ "code": "switch_prepayment", "dp_id": 12, "type": "Boolean", "values": "{}" }, 
{ "code": "balance_energy", "dp_id": 13, "type": "Integer", "values": "{\"unit\":\"kW·h\",\"min\":0,\"max\":99999999,\"scale\":2,\"step\":1}" }, 
{ "code": "clear_energy", "dp_id": 14, "type": "Boolean", "values": "{}" }, 
{ "code": "charge_energy", "dp_id": 15, "type": "Integer", "values": "{\"unit\":\"kW·h\",\"min\":0,\"max\":50000,\"scale\":0,\"step\":1}" }, 
{ "code": "switch", "dp_id": 16, "type": "Boolean", "values": "{}" }, { "code": "alarm_set_2", "dp_id": 17, "type": "Json", "values": "{}" }, 
{ "code": "event_clear", "dp_id": 20, "type": "Boolean", "values": "{}" } ] }, "success": true, "t": 1655040556587, "tid": "a8210ba7ea5311ec87382aa367714e21" }`

Seems that 1 is power, 6 is metrics (wattage, voltage, ampers), 16 is switch, 14 is clear energy meter.

closed 2022-07-01 by nagyrobi