esphome/feature-requests #978: Integration new Xiaomi sensor door&windows (MCCGQ02HL)

ManuelAbril, 2020-11-03:
**Describe the problem you have/What new integration you would like**

I would like to have an integration with the new Xiaomi sensor door&windows (MCCGQ02HL) which has BLE sensor.

**Please describe your use case for this integration and alternatives you've tried:**

I would like to have a similar integration like other Xiaomi BLE devices, without Xiaomi Gateway.

**Additional context**

kar-vi-nen, 2020-11-12:
This is something I would also like to have.  I found this: 
https://github.com/custom-components/sensor.mitemp_bt/issues/115#issuecomment-724901955

That looks promising, hopefully that will also help esphome developers.

Lymkin, 2020-12-22:
Ditto

petermcandrew, 2020-12-24:
Would be great to see this

kar-vi-nen, 2020-12-26:
I did some testing and got it working, sort of. Unfortunately im not much of a coder myself.
I think this will be easy task for someone who wrote xiaomi_ble and/or xiaomi_mjyd02yla for esphome. This commit for other project (ble_monitor) contains all needed info for this sensor, i think. 
https://github.com/custom-components/ble_monitor/commit/3f27fbc633d40dca09a97dbf3e18800e0dfbe712

So, I created copy of xiaomi_mjyd02yla and modified it for MCCGQ02HL. I also had to modify xiaomi_ble components. I did not find any good class (temperature/humidity etc.) for this sensor so i tried to create my own called "opening". That isn't working yet.

It might work if you place files in this .zip to your esphome components folder, overwriting xiaomi_ble folder. 
[xiaomi_mccgq02hl.zip](https://github.com/esphome/feature-requests/files/5743246/xiaomi_mccgq02hl.zip)
Then with config like this:

sensor:
  - platform: xiaomi_mccgq02hl
    mac_address: "MA:C_:HE:RE"
    bindkey: "you_need_bind_key"
    battery_level:
      name: "Door sensor battery level"

You might get battery level reading out of it. I think it reports it every 24h or after 30seconds without battery. Im not sure if this works as im getting 100% all the time.

For example:
[16:45:19][D][xiaomi_ble:311]: Got Xiaomi MCCGQ02HL (MA:C_:HE:RE):
[16:45:19][D][xiaomi_ble:320]:   Battery Level: 100%

Door state is also shown in the logs, but that "opening" device class is half way done so you cant add opening: line to your config.

[16:46:57][D][xiaomi_ble:311]: Got Xiaomi MCCGQ02HL (MA:C_:HE:RE):
[16:46:57][D][xiaomi_ble:344]:   Opening:  on
[16:46:59][D][xiaomi_ble:311]: Got Xiaomi MCCGQ02HL (MA:C_:HE:RE):
[16:46:59][D][xiaomi_ble:344]:   Opening:  off

I have not event tested light sensor at all. Hope this helps. 
I can help with testing, i have esp32 ,two MCCGQ02HLs and bindkeys for them.

maciak, 2021-01-26:
Hi,
Was there any progress on this?

adamklis, 2021-02-16:
Hi, I have added an MCCGQ02HL component as a binary sensor in the forked repository. Component returns values for opening, light (binary), and battery level (percentage). The opening and light are working fine for me, but I can't test the battery level. It returns 100% once a day.
[My commit in the forked repository](https://github.com/adamklis/esphome/commit/52ebc33dcb7ff7764c14670a7d8472568a5f4828) and example part of the configuration:
```
sensor:

binary_sensor:
  - platform: xiaomi_mccgq02hl
    name: door sensor
    mac_address: MA:C_:AD:DR:ES:S1
    bindkey: "bind_key"
    battery_level:
      name: "Door sensor battery level"
    light:
      name: "Door sensor light"
    opening:
      name: "Door sensor open"
```
This solution is sufficient for me. You can test it and add it to the esphome repository if you want. I am testing it by replacing esphome directory in python3/dist-package directory.

The door sensor returns 0x01 if a door is closed and 0x00 if open. After one minute if the door opened returns 0x02. I think it depends on settings but I have no Xiaomi gate to check it.

maciak, 2021-02-16:
Awesome! If this could be a merged it would be super great :)

kar-vi-nen, 2021-02-17:
Hi, 

Nice work! I have tested it with two sensors and i can confirm that it works like it should!
I have no xiaomi gate either. My sensors report battery level after 1 minute without battery. But it's 100% always.

maciak, 2021-02-17:
Hi,
How to install it on pi4? I dont really know how to get to filesystem there. I have Operating System
Home Assistant OS 5.11

roondar, 2021-02-18:
> Hi, I have added an MCCGQ02HL component as a binary sensor in the forked repository. Component returns values for opening, light (binary), and battery level (percentage). The opening and light are [...]

Hey great work but why you didn't create a pull Request ?

kar-vi-nen, 2021-02-18:
> Hi,
> How to install it on pi4? I dont really know how to get to filesystem there. I have Operating System
> Home Assistant OS 5.11

If you are planning to use pi4s onboard bluetooth you don't actually need esphome. I think what you need is this https://community.home-assistant.io/t/passive-ble-monitor-integration-xiaomi-mijia-ble-mibeacon-monitor/177352

maciak, 2021-02-20:
> If you are planning to use pi4s onboard bluetooth you don't actually need esphome. I think what you need is this [...]

Hey, I'm using this currently but it's almost out of range and sometimes just does not work. I have an esphome on esp32 that reads some sensors already closer and would be great to add support for it there.

cdu913, 2021-03-07:
> Hey, I'm using this currently but it's almost out of range and sometimes just does not work. I have an esphome on esp32 that reads some sensors already closer and would be great to add support for [...]

Hi, any success on this ?

maciak, 2021-03-08:
> Hi, any success on this ?

Nope. Still don't know how to add this to filesystem on Hassio

morph027, 2021-03-27:
Just received my sensors and got it working using @kar-vi-nen code :tada: :heart:

I've created a PR in https://github.com/esphome/esphome/pull/1645

maciak, 2021-06-06:
Any changes to get this moving forward? :)

Tien76, 2021-06-12:
> Hi, I have added an MCCGQ02HL component as a binary sensor in the forked repository. Component returns values for opening, light (binary), and battery level (percentage). The opening and light are [...]

Hi, thanks for your guided. However, I don't know how to read Value of sensor which returned. I try to run some python code on google but it just returns 0x00 and not change the Value. I'm also not sure about the Characteristic UUID correct. Can you help a tutorial about code (python) and provide some document about MCCGQ02HL. 

Hope you reply soon and many thanks.

gianlucaf81, 2021-11-26:
hello, could you explain how to integrate this modification? Where should the files be copied? Thanks

szilardcsotya, 2021-12-03:
I have a problem integrated this sensor, the following screen appears uploading the file:
INFO Reading configuration /config/esphome/ablaknyitas-teszt.yaml...
Failed config

binary_sensor.xiaomi_mccgq02hl: [source /config/esphome/ablaknyitas-teszt.yaml:30]
  
  Platform not found: 'binary_sensor.xiaomi_mccgq02hl'.
  platform: xiaomi_mccgq02hl
  mac_address: E4:AA:EC:4B:A3:92
  bindkey: 6a756401718b9abcb2ea595edb52d057
  name: Konyha ablak Sensor
  light: 
    name: Konyha ablak Light
  alert: 
    name: Konyha ablak Alert
  battery_level: 
    name: Konyha ablak Battery Level
  battery_voltage: 
    name: Konyha ablak Battery Voltage

This is the config file:
sensor:
binary_sensor:
  - platform: xiaomi_mccgq02hl
    mac_address: "MAC address"
    bindkey: "the bind key"
    name: "Konyha ablak Sensor"
    light:
      name: "Konyha ablak Light"
    alert:
      name: "Konyha ablak Alert"
    battery_level:
      name: "Konyha ablak Battery Level"
    battery_voltage:
      name: "Konyha ablak Battery Voltage"

Thanks for the reply.

ahpohl, 2022-01-20:
Has anybody received the battery status?

I have a MCCGQ02HL running for 5 days on my hamster wheel and the open and light items are working great. The hamster is running 400 revolutions per hour during the night totaling on average 1.6 km and already drained the stock battery! Hopefully the Varta CR2032 I have put in now will last a bit longer.

I made some code comments in the PR to get it to compile.

szilardcsotya, 2022-01-20:
> Has anybody received the battery status?
> 
> I have a MCCGQ02HL running for 5 days on my hamster wheel and the open and light items are working great. The hamster is running 400 revolutions per hour during the night totaling on average 2500 open/close events per day and already drained the stock battery! Hopefully the Varta CR2032 I have put in now will last a bit longer.
> 
> I made some code comments in the PR to get it to compile.

I replaced every bluetooth door and window sensor to the zigbee version, work a lot better and faster than the MCCGQ02HL which is not a good choice for HA.

closed 2022-07-01 by nagyrobi