esphome/feature-requests #981: New Device: M5Stack ATOM Lite

yngndrw, 2020-11-05:
**Describe the problem you have/What new integration you would like**

Please add support for the new M5Stack ATOM Lite device:
https://docs.m5stack.com/#/en/core/atom_lite

This is a small low-cost device based upon the ESP32-PICO with a single button, a single NeoPixel, a Grove port and six additional IO pins. The device is within a small plastic case which also makes it convenient as a Wifi -> Bluetooth bridge.

**Please describe your use case for this integration and alternatives you've tried:**

No matching device exists within the ESPHome Wizard.

**Additional context**

randybb, 2020-11-05:
From Board perspective ESPHome supports everything that is supported by PlatformIO. M5Stack ATOM uses ESP32-PICO that is defined as "[ESP32 Pico Kit](https://docs.platformio.org/en/latest/boards/espressif32/pico32.html)" Board.
For pin mapping you will need to look into their documentation, or back side of that device (this is really nice on M5 devices).

yngndrw, 2020-11-05:
I was a little unsure about the importance of these board mappings, from the code I can see that there's a list of pins defined but in some definitions they are completely empty - Are these just used to provide a template?

From the processor perspective, the ATOM Lite claims to use a ESP32-PICO-D4 which is meant to be 240MHz dual core with 4MB of flash and 520KB of RAM and they say to set it as the M5StickC in the Arduino IDE. The ESP32-PICO is said to only have 320KB of RAM. Like I say, my consufusion is around which of these parameters are relevant to ESPHome and which are irrelevant - I'm thinking that the closest would be the M5StickC as the processor matches and it already exists within the ESPHome wizard?

randybb, 2020-11-05:
Nope. There is no template specific for different boards. It only sets name, platform, board, WiFi parameters and OTA password in generated yaml file. Sure, it is not very convenient for beginners, but to be honest it is not so difficult - you just need to select the right "board", define pinouts and combine all components you need together.

The board definitions for [ESP32 Pico Kit](https://docs.platformio.org/en/latest/boards/espressif32/pico32.html) and for [M5Stick-C](https://docs.platformio.org/en/latest/boards/espressif32/m5stick-c.html) are the same, so it is up to you - the difference is just in name.

yngndrw, 2020-11-05:
Thank you, the documentation on what exactly the board is used for isn't great so I appreciate you listing the important parts. The pins shouldn't be an issue, as you say the M5 documentation is really nice for this. I'll close this issue.

Thanks!

closed 2020-11-05 by yngndrw

Davst, 2021-03-08:
@yngndrw Seeing as I came here looking for this answer, the board name should still be added for better user experience.