esphome/feature-requests #983: Ability to disable mDNS in ESPHome firmware

JayNewstrom, 2020-11-05:
**Describe the problem you have/What new integration you would like**

I would like to disable mDNS in ESPHome firmware. I'm trying to build a device with minimal traffic, and this is one of the things that cause it to use more data than hand written firmware.

**Please describe your use case for this integration and alternatives you've tried:**

I've disabled and blocked mDNS on my router, but the packets are still sent from my ESPHome device.

**Additional context**

It looks like this functionality is driven by configuration near the [wifi component](https://esphome.io/components/wifi.html). I see the code that makes the mDNS calls here: https://github.com/esphome/esphome/blob/dev/esphome/core/util.cpp#L47

I propose we add a configuration value to the wifi component, and simply not make calls for mDNS.

I'm aware of the drawbacks of this, and OTA will be made more difficult without it.

mspinolo, 2021-03-19:
This would also tie up with this suggestion:

https://github.com/esphome/feature-requests/issues/769

oxan, 2021-09-22:
Closing as this is now implemented.

closed 2021-09-22 by oxan