esphome/feature-requests #987: OTA Update over MQTT

jperquin, 2020-11-07:
**Describe the problem you have/What new integration you would like**
Upload FW.bin over MQTT.
Could work by publishing something like this:
mytopic/sys/update = http://myserver.ip/update/version/fw.bin

**Please describe your use case for this integration and alternatives you've tried:**
I have a good number of ESPHome devices deployed to different remote sites that communicate over MQTT via WAN to a centrally hosted MQTT broker. Unless we start opening up ports on the local routers at those remote sites, OTA updates via the standard process won't work.

**Additional context**
An (unrelated) implementation that works well in Arduino: https://github.com/YorffoeG/diyscip/blob/master/src/OTAUpdate.cpp

jperquin, 2020-11-15:
A potential route described here: https://harizanov.com/2019/01/esp32-ota-using-mqtt-over-ssl-as-transport-and-node-red-as-the-bin-image-provider/

MagMeister, 2021-08-13:
+1 for this feature request

nkhome, 2022-01-16:
+++ this feature will make me and other integrators happy!!!
I have a 100+ devices outside LAN and I can't to update it remotely!!!

PecceG2, 2022-04-02:
+1
Having remote devices has always been a hassle, although it has improved a lot in recent years, this feature would be one of the main changes to make remote devices an elegant and simple solution.

wizzor, 2022-05-11:
I'd very much like this as well.

nagyrobi, 2022-07-05:
This request is in reality update via HTTP, just sending a Tasmota-like command to trigger it.
Closing it as duplicate of https://github.com/esphome/feature-requests/issues/801

closed 2022-07-05 by nagyrobi