esphome/feature-requests #994: Dallas onewire DS2423

ludogomez, 2020-11-15:
**Describe the problem you have/What new integration you would like**

is it possible to add a Dallas onewire device : DS2423, it's a counter.

**Please describe your use case for this integration and alternatives you've tried:**

I want to use it for a weather station in addition with DS18b20.

**Additional context**

you can find an arduino library : https://github.com/jbechter/arduino-onewire-DS2423

randybb, 2020-11-15:
This might be handy in some cases...
There are multiple requests for one wire abstraction, since it is currently used only the well known DS18B20 temp sensors.
Like mine for DS2413 (1-Wire Dual Channel Addressable Switch) https://github.com/esphome/feature-requests/issues/432 or any other addressable switches and then for bigger one-wire networks you will need one wire master https://github.com/esphome/feature-requests/issues/621 that will handle timing and can drive mosfet for stronger pull-up.

argo69, 2020-11-19:
Hello,
+1 , same request for me.
Even if DS2423 is discontinued, will be nice if one wire device is extended for that one.
Thanks a lot for the great job done yet.

lwqcz, 2020-11-20:
Would not be better to use some modern, supported chips and not ones that are EOL?

argo69, 2020-11-20:
For sure. I'm ready to use a modern chip but nothing found doing the same job:
2 counters and memory low power used on whether station.

Creo2005, 2022-02-02:
https://community.home-assistant.io/t/1-wire-counters-ds2423-and-home-assistant/219276

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/432

closed 2022-07-01 by nagyrobi