esphome/feature-requests #996: Support for ESP32-S devices

Blaise70, 2020-11-16:
**Describe the problem you have/What new integration you would like**

Do you plan to support the new ESP32 variant from Espressif called ESP32-S2. These new chips and dev boards have a single LX7 core and no BT, but WiFi only. I guess they will become a cheaper alternative for ESP32 where no BT and dual core is required.

**Please describe your use case for this integration and alternatives you've tried:**

No actual use case yet. I just started to play with these new Saola dev boards and thought it worth a question.

**Additional context**

misery, 2020-12-26:
Tried current development versions here: https://github.com/esphome/feature-requests/issues/871

misery, 2021-02-05:
https://github.com/platformio/platform-espressif32/releases Version 3 has support for ESP32-S2 now.

flpaoli, 2021-04-25:
@misery I tried updating Platformio and esphome to dev, then doing below:

In ~/.platformio/platforms:
`cp -R espressif32 'espressif32@3.1.1'`

Then in ~/.local/lib/python3.8/site-packages/esphome edited pins.py and added this line at line 380:
`    "esp32-s2-saola-1": {},`

I get the system to progress, but it fails claiming the board doesn't support arduino framework:

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-saola-1.html
PLATFORM: Espressif 32 (3.0.0) > Espressif ESP32-S2-Saola-1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.210126 (1.0.4) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32s2 1.80400.210211 (8.4.0)
Error: This board doesn't support arduino framework!

deosrc, 2021-06-12:
Also looking for this as the [Adafruit MagTag](https://www.adafruit.com/magtag) is based on the ESP32-S2 WROVER.

I've currently managed to get ESPHome onto it, connected to the wifi, talking with MQTT, the Neopixel lights working and the "up" button working. When I try to set up other buttons, they either aren't reporting being pressed or ESPHome complains that the pin is output only. I tried the screen but that's a long way off yet. I've not tried the other components.

Here's my config so far (wifi and MQTT moved to packages):

```yaml
substitutions:
  device_id: office-magtag
  device_name: Office MagTag

# Device Details
esphome:
  name: $device_id
  platform: esp32
  board: esp32dev
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
      - platformio/tool-esptoolpy @ ~1.30100
    board_build.mcu: esp32s2

# Common
packages:
  base: !include common/base.yaml

mqtt: !include common/mqtt.yaml

# Lights
light:
  - platform: fastled_clockless
    chipset: NEOPIXEL
    pin: GPIO1
    num_leds: 4
    name: ${device_name} Lights
    effects:
      - pulse:
      - random:
      - strobe:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:

switch:
  - platform: gpio
    pin:
      number: GPIO21
      inverted: yes
    name: ${device_name} Lights Power
    restore_mode: ALWAYS_ON
    internal: true

# Buttons
binary_sensor:
  - platform: gpio
    name: ${device_name} Button 2
    pin:
      number: GPIO14
      inverted: yes
```

jlunz, 2021-07-14:
Just for reference, here is my current config for a [ESP32-S2-Saola-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html):
```yaml
esphome:
  name: esp32-s2
  platform: ESP32
  board: esp32dev
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
      - platformio/tool-esptoolpy @ ~1.30100
    board_build.mcu: esp32s2

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: ''
  password: ''

light:
  - platform: fastled_clockless
    id: rgb
    name: "RGB GPIO 18"
    chipset: WS2812
    rgb_order: GRB
    pin: GPIO18
    num_leds: 1
    
binary_sensor:
  - platform: gpio
    name: "Boot Button"
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: true
```

lucky79, 2021-07-27:
@jlunz Thanks for your code, unfortunately I am getting this error

Tool Manager: Installing platformio/toolchain-xtensa32s2 @ ~1.80400.0
Error: Could not find the package with 'platformio/toolchain-xtensa32s2 @ ~1.80400.0' requirements for your system 'linux_aarch64'

Are you running 32 or 64bit HA?

Thanks!

jlunz, 2021-07-27:
@lucky79 That looks like there is no toolchain support for that particular version available on aarch64.
I am running the build inside a container on amd64, for reference this is the head of the build log:

```shell
Processing esp32-s2 (board: esp32dev; framework: arduino; platform: platformio/espressif32@3.2.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-xtensa32s2 @ ~1.80400.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa32s2 @ 1.80400.210211 has been installed!
Tool Manager: Installing platformio/toolchain-esp32s2ulp @ ~1.22851.0
Downloading  [####################################]  100%          
Tool Manager: toolchain-esp32s2ulp @ 1.22851.191205 has been installed!
Tool Manager: Installing git+https://github.com/espressif/arduino-esp32.git#master
git version 2.20.1
Cloning into '/root/.platformio/.cache/tmp/pkg-installing-mtj3glip'...
Checking out files: 100% (6484/6484), done.
Tool Manager: framework-arduinoespressif32 @ 0.0.0+sha.65eafd1 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.30100
Downloading  [####################################]  100%          
Tool Manager: tool-esptoolpy @ 1.30100.210531 has been installed!
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.65eafd1 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32s2 1.80400.210211 (8.4.0)
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <ESPmDNS> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <Wire> 2.0.0
|-- <FastLED> 3.3.2
|   |-- <SPI> 2.0.0
|-- <Update> 2.0.0
|-- <WiFi> 2.0.0
...
```

I did a quick check and it seems sth. has changed, now getting:
```shell
Linking .pioenvs/esp32-s2/firmware.elf
/root/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp32-s2/firmware.elf] Error 1
```

ChrisCrewdson, 2021-07-31:
I get the same error.

```
INFO Reading configuration /config/esphome/magtag.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/magtag
Processing magtag (board: esp32dev; framework: arduino; platform: platformio/espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.c5a1f3e 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32s2 1.80400.210211 (8.4.0)
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <FS> 2.0.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <FS> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <Update> 2.0.0
|-- <WiFi> 2.0.0
Linking /data/magtag/.pioenvs/magtag/firmware.elf
/root/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
*** [/data/magtag/.pioenvs/magtag/firmware.elf] Error 1
========================== [FAILED] Took 7.88 seconds ==========================
```

mikenabhan, 2021-08-18:
@ChrisCrewdson Are you able to share your yaml for the magtag?  I am gonna try to dig into this.

ChrisCrewdson, 2021-08-18:
> @ChrisCrewdson Are you able to share your yaml for the magtag? I am gonna try to dig into this.

Sure. Here's what I have right now:
```
esphome:
  name: magtag
  platform: esp32
  board: esp32dev
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
      - platformio/tool-esptoolpy @ ~1.30100
    board_build.mcu: esp32s2

logger:
api:
ota:
  password: !secret ota_password
wifi:
  ssid: "Monkey House"
  password: !secret wifi_password
  ap:
    ssid: "Magtag Fallback Hotspot"
    password: !secret wifi_ap_password
captive_portal:
mqtt: !include common/mqtt.yaml
```

mikenabhan, 2021-09-01:
Alright, I got it to compile.  Here is my starter GIST:

https://gist.github.com/mikenabhan/a14a74a04beac4bff15a32f6d3a2aa92

This will autogenerate the appropriate platformio.ini.  Thanks @Zagrophyte and @ChrisCrewdson for putting me on the right path.

sjtrny, 2021-09-01:
What's the process for uploading/flashing the magtag? I think I've bricked mine.

I got it to show up as ESP32-S2 in System Information by entering ROM boot loader mode and it seemed to upload ok but I can't get any log output from the serial port.

Zagrophyte, 2021-09-01:
I believe you can just re-install the boot loader via the method outlined
here:
https://learn.adafruit.com/adafruit-magtag?view=all#install-uf2-bootloader

Do we think we’ll need any special considerations here for PlatformIO to
avoid stepping on the UF2 portion?

mikenabhan, 2021-09-01:
I had to overwrite it. I also had to use esptool to flash because no other tool I had seemed to want to do it.

deosrc, 2021-09-01:
Ok I've been playing around a bit. Big thanks to @mikenabhan for the gist to get me a bit further.

So far I have working:
- OTA updates
- Wifi Logger
- Neopixel LEDs
- Red Status LED
- Button 2

I have also tried to configure but ran into issues with:
- Captive Portal (Compiler error if included)
- Button 3 (Flagged as flash interface)
- Button 1 & 4 (Always marked as "On" in HA)
- ADC (For light sensor, compiler error)
- Display (Flagged as flash and input only pins)

If I have the pins right in the yaml config, it looks like the pin validation is going to need updating for ESP32-S2 devices.

[MagTag Pins](https://learn.adafruit.com/assets/102127)
[ESP32-S2 DataSheet](https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf) (See section 2.2 Pin Descriptions)

Here is the config I'm using:
```yaml
substitutions:
  device_id: office-magtag
  device_name: Office MagTag

# Device Details
esphome:
  name: $device_id
  platform: esp32
  board: esp32dev
  platformio_options:
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0
      - platformio/tool-esptoolpy @ ~1.30100
      - espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r1
      - toolchain-xtensa32s2 @ 1.80400.210211
      - toolchain-esp32s2ulp @ 1.22851.191205
    board_build.mcu: esp32s2
    framework: arduino
    build_unflags:
      - -DARDUINO_ESP32_DEV
      - -DARDUINO_VARIANT="esp32"
    build_flags:
      - -DARDUINO_MAGTAG29_ESP32S2
      - -DARDUINO_SERIAL_PORT=1
      - -DARDUINO_VARIANT="adafruit_magtag29_esp32s2"
      - -DCORE_DEBUG_LEVEL=5
      - -DBOARD_HAS_PSRAM
    # https://community.platformio.org/t/board-definition-for-adafruit-magtag/20747/6

# Common
packages:
  base: !include common/base_modified.yaml

mqtt: !include common/mqtt.yaml

switch:
  - platform: gpio
    pin:
      number: GPIO21
      inverted: yes
    name: ${device_name} Aux Power # Used for LEDs and Light Sensor
    restore_mode: ALWAYS_ON
    internal: true # Don't export this, just leave it always on

# Lights
light:
  - platform: fastled_clockless
    chipset: NEOPIXEL
    pin: GPIO1
    num_leds: 4
    name: ${device_name} Lights
    effects:
      - pulse:
      - random:
      - strobe:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
  - platform: status_led
    name: ${device_name} Status LED
    pin: GPIO13

# Light Sensor
# FIXME: src/esphome/components/adc/adc_sensor.cpp:65:21: error: 'ADC_WIDTH_BIT_12' was not declared in this scope
# sensor:
#   - platform: adc
#     pin: A3
#     name: ${device_name} Light Sensor

# Buttons
binary_sensor:
  # FIXME: Always showing as "On" in HA for some reason???
  # - platform: gpio
  #   name: ${device_name} Button 1
  #   pin:
  #     number: GPIO15
  #     inverted: yes

  - platform: gpio
    name: ${device_name} Button 2
    pin:
      number: GPIO14
      inverted: yes
  # FIXME: Failing Validation Checks as this is a Flash pin on ESP32s
  # - platform: gpio
  #   name: ${device_name} Button 3
  #   pin:
  #     number: GPIO11
  #     inverted: yes
  # FIXME: Always showing as "On" in HA for some reason???
  # - platform: gpio
  #   name: ${device_name} Button 4
  #   pin:
  #     number: GPIO12
  #     inverted: yes
# Display
# font:
#   - file: "fonts/Comic Sans MS.ttf"
#     id: font1
#     size: 8

# spi:
#   clk_pin: GPIO36 # FIXME: Flagged as input only pin
#   mosi_pin: GPIO35
#   miso_pin: GPIO37 # FIXME: Flagged as input only pin

# display:
#   - platform: waveshare_epaper
#     cs_pin: GPIO8 # FIXME: Flagged as flash pin
#     dc_pin: GPIO7 # FIXME: Flagged as flash pin
#     busy_pin: GPIO5
#     reset_pin: GPIO6 # FIXME: Flagged as flash pin
#     model: 2.90in
#     full_update_every: 30
#     lambda: |-
#       it.print(0, 0, id(font1), "Hello World!");
```

mikenabhan, 2021-09-02:
@deosrc What is in your two !includes?  I can't get it to boot using that gist.

deosrc, 2021-09-02:
> 
> 
> @deosrc What is in your two !includes? I can't get it to boot using that gist.
> 
> FWIW I had the same issues with the pin mapping, I have a couple of ideas for workarounds though that I'll try once I can boot.

Sorry I missed them off because of some of the content. I've tweaked them and updated the gist with the extra files.

I made a start at trying to update the pin validation following what has been done for the ESP32-C3s. The pin validation errors have gone but I didn't manage to get anything additional working. Tbh I'm a bit out of my depth on this one.

Commit and branch with my changes on are here if it helps anyone:

Commit: https://github.com/deosrc/esphome/commit/958ac6704ec071a87b5f1ebd4571018b8c3f85ad
Branch: https://github.com/deosrc/esphome/tree/feature/magtag :warning: Highly experimental

If you're using the branch, you''ll need to update the device details in the config:
```yaml
esphome:
  name: $device_id
  platform: esp32
  board: adafruit-magtag # Change to this
  platformio_options:
    board: esp32dev # Add this
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
    platform_packages:
      ...
```

stumpylog, 2021-09-09:
Does the recent release of [espressif/arduino-esp32](https://github.com/espressif/arduino-esp32/releases/tag/2.0.0) mean this is likely to be in esphome more easily now?

It specifically mentions "Support for ESP32-S2"

comatose-tortoise, 2021-09-09:
> Does the recent release of [espressif/arduino-esp32](https://github.com/espressif/arduino-esp32/releases/tag/2.0.0) mean this is likely to be in esphome more easily now?
> 
> 
> 
> It specifically mentions "Support for ESP32-S2"

Also wondering about this!

mikenabhan, 2021-09-12:
This is starting to get off topic, it's a step forward but the biggest thing is that platform.io is going to have to support it.  Once that happens, it should be relatively quick to come into ESPHome.

eisenhowerj, 2021-09-13:
@mikenabhan its supported now in platformio:

With a freshly pip install esphome:
`~/tmp via 🐍 v3.7.8 (venv) 
﬌ platformio --version
PlatformIO Core, version 5.1.1

~/tmp via 🐍 v3.7.8 (venv) 
﬌ platformio boards | grep -i ESP32S2
featheresp32-s2                  ESP32S2  240MHz       4MB      320KB   Adafruit ESP32-S2 Feather Development Board
esp32-s2-kaluga-1                ESP32S2  240MHz       4MB      320KB   Espressif ESP32-S2-Kaluga-1 Kit
esp32-s2-saola-1                 ESP32S2  240MHz       4MB      320KB   Espressif ESP32-S2-Saola-1
`

RAYs3T, 2021-11-08:
@randybb So you're running ESP32-S2 chips with this?

randybb, 2021-11-08:
Nope, my assumption is just based on the documentation.

Sebmaster, 2021-11-08:
esp-idf does indeed allow running this on a Feather S2 for me. Even OTA upgrade works.

I had to patch the flash pin check out since the Stemma QT port uses pins 8+9.

lucky79, 2021-11-09:
I was looking forward to see support for ESP32-S2-Saola-1, but guess what.. The same problem I had before it was officially supported.

Using HA Blue HW (Odroid N2)

INFO Reading configuration /config/esphome/esp32-saola.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-s2 (board: esp32-s2-saola-1; framework: espidf; platform: espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-xtensa32s2 @ ~1.80400.0
Error: Could not find the package with 'platformio/toolchain-xtensa32s2 @ ~1.80400.0' requirements for your system 'linux_aarch64'

misery, 2021-11-09:
This works for me

```
esp32:
  board: esp32-s2-kaluga-1

esphome:
  name: dummy
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

status_led:
  pin: GPIO2
```

RAYs3T, 2021-11-09:
@misery  if I try the first configuration you mentioned with my `Olimex ESP32-S2-WROVER` it crashes (the build itself works), which is interesting, because the chip should be the same.

Also tried this with the `esp32-s2-saola-1`
```
Showing logs:
[11:29:44]ESP-ROM:esp32s2-rc4-20191025
[11:29:44]Build:Oct 25 2019
[11:29:44]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[11:29:44]Saved PC:0x4000f79a
[11:29:44]Invalid chip id. Expected 2 read 0. Bootloader for wrong chip?
[11:29:44]ets_main.c 386 
[11:29:44]ESP-ROM:esp32s2-rc4-20191025
[11:29:44]Build:Oct 25 2019
[11:29:44]rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[11:29:44]Invalid chip id. Expected 2 read 0. Bootloader for wrong chip?
[11:29:44]ets_main.c 386 
[11:29:46]ESP-ROM:esp32s2-rc4-20191025
[11:29:46]Build:Oct 25 2019
[11:29:46]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[11:29:46]Saved PC:0x4000f79a
[11:29:46]Invalid chip id. Expected 2 read 0. Bootloader for wrong chip?
[11:29:46]ets_main.c 386 
```

misery, 2021-11-09:
Ah, ok. I have a ESP32 S2 "Ai-thinker ESP 12K" [1] that runs without problems.

Did you try the esp-idf, too?

[1] https://www.mischianti.org/2021/02/27/esp32-s2-ai-thinker-esp-12k-high-resolution-pinout-and-specs/

RAYs3T, 2021-11-09:
Yes tried the idf aswell even with a lot of different platform options and settings. I'm at binary  27 now and still can't get it to work. Was a mistake buying the S2...

But I really wonder what's the reason for the bootloader error. It should be exactly the same chip and therefore bootloader as the SALOA.

RAYs3T, 2021-11-15:
FYI: Just tried to get this device running with the normal arduino studio and Arduino framework 2.0.1. Works normally.
But even with that version configured in the build it still fails with esphome (bootloeader code error).

So if anyone has a clue what I could do that would be appreciated.

volca, 2021-11-16:
> It's not possible to flash this with EsphomeFlasher (`Esp Chip Detection failed : Unexpected UART datacode value 0x000000500`).
> 
> Uploading it manually with `esptool.py` works but the code in the esp loops as it crashes with `invalid header: 0x6f742064`.
> 
> Olimex ESP32-S2-DevKit-Lipo

I tried flash with `esptool.py` and esp32 s2 works

```
esptool.py -p /dev/cu.SLAB_USBtoUART write_flash --flash_freq 40m --flash_size detect -z --flash_mode dio  0x1000 bootloader_dio_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware.bin
```

You can find out all bin files in the `.esphome` directory.

roger-, 2021-11-19:
FYI this works in Arduino mode:

```yaml
esphome:
  name: XXX
  platformio_options:
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
    board_build.mcu: esp32s2
    platform_packages:
    - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1

esp32:
  board: esp32-s2-saola-1
  variant: ESP32S2
```

jslawek, 2021-11-20:
Hi! Thanks for the sample configs, arduino mode compiles for me on ESP32S2.
But when I try to add ADC, for example:
```
sensor:
 - platform: adc
   pin: 7
   name: "vbat"
```

I get compilation error:

```
Compiling .pioenvs/test3/src/esphome/core/helpers.cpp.o
src/esphome/components/adc/adc_sensor.cpp: In member function 'virtual void esphome::adc::ADCSensor::setup()':
src/esphome/components/adc/adc_sensor.cpp:26:21: error: 'ADC_WIDTH_BIT_12' was not declared in this scope
   adc1_config_width(ADC_WIDTH_BIT_12);
                     ^~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:26:21: note: suggested alternative: 'ADC_WIDTH_BIT_13'
   adc1_config_width(ADC_WIDTH_BIT_12);
                     ^~~~~~~~~~~~~~~~
                     ADC_WIDTH_BIT_13
*** [.pioenvs/test3/src/esphome/components/adc/adc_sensor.cpp.o] Error 1
```
Am I missing some obvious include?

Halbgeviertstrich, 2021-12-12:
> @misery if I try the first configuration you mentioned with my `Olimex ESP32-S2-WROVER` it crashes (the build itself works), which is interesting, because the chip should be the same.
>
> Also tried [...]

Currently trying to get the ESP32-S2-SAOLA-1R to work with esphome
After a lot of trouble I got esphome to compile using

`esp32:
  board: esp32-s2-saola-1
  variant: ESP32S2
  framework:
    type: esp-idf
    version: recommended`
 but I end up with the same error message as misery does when using the flasher tool. 

Compiling using Arduino currently fails like this:

> INFO Reading configuration /config/esphome/esp32-s2-ard.yaml...
> INFO Generating C++ source...
> INFO Compiling app...
> Processing esp32-s2-ard (board: esp32-s2-saola-1; framework: arduino; platform: 
> https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream)
> --------------------------------------------------------------------------------
> * Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
> * Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
> * Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
> * Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
> * Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
> * Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
> HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
>  - framework-arduinoespressif32 0.0.0+sha.d218e58 
>  - toolchain-riscv32-esp 8.4.0+2021r2 
>  - toolchain-xtensa-esp32s2 8.4.0+2021r2
> Dependency Graph
> |-- <AsyncTCP-esphome> 1.2.2
> |-- <WiFi> 2.0.0
> |-- <FS> 2.0.0
> |-- <Update> 2.0.0
> |-- <ESPAsyncWebServer-esphome> 2.1.0
> |   |-- <AsyncTCP-esphome> 1.2.2
> |-- <DNSServer> 2.0.0
> |-- <ESPmDNS> 2.0.0
> Compiling /data/esp32-s2-ard/.pioenvs/esp32-s2-ard/src/esphome/components/api/api_connection.cpp.o
> sh: 1: xtensa-esp32s2-elf-g++: not found
> Compiling /data/esp32-s2-ard/.pioenvs/esp32-s2-ard/src/esphome/components/api/api_frame_helper.cpp.o
> sh: 1: xtensa-esp32s2-elf-g++: not found
> Compiling /data/esp32-s2-ard/.pioenvs/esp32-s2-ard/src/esphome/components/api/api_pb2.cpp.o
> sh: 1: xtensa-esp32s2-elf-g++: not found
> *** [/data/esp32-s2-ard/.pioenvs/esp32-s2-ard/src/esphome/components/api/api_connection.cpp.o] Error 127
> Compiling /data/esp32-s2-ard/.pioenvs/esp32-s2-ard/src/esphome/components/api/api_pb2_service.cpp.o
> *** [/data/esp32-s2-ard/.pioenvs/esp32-s2-ard/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
> *** [/data/esp32-s2-ard/.pioenvs/esp32-s2-ard/src/esphome/components/api/api_pb2.cpp.o] Error 127
> sh: 1: xtensa-esp32s2-elf-g++: not found
> *** [/data/esp32-s2-ard/.pioenvs/esp32-s2-ard/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
> ========================== [FAILED] Took 3.69 seconds ==========================

MFlasskamp, 2022-01-26:
@misery ESP32-S2 + ESP32-IDF + ADC was fixed by @martgras with https://github.com/esphome/esphome/pull/3117

Simple example:
```yaml
esphome:
  name: esp32s2-01

esp32:
  board: esp32-s2-saola-1
  variant: esp32s2
  framework:
    type: esp-idf

sensor:
  - platform: adc
    pin: GPIO3
    name: "esp32s2 adc"
```

RAYs3T, 2022-02-02:
No matter what this still dosen't work for the Olimex ESP32-S2 as we still get the bootloader error mentioned above (https://github.com/esphome/feature-requests/issues/996#issuecomment-964019142) .

MFlasskamp, 2022-02-02:
I think I had similar problems with an Ai-Thinker ESP-12K module and fixed that by flashing the factory image build by esphome to the module at address `0x0000` with esptool:

```
.\esptool.exe --port COM3 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x0000 <<PATH-TO-ESPHOME>>\.esphome\build\<<DEV-NAME>>\.pioenvs\<<DEV-NAME>>\firmware-factory.bin
```

RAYs3T, 2022-02-03:
## Work-A-Round for ESP32-S2 chips
Awesome, this actually worked

For everyone else having these bootloader issues with the ESP32-S2

### 1. Go to your ESPHome Dashboard 
### 2. Choose the option `Plug into this computer`
[image]
### 3. Choose "Download Project" - This will give you a `<nodename>-factory.bin` file.
[image]
[image]
### 4. install [ESPTool](https://docs.espressif.com/projects/esptool/en/latest/esp32/) and flash the device using this command (thanks @MFlasskamp )
```shell
esptool.py --port COM3 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x0000 downloaded_firmware-nodename-factory.bin
```
Replace COM3 with something like `/dev/ttyUSB0` if your on Linux or the correct COM-Port on windows.

**After this you can update the firmware of the ESP32-S2 via the WebUpdater**

## What's causing the issue?
My assumption is that the ESPHomeFlasher utility is starting to write from the wrong storage address, as it is not writing to `0x0000`

RAYs3T, 2022-02-03:
@flpaoli This is unfortunately not quickly solvable. It seems like the IDF framework has way less implementations then the Arduino framework

ashald, 2022-03-14:
> esp-idf does indeed allow running this on a Feather S2 for me (arduino worked with a lot of hacks). Even OTA upgrade works.
> 
> I had to patch the flash pin check out of esphome since the Stemma QT port uses pins 8+9.
> 
> Relevant sections:
> 
> ```
> esphome:
>   name: $devicename
>   platformio_options:
>     board_build.variant: um_feathers2
> 
> esp32:
>   board: featheresp32-s2
>   variant: esp32s2
>   framework:
>     type: esp-idf
>     version: recommended
> i2c:
>   - id: stemma
>     sda: 8
>     scl: 9
>     scan: true
> ```

@Sebmaster does it still work for you? Could you share instructions on how to flash?
I've got Feather with S2, and it won't show up as a device until I do that dance with hold-boot->press-reset->release boot. And after I flash it, I can never see logs or whatever.

Tried config like yours:
```
esphome:
  name: s2

  platformio_options:
    board_build.variant: um_feathers2

esp32:
  board: featheresp32-s2
  variant: esp32s2
  framework:
    type: esp-idf
    version: recommended

logger:

sensor:
  - platform: uptime
    update_interval: 10s
    id: up
```

Sebmaster, 2022-03-14:
@ashald Yeah you have to do the button dance if you want to flash over USB, but thankfully with esp-idf OTA is supported, so should only be necessary once.

It then requires that you reset manually once to actually start running (which will kill your logs), but you then should be able to do `esptool logs` and that should work (or if you do OTA that should also work automatically).

(all from memory, my setup's stable rn so I'm not gonna mess with it to retest)

ashald, 2022-03-14:
@Sebmaster Thanks for a quick reply! I guess I have to try adding Wi-Fi... I want to just see it work, but the issue is that once I flash it and reconnect (wasn't sure if just pressing reset would be sufficient), it never shows up under `/dev` again. :( It only quickly blinks orange near the connector and I'm not sure if that's a sign things OK or not.

Sebmaster, 2022-03-14:
@ashald Ah yeah, I always use wifi, I think it may be required (afaik feather s2 is missing the regular serial port IO hookup that logs would flow over).

ashald, 2022-03-14:
@Sebmaster oooh, that would explain. Will try, thanks! And the last question - did you try to make it work with Arduino core?

Sebmaster, 2022-03-14:
@ashald I think I had it running in some weird hacky config at one point with a specific back of versions (no working USB logs nor OTA either), but as soon as esp-idf came out it just worked so much better.

vrutkovs, 2022-06-14:
OTA seems to work on ESP-12K with
```
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
    build_unflags:
      - -DARDUINO_ESP32_DEV
      - -DARDUINO_VARIANT="esp32"
    build_flags:
      - -DARDUINO_SERIAL_PORT=1
      - -DCORE_DEBUG_LEVEL=5
      - -DBOARD_HAS_PSRAM
```

shahms, 2022-06-25:
I managed to get an Unexpected Maker FeatherS2 working with the configuration:
```
esphome:
  name: ${device-name}

esp32:
  board: um_feathers2
  variant: esp32s2
  framework:
    type: esp-idf
    platform_version: "4.3.0"
```
Notably, platform_version: "4.4.0" fails with `Error: Unknown development platform 'espressif32'`

Omitting `variant` fails with:
```
Traceback (most recent call last):
  File "/home/shahms/.local/bin/esphome", line 8, in <module>
    sys.exit(main())
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/__main__.py", line 931, in main
    return run_esphome(sys.argv)
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/__main__.py", line 909, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/config.py", line 972, in read_config
    res = load_config(command_line_substitutions)
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/config.py", line 827, in load_config
    return _load_config(command_line_substitutions)
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/config.py", line 815, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/config.py", line 742, in validate_config
    result.run_validation_steps()
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/config.py", line 124, in run_validation_steps
    task.step.run(self)
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/config.py", line 498, in run
    validated = schema(self.conf)
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/home/shahms/.local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/home/shahms/.local/lib/python3.10/site-packages/voluptuous/validators.py", line 229, in _run
    return self._exec(self._compiled, value, path)
  File "/home/shahms/.local/lib/python3.10/site-packages/voluptuous/validators.py", line 353, in _exec
    v = func(path, v)
  File "/home/shahms/.local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/home/shahms/.local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/home/shahms/.local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/voluptuous_schema.py", line 118, in validate_mapping
    not isinstance(key.default, vol.Undefined)
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/config_validation.py", line 1450, in default
    if CORE.is_esp8266:
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/core/__init__.py", line 590, in is_esp8266
    return self.target_platform == "esp8266"
  File "/home/shahms/.local/lib/python3.10/site-packages/esphome/core/__init__.py", line 586, in target_platform
    return self.data[KEY_CORE][KEY_TARGET_PLATFORM]
KeyError: 'target_platform'
```

nagyrobi, 2022-07-01:
https://github.com/esphome/esphome/pull/1433

codersaur, 2022-07-09:
Using the **dev branch** of esphome, I have the UM Feather**S3** working with the following configuration (works for arduino framework but not esp-idf):

        esphome:
          name: $device_name
        
        esp32:
          board: um_feathers3
          variant: esp32s3
          framework:
            type: arduino
            version: 2.0.3
            platform_version: 5.0.0

roger-, 2022-07-31:
Anyone able to get BLE support working on the Wemos S2 mini with the Arduino framework?

I'm using 

```yaml
esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: arduino
    version: 2.0.4
    platform_version: 5.0.0
```

But compilation fails:

```
Compiling .pioenvs/bridge/src/esphome/components/fan/fan.cpp.o
In file included from src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.h:5,
                 from src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:3:
src/esphome/components/esp32_ble_tracker/queue.h:11:10: fatal error: esp_gap_ble_api.h: No such file or directory

*************************************************************************
* Looking for esp_gap_ble_api.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_gap_ble_api.h"
* Web  > https://registry.platformio.org/search?q=header:esp_gap_ble_api.h
*
*************************************************************************

 #include <esp_gap_ble_api.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/bridge/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp.o] Error 1
```

shiSHARK, 2022-08-12:
Compilation error on Raspberry pi and on Mac M1-Pro when trying to build for Wemos Lolin S2 mini

```
esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: esp-idf
    platform_version: https://github.com/jhamhader/platform-espressif32#8bc085573e4f6986ffcda7050b9607739f3c6976
```

error message:
```
INFO Reading configuration /config/esphome/s2-mini.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing s2-mini (board: lolin_s2_mini; framework: espidf; platform: https://github.com/jhamhader/platform-espressif32#8bc085573e4f6986ffcda7050b9607739f3c6976)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-esp32s2ulp @ ~1.22851.0
Error: Could not find the package with 'platformio/toolchain-esp32s2ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'
```

lucky79, 2022-08-13:
Same problem on Odroid N2+. I think it only works on a Pi with 32bit Hass.

Bart-1992, 2022-09-18:
I'm trying to upload a succesfully compiled program to my ESP32S2, but i'm running into the following error from ESPtool: 

[image]

The configuration : 
[image]

Something goes wrong with the .bin files, i've manually added this "patched_bootloader.bin", just by copying a normal ESP32 bootloader and renaming it into the specific folder, but without results unfortunately. 

I've also tried to upload the bin files manually using : 

`1. esptool -p COM<PORT> write_flash 0x0 esp32_bootloader_v4.bin
2. esptool -p COM<PORT> write_flash 0x010000 firmware.bin`
I've also tried the factory-firmware variant at 0x0, and 0x01 but all without result. 

Does anyone know how I should upload the firmware files to the board? 

The following files are created by the compiler : 

[image]

ssieb, 2022-09-18:
You've mixed up too many things in that config to expect it to work.
This config compiles for me:
```yaml
esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```
This issue has way too much stuff in it.  It should be closed and any specific items put in separate issues.

bmustata, 2022-09-21:
Tks, this version worked for me. Also web_server is working.

mc: esp32s2
board: esp32-s2-carbon-s2

alfwro13, 2022-09-30:
> You've mixed up too many things in that config to expect it to work. This config compiles for me:
>
> ```yaml
> esp32:
>   board: esp32-s2-saola-1
>   framework:
>     type: arduino
>     version: 2.0.3 [...]

yey - that worked for me. However looking at the esphome log files I can see it complaining:
```
INFO Reading configuration /config/esphome/test.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Starting log output from 111.222.333.444 using esphome API
INFO Successfully connected to 111.222.333.444
[14:58:04][I][app:102]: ESPHome version 2022.8.0 compiled on Sep 30 2022, 14:56:22
```

gretel, 2022-11-18:
@alfwro13 this also werks for these ~2.50 euro cheapos:
[image]. thanks a lot!

jesserockz, 2022-11-21:
Closed in esphome/esphome#3564

closed 2022-11-21 by jesserockz