esphome/feature-requests #997: "Soft" strobe

Joshfindit, 2020-11-16:
**Describe the problem you have/What new integration you would like**

Strobe is great, but I would like something where it fades between the values:

```
      - soft-strobe:
          name: Soft Strobe 60bpm Heartbeat Effect
          colors:
            - state: True
              brightness: 100%
              red: 100%
              green: 100%
              blue: 100%
              white: 100%
              duration: 50ms
            - state: False
              duration: 50ms
            - state: True
              brightness: 100%
              red: 100%
              green: 100%
              blue: 100%
              white: 100%
              duration: 700ms
            - state: False
              duration: 200ms
```

**Please describe your use case for this integration and alternatives you've tried:**

Have not tried Lambdas as I have not been able to get them to work yet

**Additional context**

lhartmann, 2024-04-21:
Adding duration to strobe would be awesome. I was about to ask for that and found this post, so bumping it...

closed 2024-04-28 by jesserockz

Joshfindit, 2024-04-29:
📈🎉🥳