esphome/feature-requests #998: Add Observability to slow_pwm

michalsteyn, 2020-11-17:
**Describe the problem you have/What new integration you would like**
The internal values of the slow_pwm is currently hidden. I would like to know what the float value of the slow_pwm is, and whether the output is on or off. 

**Please describe your use case for this integration and alternatives you've tried:**
I converted a bang bang climate controller to a pid climate controller. I could observe the state of the climate controller in MQTT and calculate energy usage for a heater. 
I would like to know when the heater is actually switched on or off when I use a slow_pwm output. It would also be nice to know the actual pwm value.

**Additional context**
Here is an extract of my config:
```
output:
  - platform: slow_pwm
    pin: GPIO12
    id: heater_pwm
    period: 10min

sensor:
  - platform: dht
    pin: GPIO14    
    model: SI7021    
    temperature:
      name: "Shed Temp"
      id: temp_shed

climate:
  - platform: pid
    name: Shed Climate
    id: bang_bang_shed_climate
    visual:
      min_temperature: 10 °C
      max_temperature: 30 °C
      temperature_step: 0.1 °C
    sensor: temp_shed
    default_target_temperature: 16 °C    
    control_parameters:
      kp: 1.09135
      ki: 0.00067
      kd: 446.08807
    heat_output: heater_pwm   
<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE FEATURE REQUEST WILL BE CLOSED AUTOMATICALLY. -->

ssieb, 2021-12-13:
You could use a template output in between to capture the value.

nagyrobi, 2022-03-24:
can be done with new slow_pwm which is not tied to output