esphome/feature-requests #1000: HDC1080 heater

N3bukadnezar, 2020-11-18:
The webpage of the [HDC1080 ](https://esphome.io/components/sensor/hdc1080.html)states:
> Currently, the platform doesn’t support activating the built-in heater, as it seems to only be rarely of use. If you need it, please open an issue.

It would be nice if one could use the heater capabilities of the HDC1080. If the sensor is used in high humidity environments condensation can form, which is detected by the sensor, and the reading drops to 0% humidity.

One could maybe add that after X measurements/time the sensor heats up for Y min or manual triggering via a button/mqtt signal.

travnick, 2022-11-29:
I'm facing a long exposure humidity offset issue on my two setups (I have now about +10% humidity readings compared to two other 3rd party devices that show a humidity)

It would be nice to somehow configure a temperature/humidity range of enabling/disabling heater. If such "automation" is possible by other esp configuration features, then simple on/on is enough.

marcoboot, 2023-03-10:
I have this sensor in my weather station, and I get frequently 100% humidity. I'm curious what the difference will be with the heater manually (or automatically) switched on.

deZeus89, 2024-02-18:
Like it is already done in the SHT4x integration, it would be nice if the heater capabilities of the HDC1080 were also implemented in esphome. In high humidity environments, the "creep" rises and the humidity values are getting an offset...