esphome/feature-requests #1001: Consider adding support for the VL53L1X Time Of Flight Distance Sensor up to 400 cm

frederickjh, 2020-11-18:
**Describe the problem you have/What new integration you would like**

I would like to see the VL53L0X time of flight distance sensor supported by ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**

The currently supported time of flight sensor, VL53L0X, has a maximum distance of 200 cm which would be too short for my application of using it to sense the level of water in the rain water tanks I am considering purchasing. The VL53L1X has a maximum distance of 400 cm which would be enough to sense to the bottom of any rain water tank I am considering purchasing.

**Additional context**

The current time of flight sensor in ESPHome, the VL53L0X, uses the [pololu/vl53l0x-arduino](https://github.com/pololu/vl53l0x-arduino) library. [Pololu](https://www.pololu.com) also has a library for the VL53L1X, [pololu/vl53l1x-arduin](https://github.com/pololu/vl53l1x-arduino) that could possibly be used for the VL53L1X ToF sensor.

Polo does state on their website:
> For for similar but shorter-range sensors, see our 200 cm VL53L0X carrier and 60 cm VL6180X carrier. Both of these are physical drop-in replacements for the VL53L1X carrier, **but they have different APIs**, so software for the VL53L1X will need to be rewritten to work with the VL53L0X or VL6180X.

So we know that the VL53L0X and VL53L1X ToF sensors use different APIs.

Thanks in advance for your consideration of this feature request!
Frederick

frederickjh, 2020-11-18:
I also found that Pololu also has a [ST VL53L1X API](https://www.st.com/content/st_com/en/products/embedded-software/proximity-sensors-software/stsw-img007.html) implementation for Arduino, [pololu/vl53l1x-st-api-arduino](https://github.com/pololu/vl53l1x-st-api-arduino). This has all the functionality of the ST API. 

For use of the VL53L1X as a rain water tank level sensor this also provides the ability to calibrating the sensor to work well under a cover glass, which would be desirable so that the sensor could be placed in a water tight container but still function well.

The [pololu/vl53l1x-arduino](https://github.com/pololu/vl53l1x-arduino) library does not have this functionality nor the functionality to select a smaller region of interest (ROI)), and it has less robust error checking.

rradar, 2021-01-25:
See https://github.com/esphome/feature-requests/issues/836

closed 2021-01-25 by rradar

frederickjh, 2021-01-26:
Sorry about opening a duplicate issue. Usually I check to see if an issue is open before creating a new on.