esphome/feature-requests #1004: using "on_time" from lamdba

blademckain, 2020-11-19:
**Describe the problem you have/What new integration you would like**
it would be useful to have the ability to dynamically define the on_time function

**Please describe your use case for this integration and alternatives you've tried:**
For example, if you want to insert alarms at specific times of the day (which can change over time), such as an alarm clock

**Additional context**

benquan, 2021-07-23:
Same Issue.

I would like to define a variable in Home Assistant, and then assign that variable to on_time parameters. That way you can define the schedule without compiling the microprocessor every time. At the moment on_time does not accept lambdas

nagyrobi, 2022-07-07:
For an alarm clock project example, this may be interesting: https://github.com/mmakaay/esphome-alarm-clock
@mmakaay

mmakaay, 2022-07-08:
Dynamic would be a bad idea, since the code would have to execute the lambda function to see if anything changed. This is a waste of CPU time.
A better implementation (something that was lacking for me in the mentioned alarm clock project) was an alarm-like component, for which the alarm time can be configured using an action. Being able to have multiple of these components (for multiple alarm) would be a nice one too. 
When the alarm clock is finished (currently working on the last bit: the actual audio for waking up), I plan to work on an alarm component like this.

soeffi, 2023-06-10:
@mmakaay, did you look into that in the meantime? Would be a great feature to have.

> Dynamic would be a bad idea, since the code would have to execute the lambda function to see if anything changed. This is a waste of CPU time.

I think #2100 is requesting the exact same thing. In that thread, there was a proposal to link the cron config to an entity instead of a lambda. Could be number entities for hour/minute/second, or better the new time entity when #2267 is implemented.

I'm not familiar with ESPhome internals, but maybe that allows to update the schedule based on that entity - whenever the entity state changes?