esphome/feature-requests #1005: Add support for si4730/5 boards

SkalkaA, 2020-11-19:
**Describe the problem you have/What new integration you would like**
Support for connecting a radio module (si4735)

**Please describe your use case for this integration and alternatives you've tried:**
There are some projects online of people connecting this board to an ESP/Arduino, controlling the signal and playing audio through a speaker. It would be nice to integrate something similar into esphome, to make these radios even smarter.
Being able to control them through HA etc. would make the possibilities limitless.

**Additional context**
Here is a link to a library for the module
https://github.com/pu2clr/SI4735
and some radios
https://create.arduino.cc/projecthub/mircemk/homemade-arduino-si4730-all-band-receiver-lw-mw-sw-fm-cbfdd6
https://www.youtube.com/watch?v=ekrlcRn1qnE&ab_channel=RicardoCaratti

nagyrobi, 2022-07-08:
Also RDA5807M...

nima-1102, 2024-10-21:
+1 Si4703 module

lilmansplace, 2025-04-07:
+1  Si4703 module