esphome/feature-requests #1009: MAX30102 - Pulse Oximeter and Heart-Rate Sensor

pmoya-in-the-web, 2020-11-20:
**Describe the problem you have/What new integration you would like**

I would like to configure a MAX30102 - Pulse Oximeter and Heart-Rate Sensor with ESPHome

**Please describe your use case for this integration and alternatives you've tried:**

Support of this sensor directly from ESPHome

**Additional context**

ivlis, 2021-01-07:
Here is the source code for Arduino: https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library

I wonder if this can be integrated with ESPHome.

lerra, 2021-01-21:
I would love to be able to capture such sensordata with esphome!

Ranchuking, 2021-08-15:
I would like to know how to intergrate in ESPHome too.

dpalmigiano, 2021-11-15:
I work for a man with spinal muscular atrophy and we're trying to expand home assistant to have the most important data. This is one metric I would really like to implement.

ivlis, 2021-11-16:
I would really NOT rely on this sensor for anything other than "for fun". It is very inaccurate and works only if located precisely on the fingertip. This is NOT something you find in a smartwatch keep aside professional heart monitors.

randybb, 2021-11-16:
Just buy a pulse oximeter that is streaming values through BLE.

dpalmigiano, 2021-11-22:
Oh, thank you. We have this, right here: https://www.amazon.com/gp/product/B087Q724QM/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1, and I was able to get the values displayed to a terminal. The rest of the details would be off topic, but thanks for steering me away from this. I actually had no idea.

I would like additional help tho, so if anyone knows a lot, could you send me an email? 

pulseox@donpalm.dev 

Subject: June881 {the rest of your subject}

Ranchuking, 2023-10-10:
> are there any achievements to this I would also be interested to have this sensor in esphome

Not yet.

Extreme79, 2024-03-17:
I'm searching the exactly same thing.
Instead of writing custome code, would be great to have an esphome support!