esphome/feature-requests #1010: SSD1283A 130x130 1.6" display

idavydov, 2020-11-21:
**Describe the problem you have/What new integration you would like**

No support for SSD1283A display. This is a 1.6" 130x130 transflective TFT display with SPI interface.

**Please describe your use case for this integration and alternatives you've tried:**

I suppose this display has good visibility in sunlight.

**Additional context**

There's some documentation available at [lcdwiki](http://www.lcdwiki.com/1.6inch_SPI_Module_SSD1283A_SKU:MSP1601). It seems to work well with [this](https://github.com/ZinggJM/SSD1283A) library.

I wonder if there is something similar available already. If there is, I could try to update it to work with SSD1283A.

rradar, 2021-05-16:
Maybe relevant: https://github.com/ZinggJM/SSD1283A

MikelMirror, 2023-09-14:
Hi, news about this?