esphome/feature-requests #1012: Integrate BLE tracking with Room Assistant Companion (iOS)

eximo84, 2020-11-23:
**Describe the problem you have/What new integration you would like**

The room assistant project tracks has developed an iOS companion app which allows a static Bluetooth UUID. As such it’s possible to track iPhone devices with this using BLE. 

However Esphome doesn’t appear to work with this as I cannot get my esp32 to find the advertised GAT UUID code. 

**Please describe your use case for this integration and alternatives you've tried:**

Useful for room tracking of devices into home assistant without having to run additional hardware for room assistant. 

**Additional context**

ronytomen, 2020-12-17:
Very interesting and cool work around that Room Assistant team has done!

alexdudgeon, 2021-03-15:
The UUID tracker also does not work with the Android companion version. The UUID format from the app seems to be different to what esphome is expecting. Is this a Home Assistant companion issue, or esphome issue?

mKeRix, 2021-03-15:
The support needs to be implemented in ESPHome - the HA Android app simply broadcasts iBeacons, which can be detected already. On iOS you can't do that in the background however, which is why ESPHome would need to support an additional discovery flow (which would then yield the UUID shown in the companion app). You can take a look at the room-assistant code for the relevant bits, or I can also give a high level overview of the process if you want.

alexdudgeon, 2021-03-15:
I can't detect the beacons from Companion app in ESPhome, even when setting the UUID to the same as app.
My next step was to look at Room-assistant, but I had a few already conveniently-located ESP32s, so would be ideal if I can get it working

mKeRix, 2021-03-15:
Ah, I'm sorry - I misread your comment. I think to track the Android app beacons via ESPHome you'd need to use the `service_uuid` option of the `ble_presence` platform. Note that this doesn't seem to include major or minor (so if the Android app broadcasts `xxxxyyyy-1-1` you'd only want to configure `xxxxyyyy` in ESPHome). At least that's my interpretation of the docs, as I don't actually use this feature of ESPHome.

alexd321, 2021-03-15:
Yeah I have tried that, but no luck. I will log a separate bug

franfos, 2021-06-11:
Hi @alexd321 ,
did you finally get ESPHome detect the beacons from companion app?
I'm trying to do the same with no luck.

alexd321, 2021-06-11:
no luck 😢

cryptedx, 2022-04-22:
Now I know why it's not working. I also tried the room assistant app with my iPhone.