esphome/feature-requests #1013: Add waveshare "2.9inch e-Paper HAT (D)"

randompee, 2020-11-23:
waveshare 2.9inch e-Paper HAT (D) tested compatible to the cheap MH-ET Live 2.9 inch E-paper Module (e.g. https://de.aliexpress.com/item/4001279382105.html?spm=a2g0s.9042311.0.0.68c34c4djtXUBM, < €16).  For tests I used the Arduino library you can find here: https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board
**Describe the problem you have/What new integration you would like**

The 2.9 inch waveshare displays available in ESPhome did not work for the MH-ET module, but I would very much like to use it in ESPhome (CO2 sensor with local display).
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nuttytree, 2021-02-10:
@randompee, I think I can probably add this for you although I will need you to test it as I don't have one of these.  Can you tell me what display type you needed to use on the WaveShare board (A or B)?

nuttytree, 2021-02-15:
@randompee I have a PR up for this now, can you test and let me know how it goes.

velaar, 2021-02-15:
@nuttytree It looks like startup commands are similar to 1.54 v2 https://github.com/esphome/issues/issues/1032 can it also be added? I can test it as needed.

randompee, 2021-02-25:
@nuttytree Unfortunately, it does not work for me:
[16:44:03][E][waveshare_epaper:107]: Timeout while displaying image!

Board: az-delivery-devkit-v4, but I also tried esp32dev without the specific platformio_options. I used the VSPI bus.

nuttytree, 2021-03-02:
@randompee this change has not been merged yet so it would not be in the 1.17.0.dev0 branch, can you pull my branch and give it a try?

randompee, 2021-03-02:
@nuttytree Actually, I use "the latest bleeding edge version", and that gives me "Version: 1.17.0-dev". When I look into waveshare_epaper.cpp, it already has your changes merged, I noticed this at once because I always have to change "// COMMAND DISPLAY UPDATE CONTROL 2" for my TTGO T5_V2.3_2.13!
I have to say, though, that This time around I also had no success when I tried the board with several sketches, using the Arduino IDE. Maybe I killed it, or did not set the SPI pins at the right place for these Arduino IDE tests  :-(
By the way, K-r-i-s-t-i-a-n Added Waveshare 2.90inch V2 e-ink display 13 days ago (https://github.com/esphome/esphome/commit/c92c439d17cb71789d7a342246459e7917f1d932).

randompee, 2021-03-07:
@nuttytree So sorry! I found that I had a wiring problem. That taken care of I still had no luck with your model in ESPHome.

When I went back to the GxEPD2 library examples, though, this time I persisted until I had succes. The compatible GxEPD2 model for my MH-ET Live display is "**GxEPD2_290_T5**". Unfortunately, nobody seems to have come up with a collection of the different names for the same e-paper modules (or compatibles). Thus, frequent mix ups and misunderstandings...

boyfromgermany, 2024-01-14:
Hello,

i have the same Board, have you managed to get it working ?

GraffJosh, 2024-01-22:
Same here. I have a flexible 2.9, which I believe is the 2.9in-d module. 

Very similar environment, trying to set up the ESP32 Driver Board from Waveshare. 

The error I'm getting is "Timeout while displaying image!"

closed 2025-02-11 by clydebarrow