esphome/feature-requests #1014: Add support for PCA9536 I/O module

Tuckie, 2020-11-24:
**Describe the problem you have/What new integration you would like**

I would like to add support for the PCA9536. 

**Please describe your use case for this integration and alternatives you've tried:**

This would allow me to add this module: https://store.ncd.io/product/4-channel-i2c-pca9536-optical-isolated-digital-input-with-i2c-interface/ to my setup. 

**Additional context**

This should be relatively similar to the existing PCF8574

nagyrobi, 2023-03-24:
PCA9536 and PCA9554 are compatible with each other?

randybb, 2023-03-25:
Yes, there are thee expanders, only difference is their address and number of GPIOs.
PCA9536 has a fixed address 0x41 and only 4 GPIOs, and then we have 8 GPIO (+ IRQ) versions PCA9554 with configurable address 0x20-0x27 and PCA9554A is the same, but with address range 0x38-0x3f.

nagyrobi, 2023-03-25:
Thanks, indeed: https://esphome.io/components/pca9554.html

closed 2023-03-25 by nagyrobi