esphome/feature-requests #1026: VPN for esp devices

christophermodom, 2020-11-30:
**Describe the problem you have/What new integration you would like**

I have a desire to put a device on another network, about 50 miles away, to monitor conditions on a boat. After using, with great success, ZeroTier in Home Assistant it dawned on me how great it would be to have ZeroTier directly loaded onto an esp so I could connect to the device remotely and add it to the HA front end.   I know putting devices on other networks has security risks, however ZeroTier resolves this.  This addition to esphome may open up a world of possibilities. I’m no expert and this may be impossible or maybe there is a solution already, but I thought this would be a great feature.

**Please describe your use case for this integration and alternatives you've tried:**

ZeroTier integration to remotely access devices on other networks.  I have not tried or know of alternatives.

**Additional context**

lwqcz, 2020-12-16:
ESP chips now-a-days lacks enough compute power and other HW resources needed for such a task as VPN is. Checkout DietPi linux for Raspberry Pi, as it has VPNs build in, easily useable.

ZsBT, 2021-03-03:
A Mikrotik router is quite cheap which can arrange the vpn for the device.

dougle03, 2021-08-19:
> A Mikrotik router is quite cheap which can arrange the vpn for the device.

Have you managed to get a router to act as a ZeroTier bridge? I've tried allsorts, I can get node to node working well, but bridging two networks via ZeroTier so far eludes me.

ZsBT, 2021-08-19:
Had no experience with ZeroTier yet. Just an OpenVPN server on my RPI4 Hass
and a Microtik acting as a VPN client.

Peter Kay ***@***.***> ezt írta (időpont: 2021. aug. 19., Csü
15:24):

davey, 2021-10-01:
Similar request for VPN (with WireGuard) in #1444, but this one might be easier to implement, as there's already an Arduino library with a supposedly working example.

nagyrobi, 2022-07-01:
API encryption is supported out of the box: https://esphome.io/components/api.html but you need some kind of static IP on the remote side.

As previous posters mentioned, it's better to install a dedicated VPN device at the remote side to handle the traffic, this brings benefit of using multiple devices via the same VPN channel.

PablexXXXX, 2023-05-16:
I also want to connect some of my nodes from outside the network where my HA instance resides. I think VPN is not necesary, just a publicly accesible instance, Of course the API is designed to work with the HA server on the same network. That is what needs to be changed. We need the ability to specify a URL of the HA server in the YAML.

nagyrobi, 2023-05-16:
Guys, you should read the docs. It works the other way around - ESPHome nodes act as 'servers' and Home Assistant connects to them.

HeMan, 2023-10-03:
Is the Wireguard component (https://esphome.io/components/wireguard.html), that was released in 2023.9, working for you?

closed 2024-06-07 by nagyrobi