esphome/feature-requests #1029: RC522 with I2C support

sj3fk3, 2020-12-02:
**Describe the problem you have/What new integration you would like**
In order to get the wonderful M5Stack with RFID Face to work, the already finished RC522 integration (#1298) also needs I2C support. Analoge to what is been done in #1302 for the Pn532.

https://m5stack.com/collections/m5-module/products/rfid-rc522-panel-for-m5-faces

**Please describe your use case for this integration and alternatives you've tried:**
The alternative would be to compile in a custom component

**Additional context**
I'm perfectly willing to donate, test and help out in any other way, but can't do much more than print("hello world")

docteurzoidberg, 2020-12-02:
it's already implemented in dev

sj3fk3, 2020-12-02:
AFAIK only SPI is in dev, not i2c, but if so would you please point me to it? Because I only see SPI.

tamerlan311, 2020-12-03:
https://next.esphome.io/components/binary_sensor/pn532.html

# Example configuration for I²C (choose which one!)
pn532_i2c:
  update_interval: 1s

sj3fk3, 2020-12-03:
The example is for **pn532**, my feature request is for the **RC522**. 

Was my description _that_ unclear?

glmnet, 2020-12-21:
I can try to get it work with my rc522 board, it's supposed to have i2c interface too, I don't have that m5 thing though

sj3fk3, 2020-12-21:
The M5 is just a ESP32 in the end and I think all RC522 do SPI and ic2 by design? Would be great if you could, if you want I could ship you a set.. That m5stack is really great stuff (I'm starting to because a big fan).

glmnet, 2020-12-23:
[image]

Having lots of fun with this request... turns out the trace you have to remove to run this module in i2c is... under the chip. So had to remove it cut the trace and put it back. For added fun I just added flying wires so I can setup either mode. Did not work at first time, had to reposition the thing a few times. I believed I baked it too much but seems to work in SPI. NOW TO CODE

sj3fk3, 2020-12-23:
OMG! So cool!

closed 2021-01-12 by glmnet