esphome/feature-requests #1030: Folders for the dashboard

jbuszkie, 2020-12-03:
**Describe the problem you have/What new integration you would like**

As I add more and more nodes, the dashboard will get pretty cluttered.
I'd like to see the ability to have a folder system for the dashboard.  This way the nodes can be grouped together 
by a specific project or device type.
For example I would have, possibly, a ESP8266 folder and a ESP32 folder.
Or I have  a bunch of relay nodes to turn on and off my x-mas lights.  All those nodes could go into a "xmas lights" folder or something like that.

**Please describe your use case for this integration and alternatives you've tried:**

This is for when you have too many nodes.

**Additional context**

Someone else mentioned it here
https://github.com/esphome/feature-requests/issues/465#issuecomment-551167377
Not sure if another issue was created for this.

jbuszkie, 2020-12-03:
Looks like there was one opened alreaady but nothing came of it?
https://github.com/esphome/feature-requests/issues/867

Could this be merged?  (is there such a thing?)

glmnet, 2020-12-28:
Duplicate of #867

@jbuszkie indeed, it's called duplicate. thanks for pointing out

to support the FR use the 👍  on the original FR. Thanks

closed 2020-12-28 by glmnet