esphome/feature-requests #1031: Support for mcp4725 - 3.3v 12 bit i2c dac with EEPROM

fdepalo, 2020-12-04:
Hello, we are about to release a new i/o board that includes among all other chips already supported by Tasmota mcp4725 for dual or quad 0-10v output ( mainly for valves, dimmers, fan with 0-10v input )

It is possible to add support to tasmota? 

If dev requires so, we can ship him a board prototype for tests and development/integration.

Thank You

[MCP4725.pdf](https://github.com/esphome/feature-requests/files/5642608/MCP4725.pdf)

JJK801, 2020-12-15:
@fdepalo ☝️  Merry christmas 🎄

closed 2021-02-06 by glmnet

mlavelle42, 2021-04-11:
> > Hello, we are about to release a new i/o board that includes among all other chips already supported by Tasmota mcp4725 for dual or quad 0-10v output ( mainly for valves, dimmers, fan with 0-10v input )
> > It is possible to add support to tasmota?
> > If dev requires so, we can ship him a board prototype for tests and development/integration.
> > Thank You
> > [MCP4725.pdf](https://github.com/esphome/feature-requests/files/5642608/MCP4725.pdf)
> 
> Hi fdepalo, i can do it... but i would need some chips of them

Did anything come of this? I'm interested in a 0-10VDC output too.

glmnet, 2021-04-12:
Mcp4725 is available in dev and beta