esphome/feature-requests #1032: Integrate trackR pixel devices

axwell, 2020-12-05:
**Describe the problem you have/What new integration you would like**

I want to be able to use trackR pixe BLE beacon as a binary_sensor.
https://www.thetrackr.com/

**Please describe your use case for this integration and alternatives you've tried:**

Currently i have to press the pair button in order for esphome to see it, this works for a few minutes after that the beacon will disable because no connection was made, i supose some sort of pairing needs to be done, similar to the app in order for it to always broadcast.

**Additional context**

rpavlik, 2021-04-26:
I wonder if https://github.com/esphome/esphome/pull/1177 would satisfy it - it would let you have a connection.

rpavlik, 2022-07-07:
well I didn't actually try it, it was just a hypothesis as a basis. Would still need an integration...