esphome/feature-requests #1033: GUI change for the color blind people

TheGabeMan, 2020-12-06:
**Describe the problem you have/What new integration you would like**

In the GUI it seems that node status online / offline is shown as a small red or green bar. As I'm colour-blind, I'm unable to see the difference. It would be better if there was a clearer distinction between online / offline. Maybe write the word offline / online in the bar? Or somewhere show a different shaped icon? 

**Please describe your use case for this integration and alternatives you've tried:**

A lot of colour-blind people would love this.

**Additional context**

kylegordon, 2020-12-07:
Jeez, I never even noticed that the bar was different. I am *slightly* red/green colour blind, as in they're not as vivid as others see them. I thought the bar was just some pointless UI fluff. It's only when I really focus on it that I can start to tell that they're different colours.

randybb, 2020-12-08:
We already have a feature request https://github.com/esphome/feature-requests/issues/890 and an issue https://github.com/esphome/issues/issues/1504 for this problem.

closed 2020-12-08 by randybb