esphome/feature-requests #1034: Support for MCCGQ02HL - Xiaomi Mijia Window Door Sensor 2, broadcasts opening state, light state and battery level.

nesror, 2020-12-07:
**Describe the problem you have/What new integration you would like**. 
MCCGQ02HL

(Xiaomi Mijia Window Door Sensor 2, broadcasts opening state, light state and battery level. Advertisements are encrypted, therefore you need to set an encryption key in your configuration, see for instructions the encryption_key option. Battery level is only send once in approximately 24 hours.)

[image]

**Please describe your use case for this integration and alternatives you've tried:**
use https://github.com/custom-components/ble_monitor 

**Additional context**

Lymkin, 2020-12-14:
I 2nd this request.

demonkol, 2020-12-24:
+1 
It would be great if there was support.

keniji, 2020-12-26:
Well I fourth this... It seems Xiaomi is planning to replace Zigbee with BLE in the new generation sensors. The first is this Windows and Door sensor. The second one is the flood detector (SJWS01LM) the support is being discussed. The third one -- a motion detector (RTCGQ02LM) will be launched soon...

petermcandrew, 2020-12-26:
> Xiaomi is planning to replace Zigbee with BLE
Where did you see this?

keniji, 2020-12-28:
> > Xiaomi is planning to replace Zigbee with BLE
> 
> Where did you see this?

Well, there is no official announcement, but as we can see all the new launched sensors by Xiaomi -- including MCCGQ02HL, SJWS01LM, RTCGQ02LM and DHKG01ZM are all BLE-based, so...

kar-vi-nen, 2021-02-06:
https://github.com/esphome/feature-requests/issues/978

dentra, 2021-08-17:
You can take it from my repo [SJWS01LM](https://github.com/dentra/esphome-components/tree/master/components/miot_sjws01lm), [RTCGQ02LM](https://github.com/dentra/esphome-components/tree/master/components/miot_rtcgq02lm), [MCCGQ02HL](https://github.com/dentra/esphome-components/tree/master/components/miot_mccgq02hl) by using great `external_components` ESPHome feature.

It implementation is quite different from ESPHome Xiaomi platform, but developed by MI specs, more clear and without magic and now under heavy development to conform need of other devices. After a some time I will try to CR it in ESPHome if the authors will approve my solution.

`DHKG01ZM` can be easily added too but I do not have them. I can instruct you how to get necessary information.

gianlucaf81, 2021-11-27:
> You can take it from my repo [SJWS01LM](https://github.com/dentra/esphome-components/tree/master/components/miot_sjws01lm), [...]

Hi I'm trying to integrate a MCCGQ02HL in HA via ESP32, I have already successfully integrated the LYWSD03MMC temperature sensors on the same ESP32. HA seem not to be receiving data from the MCCGQ02HL sensor.
From the XiaomiHome app it works perfectly.
Do you have any advice for me?
Thanks.

dentra, 2021-11-29:
> Hi I'm trying to integrate a MCCGQ02HL in HA via ESP32, I have already successfully integrated the LYWSD03MMC temperature sensors on the same ESP32. HA seem not to be receiving data from the MCCGQ02HL sensor. From the XiaomiHome app it works perfectly. Do you have any advice for me? Thanks.
it very is hard to advice with no details

gianlucaf81, 2021-11-29:
> it very is hard to advice with no details

what information can I give you to help me? Thanks

dentra, 2021-11-29:
Configuration, logs

gianlucaf81, 2021-12-01:
> Configuration, logs

Wow, thanks for the tip. Indeed, the ESPHome log gave me the answer. It was the wrong bind_key, for some reason.

szilardcsotya, 2021-12-02:
> > Configuration, logs
> 
> Wow, thanks for the tip. Indeed, the ESPHome log gave me the answer. It was the wrong bind_key, for some reason.

Can you please provide what you did to finish this project? I'm very interesting on this! I have 9 temperature sensors fully integrated in HA via ESP32 and flashed with the custom firmware and I want to add these door and window sensors to HA as well. Thanks for your prompt reply.

gianlucaf81, 2021-12-03:
> > > Configuration, logs
> > 
> > 
> > Wow, thanks for the tip. Indeed, the ESPHome log gave me the answer. It was the wrong bind_key, for some reason.
> 
> Can you please provide how you did finish this project? I'm very interesting on this! I have 9 temperature sensors fully integrated in HA via ESP32 and flashed with the custom firmware and I want to add these door and window sensors to HA as well. Thanks for your prompt reply.

I just followed the guide [here](https://github.com/dentra/esphome-components/tree/master/components/miot_mccgq02hl). it's made very well, congratulations to the author. Copy and paste the "Sample configuration" code and modify it as you wish. I have removed "Battery Voltage" and renamed it in my language (Italian) and removing "$ name". Don't forget "external_components:" line otherwise it will report an error. Upload it via OTA and go. Simple and functional. 
I used Xiaomi Gateway 3 to obtain the device-mac-address and device-bin-key,I think it can also be obtained in another way but I don't know how to help you in this. If you need more details just ask.

szilardcsotya, 2021-12-03:
> I just followed the guide [here](https://github.com/dentra/esphome-components/tree/master/components/miot_mccgq02hl). it's made very well, congratulations to the author. Copy and paste the "Sample [...]

First step is missing for me, how can I get the bindkey?

gianlucaf81, 2021-12-03:
> First step is missing for me, how can I get the bindkey?

the easiest thing in my opinion is to buy a Xiaomi gateway 3 and integrate it into HA.  it seems counterintuitive but using esp32 in many parts of the house is cheaper than using multiple gateways.

szilardcsotya, 2021-12-03:
> the easiest thing in my opinion is to buy a Xiaomi gateway 3 and integrate it into HA. it seems counterintuitive but using esp32 in many parts of the house is cheaper than using multiple gateways.

Thanks for your reply. I have get the bindey, flashed a new esp32 board, this is the file I edited:

sensor:
binary_sensor:
  - platform: xiaomi_mccgq02hl
    mac_address: "the mac address"
    bindkey: "bindkey"
    name: "Konyha ablak Sensor"
    light:
      name: "Konyha ablak Light"
    alert:
      name: "Konyha ablak Alert"
    battery_level:
      name: "Konyha ablak Battery Level"
    battery_voltage:
      name: "Konyha ablak Battery Voltage"

After I clicked on the install button this is appeared:
INFO Reading configuration /config/esphome/ablaknyitas-teszt.yaml...
Failed config

binary_sensor.xiaomi_mccgq02hl: [source /config/esphome/ablaknyitas-teszt.yaml:30]
  
  Platform not found: 'binary_sensor.xiaomi_mccgq02hl'.

I am succes with the temperature sensors and tried to add the door-window sensor but i could not. What could be the problem?

gianlucaf81, 2021-12-04:
Don't forget "external_components:" line otherwise it will report an error.

szilardcsotya, 2021-12-04:
> Don't forget "external_components:" line otherwise it will report an error.

Ran at failure again.

external_components:
  - source: github://dentra/esphome-components
  
  - platform: miot_mccgq02hl
    mac_address: "MA"
    bindkey: "BK"
      name: "Konyha ablak Sensor"
    light:
      name: "Konyha ablak Light"
    alert:
      name: "Konyha ablak Alert"
    battery_level:
      name: "Konyha ablak Battery Level"
    battery_voltage:
      name: "Konyha ablak Battery Voltage"

ESP msg:
INFO Reading configuration /config/esphome/ablakszenzor.yaml...
ERROR Error while reading config: Invalid YAML syntax:

while parsing a block mapping
  in "/config/esphome/ablakszenzor.yaml", line 31, column 5:
      - platform: miot_mccgq02hl
        ^
expected <block end>, but found '<block mapping start>'
  in "/config/esphome/ablakszenzor.yaml", line 34, column 7:
          name: "Konyha ablak Sensor"
          ^

gianlucaf81, 2021-12-06:
> while parsing a block mapping in "/config/esphome/ablakszenzor.yaml", line 31, column 5: - platform: miot_mccgq02hl ^ expected , but found '' in "/config/esphome/ablakszenzor.yaml", line 34, column 7: name: "Konyha ablak Sensor" ^

` Invalid YAML syntax`

incorrect syntax, try to correct, you know that even a space on yaml can create an error.

dentra, 2021-12-06:
with my components you does not need to obtain bindkey. It can be done automatically. Just setup xiaomi account  in `miot` platform:
```yaml
miot:
  xiaomi_account:
    username: !secret xiaomi_username
    password: !secret xiaomi_password```

It requests xiaomi servers for device bindkey when compiling firmware, so you need an internet access from host where are you running esphome.

gianlucaf81, 2021-12-06:
> ```yaml
> xiaomi_account
> ```
Very interesting, how exactly does it work? The login bypass the need for the bindkey or retrieve it from the account?

dentra, 2021-12-06:
retrieve it from the account

dentra, 2021-12-06:
bindkey should not be changed until rebonding your device in mi home app.

szilardcsotya, 2021-12-06:
> bindkey should not be changed until rebonding your device in mi home app.

I believe removing the battery restarts the device so this is why the bindkey changed.

gianlucaf81, 2022-02-02:
Still asking for your help, I'm trying to load firmware for an update but while compiling I get this error. Can you tell me what it could be? Thank you

`HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/api/api_connection.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/api/api_server.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/api/list_entities.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/api/proto.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/api/user_services.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/api/util.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/esp32/core.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/esp32/gpio_idf.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/logger/logger.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/md5/md5.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/miot/automation.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/miot/miot.cpp.o
Compiling /data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/miot/miot_decrypt.cpp.o
In file included from src/esphome/components/miot/miot.cpp:1:0:
src/esphome/components/miot/miot.cpp: In member function 'bool esphome::miot::MiBeaconTracker::parse_mibeacon_(const esphome::esp32_ble_tracker::ESPBTDevice&, const std::vector<unsigned char>&) const':
src/esphome/components/miot/miot.cpp:54:76: error: 'format_hex_pretty' was not declared in this scope
     ESP_LOGW(TAG, "Invalid MiBeacon data length: %s", format_hex_pretty(raw).c_str());
                                                                            ^
src/esphome/core/log.h:117:88: note: in definition of macro 'esph_log_w'
   esp_log_printf_(ESPHOME_LOG_LEVEL_WARN, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^
src/esphome/components/miot/miot.cpp:54:5: note: in expansion of macro 'ESP_LOGW'
     ESP_LOGW(TAG, "Invalid MiBeacon data length: %s", format_hex_pretty(raw).c_str());
     ^
src/esphome/components/miot/miot.cpp:67:35: error: 'format_hex_pretty' was not declared in this scope
              format_hex_pretty(raw).c_str());
                                   ^
src/esphome/core/log.h:117:88: note: in definition of macro 'esph_log_w'
   esp_log_printf_(ESPHOME_LOG_LEVEL_WARN, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^
src/esphome/components/miot/miot.cpp:66:5: note: in expansion of macro 'ESP_LOGW'
     ESP_LOGW(TAG, "Device data is using old data format %" PRIu8 ": %s", mib.frame_control.version,
     ^
src/esphome/components/miot/miot.cpp:146:60: error: 'format_hex_pretty' was not declared in this scope
     ESP_LOGD(TAG, "Got MiBeacon: %s", format_hex_pretty(raw).c_str());
                                                            ^
src/esphome/core/log.h:95:89: note: in definition of macro 'esph_log_d'
   esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                         ^
src/esphome/components/miot/miot.cpp:146:5: note: in expansion of macro 'ESP_LOGD'
     ESP_LOGD(TAG, "Got MiBeacon: %s", format_hex_pretty(raw).c_str());
     ^
[... cut ...]
 #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
                            ^
src/esphome/components/miot/miot_decrypt.cpp:117:5: note: in expansion of macro 'ESP_LOGW'
     ESP_LOGW(TAG, "   key: %s", format_hex_pretty(key, sizeof(bindkey_t)).c_str());  // legacy bindkey size is 12 bytes
     ^
*** [/data/termometri-bt/.pioenvs/termometri-bt/src/esphome/components/miot/miot_decrypt.cpp.o] Error 1
========================== [FAILED] Took 8.93 seconds ==========================`

[image]

rolldo, 2022-06-04:
Can i use python file directly on windows to get data from sensor ?

brynjoewill, 2022-06-08:
Hi, I'm able to see a connection to the MCCGQ02HL with the RSSI status, the entities appear in home assistant, but do not update the state.  Any help would be most appreciate.  Here are the logs:
INFO Successfully connected to garden-bridge.local
[10:09:28][I][app:102]: ESPHome version 2022.5.1 compiled on Jun  8 2022, 10:04:11
[10:09:28][C][wifi:491]: WiFi:
[10:09:28][C][wifi:353]:   Local MAC: 
[10:09:28][C][wifi:354]:   SSID: [redacted]
[10:09:28][C][wifi:355]:   IP Address: 192.168.1.24
[10:09:28][C][wifi:357]:   BSSID: [redacted]
[10:09:28][C][wifi:358]:   Hostname: 'garden-bridge'
[10:09:28][C][wifi:360]:   Signal strength: -66 dB ▂▄▆█
[10:09:28][V][wifi:362]:   Priority: 0.0
[10:09:28][C][wifi:364]:   Channel: 13
[10:09:28][C][wifi:365]:   Subnet: 255.255.255.0
[10:09:28][C][wifi:366]:   Gateway: 192.168.1.1
[10:09:28][C][wifi:367]:   DNS1: 192.168.1.1
[10:09:28][C][wifi:368]:   DNS2: 0.0.0.0
[10:09:28][C][logger:274]: Logger:
[10:09:28][C][logger:275]:   Level: VERBOSE
[10:09:28][C][logger:276]:   Log Baud Rate: 115200
[10:09:28][C][miot_mccgq02hl:014]:   Alert 'Shed_Open'
[10:09:28][C][miot_mccgq02hl:014]:     Device Class: 'problem'
[10:09:28][V][api.connection:809]: Hello from client: 'Home Assistant 2022.6.4 (::FFFF:C0A8:108)'
[10:09:29][C][ble_rssi:011]: BLE RSSI Sensor '$shed RSSI'
[10:09:29][C][ble_rssi:011]:   Device Class: 'signal_strength'
[10:09:29][C][ble_rssi:011]:   State Class: 'measurement'
[10:09:29][C][ble_rssi:011]:   Unit of Measurement: 'dB'
[10:09:29][C][ble_rssi:011]:   Accuracy Decimals: 0
[10:09:29][C][esp32_ble_tracker:712]: BLE Tracker:
[10:09:29][C][esp32_ble_tracker:713]:   Scan Duration: 300 s
[10:09:29][C][esp32_ble_tracker:714]:   Scan Interval: 320.0 ms
[10:09:29][C][esp32_ble_tracker:715]:   Scan Window: 30.0 ms
[10:09:29][C][esp32_ble_tracker:716]:   Scan Type: ACTIVE
[10:09:29][C][captive_portal:088]: Captive Portal:
[10:09:29][C][mdns:084]: mDNS:
[10:09:29][C][mdns:085]:   Hostname: garden-bridge
[10:09:29][D][api.connection:827]: Home Assistant 2022.6.4 (::FFFF:C0A8:108): Connected successfully
[10:09:32][D][esp32_ble_tracker:726]: Found device {random device} RSSI=-75
[10:09:32][D][esp32_ble_tracker:747]:   Address Type: PUBLIC
[10:09:32][D][esp32_ble_tracker:749]:   Name: 'LEDnetWF010006D75C2C'
[10:09:32][D][esp32_ble_tracker:726]: Found device  {random device}  RSSI=-85
[10:09:32][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[10:09:34][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:09:35][D][esp32_ble_tracker:726]: Found device  {random device}  RSSI=-45
[10:09:35][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[10:09:35][D][esp32_ble_tracker:726]: Found device  {random device}  RSSI=-83
[10:09:35][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[10:09:35][D][esp32_ble_tracker:751]:   TX Power: 2
[10:09:40][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:09:40][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:09:41][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:09:41][V][sensor:074]: '$shed RSSI': Received new state -32.000000
[10:09:41][D][sensor:125]: '$shed RSSI': Sending state -32.00000 dB with 0 decimals of accuracy
[10:09:41][V][miot:169]: [redacted] [098B] Duplicate data packet received: 1
[10:09:44][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:09:45][V][sensor:074]: '$shed RSSI': Received new state -32.000000
[10:09:45][D][sensor:125]: '$shed RSSI': Sending state -32.00000 dB with 0 decimals of accuracy
[10:09:45][V][miot:169]: [redacted] [098B] Duplicate data packet received: 1
[10:09:47][D][esp32_ble_tracker:726]: Found device [redacted] RSSI=-77
[10:09:47][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[10:09:48][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:09:49][D][esp32_ble_tracker:726]: Found device [redacted]RSSI=-94
[10:09:49][D][esp32_ble_tracker:747]:   Address Type: PUBLIC
[10:09:51][V][sensor:074]: '$shed RSSI': Received new state -32.000000
[10:09:51][D][sensor:125]: '$shed RSSI': Sending state -32.00000 dB with 0 decimals of accuracy
[10:09:51][V][miot:169]: [Sensor MAC] [098B] Duplicate data packet received: 1
[10:09:54][V][sensor:074]: '$shed RSSI': Received new state -32.000000
[10:09:55][D][sensor:125]: '$shed RSSI': Sending state -32.00000 dB with 0 decimals of accuracy
[10:09:55][V][miot:169]: [Sensor MAC] [098B] Duplicate data packet received: 1
[10:09:58][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:09:58][V][sensor:074]: '$shed RSSI': Received new state -32.000000
[... 18 lines cut ...]
[10:10:20][D][esp32_ble_tracker:726]: Found device 08:EF:3B:F9:41:72 RSSI=-97
[10:10:20][D][esp32_ble_tracker:747]:   Address Type: PUBLIC
[10:10:20][D][esp32_ble_tracker:749]:   Name: 'LG SH5(72)'
[10:10:23][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:10:24][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:10:27][V][sensor:074]: '$shed RSSI': Received new state -32.000000
[10:10:27][D][sensor:125]: '$shed RSSI': Sending state -32.00000 dB with 0 decimals of accuracy
[10:10:27][V][miot:169]: [Sensor MAC] [098B] Duplicate data packet received: 1
[10:10:28][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:10:29][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:10:33][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:10:35][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:10:36][V][sensor:074]: '$shed RSSI': Received new state -33.000000
[... cut ...]
[10:10:46][V][miot:169]: [Sensor MAC] [098B] Duplicate data packet received: 1
[10:10:46][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:10:50][V][esp32_ble_tracker:697]: Unhandled type: advType: 0x08
[10:10:54][D][esp32_ble_tracker:726]: Found device CC:7E:4A:88:8E:F0 RSSI=-90
[10:10:54][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[10:10:56][V][sensor:074]: '$shed RSSI': Received new state -51.000000
[10:10:56][D][sensor:125]: '$shed RSSI': Sending state -51.00000 dB with 0 decimals of accuracy

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/978