esphome/feature-requests #1037: Protect firmware upload (web interface) but leave open action controls (ie: light on/off)

Lovely-Maisonette, 2020-12-09:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

I read that authentication will be added to the web_server component. (actually I already saw the merge)
The idea of that feature was to protect the firmware upload!!!!

Unfortunately this authentication will protect the upload, but will prevent "guest" users to turn on or off the light for instance.
The authentication should protect only the upload, or alternative, there could be an option to choose, to protect all or only the firmware upload. Using the web interface to call actions by guest users is very useful!

Analogously, if you have guest, you do want them to be able to turn on and off light, clima, etc, but you do not want them to open the switches and alter the cabling inside the walls :)

alexyao2015, 2020-12-12:
Perhaps use home assistant for actually controlling the device itself?  You would be able to avoid using the web interface altogether by simply connecting it to home assistant. IMO the web server isn't really something that should be used on the daily.

glmnet, 2020-12-23:
as a workaround you can supply custom css which would hide upload stuff