esphome/feature-requests #1042: Fetch weather data from Openweathermap API

F3ELING1337, 2020-12-17:
**Describe the problem you have/What new integration you would like**

I'm building a weather sensor right now which is ESP32 based. One job of it is to pull some weather info from the openweathermap api.

**Please describe your use case for this integration and alternatives you've tried:**

The problem is that the httprequest component doesn't seem to "store" the server responses (e.g. in a variable).
So I've created a custom text sensor **owsensor.h** which leverages httpclient.h to fetch and parse data from the api.
Please let me say that I have absolutely no knowledge about programming in C++ which is why my code probably is really bad but it gets something done.
Maybe it can be helpful as a starting point of an "official" implementation:

```cpp
#include "esphome.h"
#include "HTTPClient.h"
#include "ArduinoJson.h"

class OWSensor : public PollingComponent {
public:
	Sensor *temperature_sensor = new Sensor();
	Sensor *rain_sensor = new Sensor();
	Sensor *wind_sensor = new Sensor();
	Sensor *weatherid_sensor = new Sensor();
  
	OWSensor() : PollingComponent(60000) { }
	
	const char *TAG = "owsensor";
	String openWeatherMapApiKey = "<YOURAPICODE>";
	String lat = "<YOURLATITUDE>";
	String lon = "<YOURLONGITUDE>";
	String serverPath = "https://api.openweathermap.org/data/2.5/onecall?lat=" + lat + "&lon=" + lon + "&exclude=minutely,hourly,alerts" + "&units=metric" + "&APPID=" + openWeatherMapApiKey;

	void setup() override {

	}

	void update() override {
		
		float *data = getWeatherData();
		
		temperature_sensor->publish_state(data[0]);
		
		rain_sensor->publish_state(data[1]);
		
		wind_sensor->publish_state(data[2]);
		
		weatherid_sensor->publish_state(data[3]);

	}
	
	char *httpGETRequest(const char* serverName) {
		
		HTTPClient http;
		http.begin(serverName);

		int httpResponseCode = http.GET();

		String temp = "{}"; 

		if (httpResponseCode>0) {
			ESP_LOGD(TAG, "HTTP Response code: '%i'", httpResponseCode);
			temp = http.getString();
		}
		else {
			ESP_LOGW(TAG, "Error code: '%i'", httpResponseCode);
		}
		
		http.end();
		
		char *payload = new char[temp.length()+1];
		strcpy(payload, temp.c_str());

		return payload;

	}
	
	float *getWeatherData() {
		
		char *global_json_build_buffer = nullptr;
		global_json_buffer.clear();
		JsonObject &root = global_json_buffer.parseObject(httpGETRequest(serverPath.c_str()));
		
		float *data = new float[4];
		
		data[0] = root["current"]["temp"];
		
		data[1] = root["daily"][0]["pop"];
		data[1] = data[1]*100;
		
		data[2] = root["current"]["wind_speed"];
		
		data[3] = root["current"]["weather"][0]["id"];
		
		return data;
		
	}
  
};
```

**Additional context**

florafrisia, 2021-02-04:
Hello,

Thanks for the example.
I have a small question is it possible to use `set_update_interval` in this piece of code.

Regards,

Peter

macropin, 2021-05-25:
> Maybe it can be helpful as a starting point of an "official" implementation:

Thanks for providing this example! It saved me a lot of time.

One thing worth mentioning, is that it looks like the HTTPClient invocation has changed. If you don't pass the WiFi client, it'll compile, but you'll get an exception when reading the response. So you should do something like:

```
WiFiClient client;
HTTPClient http;
http.begin(client, serverName);
```

closed 2022-07-06 by nagyrobi