esphome/feature-requests #1045: st7789v for 240x320 screen IPS?

Electronlibre2012, 2020-12-18:
**Describe the problem you have/What new integration you would like**

the st7789v is design for TTgo screen only...135x240
**Please describe your use case for this integration and alternatives you've tried:**

i try to use the st7789v custom_component from @musk95 but even if i have put 240 and 320 instead of 135 and 240 in the "st7789v.cpp" line 265 and 269 but it boot loop.
**Additional context**

nagyrobi, 2022-07-24:
Done in https://github.com/esphome/esphome/pull/3651

thewolf-oz, 2022-08-03:
I've just received one of these screens without realising the limitation. Has this been rolled out yet or just prepared for rollout?

Electronlibre2012, 2022-08-03:
still the same note on EspHome.io : 

```
Note

Currently this only supports 135x240 pixel ST7789V displays. Other sizes (e.g. 240x320, 240x240) are not supported.
```
source : [https://esphome.io/components/display/st7789v.html](url)

so we must wait the next(s) release(s) of EspHome i think