esphome/feature-requests #1048: Switch with multiple values

nbougues, 2020-12-21:
**Describe the problem you have/What new integration you would like**
It would be nice to have a multi value switch, similar to https://www.home-assistant.io/integrations/input_select/

**Please describe your use case for this integration and alternatives you've tried:**
My use case is for this switch to be virtual only ie not tied to a specific piece of hardware, just settable through the interface (or an automation)

**Additional context**
Ciurrently the Template Switch is just on.off. This would be similar, but with a variable list of states.

I may be able to implement this, but I'm not sure where to start. While I know my stuff in Python and C++, I'm not sure where to start to implement this. Any pointers welcome.

glmnet, 2020-12-23:
I would use a global variable https://next.esphome.io/guides/automations.html#bonus-2-global-variables

deosrc, 2021-12-09:
Hopefully you've sorted this by now but I think you're looking for a template select? https://esphome.io/components/select/index.html#select-on-value

```
select:
  - platform: template
    name: "Example Select"
    optimistic: true
    restore_value: true
    options:
      - Option 1
      - Option 2
      - Option 3
```

nbougues, 2021-12-12:
Thanks for the suggestions !

nagyrobi, 2022-07-06:
Implemented with `select`.

closed 2022-07-06 by nagyrobi