esphome/feature-requests #1049: New MagicHome RGBW controller with BL602 (RISC-V) support

exentio, 2020-12-21:
**Describe the problem you have/What new integration you would like**

The classic MagicHome LED controllers just switched from ESP8266 to an SoC called BL602, apparently based on RISC-V. I accidentally got one while ordering LED controllers from AliExpress, expecting to get the classic ESP one. Probably they're all gonna switch to this SoC, which is also way better (supports Bluetooth). I'd be happy to help test the porting!

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

All the info I was able to find:
https://community.openhab.org/t/risc-v-based-bl602-magichome-rgb-led-strip-controller/109704
https://forum.creationx.de/forum/index.php?thread/3098-magichome-wlan-controller-v1-2-tasmota-flashen/&postID=35506#post35506
https://www.bouffalolab.com/bl602
https://github.com/bouffalolab/bl_iot_sdk
https://github.com/bouffalolab/bl_docs
https://github.com/pine64/bl602-docs
https://github.com/spacemeowx2/blflash
https://github.com/mkroman/bouffalo-cli

randybb, 2020-12-22:
As @OttoWinter mentioned earlier "Not an ESP, so not for ESPHome. Sorry"

glmnet, 2020-12-22:
> As @OttoWinter mentioned earlier "Not an ESP, so not for ESPHome. Sorry"

Actually partially correct, the project can very well be extended for others SoC, that would make it way more difficult to maintain though. I'm not in the mood to do much research but if it works for with PlatformIO and Arduino then development is just on our side.

OFF, that SOC seems to have built in flash, that's a big advantage over classic esp8266

randybb, 2020-12-22:
Sure, theoretically it could be extended to any Arduino lib compatible devices, but this one is not compatible.
Otto denied compatibility for [Seeduino Wio Terminal](https://circuitpython.org/board/seeeduino_wio_terminal/) powered by SAMD21 that is fully compatible with Arduino libs and in this case it would not be hard...

glmnet, 2020-12-22:
#812

It's not same but is quite similar

Citrullin, 2021-02-11:
@exentio Where did you buy it? Do you have a link?

exentio, 2021-02-11:
I wasn't the one who made the order so I don't have the link, sorry

gusman80, 2021-03-01:
Hei i started fiddling around with this board and made a simple firmware to test around the PWM outputs.
i started out of the info provided right here: https://lupyuen.github.io/articles/book
the author's super kind, answers quickly on reddit and provides very good information.
I don't think its possible to support this completely different architecture within this project.

Still - for those who are stuck with magichome's $#%firmware - Here is the Layout of the new MagicHome >>

RGBW controller MagicHome ZJ-BWCE-IR-RGBW V1.2
PIN	gpio  	Asset       PWM-Channel
4	gpio3	Red		pwm3
5	GPIO4	Green	pwm4
9	ANt		RF	
16	 gpio7	RX
17	gpio8	Boot Pin
21	gpio12	IR Pin
27	gpio16	TX
29	gpio20	WHITE	pwm0
30	gpio21	BLUE	pwm1

have great fun

glmnet, 2021-03-01:
Well I believe we should support the esp32 c3 which is risc v 

unfortunately I don’t own one and can’t get them in my country. 

I accept shipments via DHL or FedEX 🙃

glmnet, 2021-03-01:
I believe that will mean supporting risc v in general. Not sure yet I’m slow to grasp new stuff

Citrullin, 2021-03-02:
> Well I believe we should support the esp32 c3 which is risc v
> 
> unfortunately I don’t own one and can’t get them in my country.
> 
> I accept shipments via DHL or FedEX upside_down_face

What country are we talking about? I contacted a couple of seller on Aliexpress. This seller told me that their [product uses the BL602](https://de.aliexpress.com/item/4000166649124.html)

glmnet, 2021-03-02:
Yeah it is

gusman80, 2021-03-02:
[image]
i ordered many different magichome layouts ... you can recognize the devices with a 100% bl602 rate by these weird octagon cornered casings

Citrullin, 2021-08-03:
> 
> 
> [image]
> i ordered many different magichome layouts ... you can recognize the devices with a 50% bl602 rate by these weird octagon cornered casings
> i ordered mine on ebay 283947655801

I can confirm this. My success rate 100% with it.

lanrat, 2021-08-03:
I ordered 4 with the diagonal corners from 3 different sellers, all contained the BL602.

RoganDawes, 2021-08-05:
I see https://github.com/pine64/ArduinoCore-bouffalo, but apparently no support yet in platformio, which is a bit of an obstacle. I note the aim to make it as compatible as possible with the ESP*, which should simplify things a bit, perhaps.
I don't really imagine that use of RiscV vs Xtensa cores should make much of a difference, that's simply down to the compiler, surely?

normanr, 2021-09-06:
fyi: I filed a bug to add support in PlatformIO: https://github.com/platformio/platformio-core/issues/4046

maxgerhardt, 2021-09-13:
If anyone can help test my PlatformIO integration (https://github.com/platformio/platformio-core/issues/4046#issuecomment-918220660), please let me know your results. You should be able to compile + upload to a BL602 based board with the Pine64 Bouffalo Arduino core.

bdraco, 2021-11-13:
> Well I believe we should support the esp32 c3 which is risc v
> 
> unfortunately I don’t own one and can’t get them in my country.
> 
> I accept shipments via DHL or FedEX 🙃

If you are still looking for some I have a few extra from testing other projects left over. Happy to ship. 

Email nick at koston dot org

geoffrothman, 2023-10-16:
For anyone stumbling on this via google search later on, here's a link to opensource firmware that can add it to esphome: https://community.home-assistant.io/t/open-source-firmware-for-new-sonoff-products-or-any-other-bl602-based-iot-device-with-youtube-guide/536935

RoganDawes, 2025-07-06:
Openshwprojects seems to be maintaining an sdk for this chip here: https://github.com/openshwprojects/OpenBL602

And examples of how to actually use the api should exist in the OpenBeken source.