esphome/feature-requests #1050: IR Climate support ZH/JT01 ZH/JT03

marcinkowalczyk, 2020-12-22:
**Describe the problem you have/What new integration you would like**

Both way support for ZH/JT01 /JT03 remotes - they are used by many brands -> protocol codes https://github.com/arthurkrupa/ac-ir-mqtt-zhjt03/blob/master/include/codes.h 

**Please describe your use case for this integration and alternatives you've tried:**

Works with https://github.com/arthurkrupa/ac-ir-mqtt-zhjt03

**Additional context**

Extension of supported remotes

cfeenstra1024, 2022-07-31:
Hi @marcinkowalczyk, would you still be interested in this? If so, would you have the ability to test this? 
I just implemented the ZH/LT-01 integration (another Chigo remote) and this one looks very similar. 
I do not have this AC myself, so I cannot test. But if you are willing, I can try to create the integration for you.

marcinkowalczyk, 2022-07-31:
Hi,

Sure. Still running this AC

cfeenstra1024, 2022-07-31:
Actually, after looking a bit deeper into the IR codes from the links in your original post, I think this is actually the same as the ZH/LT-01 remote. Could you give that a try?

Add the below to your config yaml to test:

``` yaml
external_components:
  - source:
      type: git
      url: https://github.com/cfeenstra1024/esphome
      ref: zhlt01-climate-ir
    components: [ zhlt01 ]
```

``` yaml
climate:
  - platform: zhlt01
    name: "AC"
```

All standard configuration parameters for Climate_IR apply (refer to https://esphome.io/components/climate/climate_ir.html)

marcinkowalczyk, 2022-07-31:
Hi,
Seems to work on both ZH/JT01 and ZH/LT01. 
Do you plan to add IR receiver support too?

cfeenstra1024, 2022-07-31:
Thats great! IR Receiver is also in already. Just setup the `remote_receiver` section in the config file (https://esphome.io/components/climate/climate_ir.html#using-a-receiver).

I'll connect this issue to the PR of the ZH/LT-01 integration

marcinkowalczyk, 2022-07-31:
Hi,
 I'm currenty testing with another unit with ZH/LT-01 (no access to JT01 at the moment)
 
```
 remote_receiver:
  id: rcvr
  pin: 
    number: GPIO5
    inverted: True
    mode: INPUT_PULLUP
  dump: all

# Configure the IR LED: this one sends commands to other devices
remote_transmitter:
  pin: GPIO14
  carrier_duty_percent: 50%

climate:
  - platform: zhlt01
    name: "zhlt01"
    receiver_id: rcvr
```

debug log on/off 

```
[18:18:50][C][remote_transmitter:018]: Remote Transmitter...
[18:18:50][C][remote_transmitter:019]:   Carrier Duty: 50%
[18:18:50][C][remote_transmitter:020]:   Pin: GPIO14
[18:18:50][C][climate_ir:076]: IR Climate 'zhlt01'
[18:18:50][C][climate_ir:077]:   Min. Temperature: 16.0°C
[18:18:50][C][climate_ir:078]:   Max. Temperature: 32.0°C
[18:18:50][C][climate_ir:079]:   Supports HEAT: YES
[18:18:50][C][climate_ir:080]:   Supports COOL: YES
[18:18:50][C][remote_receiver.esp8266:060]: Remote Receiver:
[18:18:50][C][remote_receiver.esp8266:061]:   Pin: GPIO5
[18:18:50][C][remote_receiver.esp8266:066]:   Buffer Size: 1000
[18:18:50][C][remote_receiver.esp8266:067]:   Tolerance: 25%
[18:18:50][C][remote_receiver.esp8266:068]:   Filter out pulses shorter than: 50 us
[18:18:50][C][remote_receiver.esp8266:069]:   Signal is done after 10000 us of no changes
[18:18:50][C][captive_portal:088]: Captive Portal:
[18:18:50][C][web_server:125]: Web Server:
[18:18:50][C][web_server:126]:   Address: 192.168.1.32:80
[18:18:50][C][mdns:084]: mDNS:
[18:18:50][C][mdns:085]:   Hostname: salon_ir
[18:18:50][C][ota:085]: Over-The-Air Updates:
[18:18:50][C][ota:086]:   Address: 192.168.1.32:8266
[18:18:50][C][api:138]: API Server:
[18:18:50][C][api:139]:   Address: 192.168.1.32:6053
[18:18:50][C][api:143]:   Using noise encryption: NO
[18:18:50][C][sntp:050]: SNTP Time:
[18:18:50][C][sntp:051]:   Server 1: '0.pool.ntp.org'
[18:18:50][C][sntp:052]:   Server 2: '1.pool.ntp.org'
[18:18:50][C][sntp:053]:   Server 3: '2.pool.ntp.org'
[18:18:50][C][sntp:054]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[18:18:57][D][remote.pronto:229]: Received Pronto: data=0000 006D 0063 0000 00EB 011D 0017 003D 0018 003D 0017 003D 0017 003D 0018 003D 0017 003D 0018 003D 0018 003D 0017 0016 0017 0015 0018 0014 0017 0015 0018 0014 0018 0014 0018 0014 0017 0015 0017 003D 0017 003D 0018 003D 0017 003D 0017 003E 0017 003D 0017 003D 0018 003D 0018 0014 0018 0014 0018 0014 0018 0014 0018 0014 0018 0014 0018 0014 0018 0015 0018 003C 0017 003D 0018 003D 0017 003E 0017 003D 0017 003E 0018 003C 0018 003C 001
[18:18:59][D][remote.pronto:229]: Received Pronto: data=0000 006D 0001 0000 0003 06C3
[18:19:01][D][remote.pronto:229]: Received Pronto: data=0000 006D 0063 0000 00EB 011D 0018 003C 0018 003C 0017 003E 0017 003E 0017 003E 0017 003E 0018 003D 0017 003D 0017 0015 0018 0014 0018 0014 0018 0014 0018 0014 0018 0015 0018 0014 0018 0014 0017 003D 0018 003C 0017 003D 0018 003C 0017 003E 0018 003C 0018 003D 0018 003D 0018 0014 0017 0015 0017 0015 0018 0014 0017 0015 0018 0014 0017 0015 0017 0015 0017 003D 0019 003B 0017 003E 0018 003D 0017 003E 0018 003C 0017 003D 0018 003C 001
[18:19:04][D][remote.pronto:229]: Received Pronto: data=0000 006D 0063 0000 00EC 011C 0017 003E 0017 003E 0018 003C 0017 003E 0017 003E 0017 003E 0018 003D 0017 003E 0017 0015 0017 0015 0017 0015 0018 0014 0018 0014 0018 0014 0018 0014 0018 0015 0018 003C 0018 003C 0018 003C 0018 003D 0017 003E 0017 003D 0018 003D 0018 003C 0018 0015 0018 0014 0018 0014 0018 0014 0017 0015 0018 0014 0018 0014 0018 0014 0018 003C 0018 003D 0018 003C 0018 003C 0018 003C 0017 003D 0017 003D 0018 003D 001
```

cfeenstra1024, 2022-07-31:
Can you set the `dump` setting to `raw` and send the log again?

marcinkowalczyk, 2022-07-31:
cooling 25, swing + airflow

```
[18:25:44][D][remote.raw:028]: Received Raw: 6136, -7391, 638, -1553, 610, -1579, 614, -1575, 614, -1579, 633, -1558, 612, -1578, 633, -1559, 614, -1576, 610, -536, 612, -538, 634, -511, 612, -536, 635, -513, 611, -536, 615, -532, 614, -534, 635, -1555, 636, -1555, 613, -1577, 636, 
[18:25:44][D][remote.raw:028]:   -1555, 636, -1556, 636, -1553, 610, -1581, 635, -1556, 610, -537, 636, -510, 613, -536, 610, -541, 633, -510, 611, -536, 638, -509, 636, -513, 608, -1580, 638, -1551, 613, -1578, 613, -1578, 640, -1550, 635, -1556, 611, -1578, 639, -1552, 635, -512, 
[18:25:44][D][remote.raw:028]:   634, -513, 613, -532, 635, -513, 609, -538, 634, -514, 633, -514, 611, -536, 609, -1581, 637, -511, 634, -1556, 635, -513, 611, -1578, 637, -510, 635, -514, 611, -1583, 609, -533, 611, -1580, 635, -514, 631, -1556, 612, -536, 634, -1554, 638, -1553, 
[18:25:44][D][remote.raw:028]:   637, -510, 612, -538, 607, -1586, 606, -1582, 634, -511, 614, -1577, 613, -537, 635, -1554, 611, -1581, 635, -1557, 635, -512, 637, -511, 635, -1556, 611, -537, 611, -1585, 607, -537, 611, -536, 610, -538, 632, -1559, 641, -508, 611, -1580, 635, -512, 
[18:25:44][D][remote.raw:041]:   610, -1582, 636, -511, 612, -538, 610, -1582, 609, -538, 609, -1582, 614, -536, 607, -1584, 663, -483, 609, -1584, 610, -1582, 613, -7370, 561
```

off:
```
[18:26:41][D][remote.raw:028]: Received Raw: 6165, -7373, 612, -1578, 638, -1552, 640, -1550, 639, -1556, 640, -1550, 638, -1555, 635, -1557, 612, -1579, 613, -536, 636, -510, 612, -537, 638, -511, 611, -537, 611, -536, 637, -512, 665, -482, 638, -1554, 613, -1579, 613, -1578, 610, 
[18:26:41][D][remote.raw:028]:   -1581, 614, -1579, 635, -1555, 638, -1554, 638, -1552, 616, -537, 608, -536, 637, -510, 637, -511, 615, -534, 615, -533, 613, -534, 638, -510, 638, -1557, 635, -1552, 612, -1583, 634, -1555, 636, -1554, 613, -1578, 613, -1579, 615, -1575, 638, -510, 
[18:26:41][D][remote.raw:028]:   613, -535, 611, -537, 637, -511, 642, -506, 613, -535, 663, -484, 636, -511, 611, -1581, 613, -1578, 635, -1557, 613, -539, 633, -1556, 612, -535, 639, -509, 612, -1580, 612, -534, 638, -511, 636, -512, 637, -1555, 634, -512, 638, -1560, 634, -1553, 
[18:26:42][D][remote.raw:028]:   612, -535, 640, -509, 638, -1553, 612, -1583, 636, -509, 638, -1553, 611, -538, 609, -1582, 638, -1552, 638, -1554, 613, -534, 612, -541, 607, -1582, 613, -533, 615, -1577, 614, -533, 613, -535, 611, -538, 635, -1556, 610, -537, 610, -1581, 615, -533, 
[18:26:42][D][remote.raw:041]:   635, -1557, 613, -534, 612, -536, 610, -1582, 618, -529, 634, -1556, 638, -511, 614, -1579, 636, -510, 610, -1582, 634, -1555, 636, -7345, 588
```

cfeenstra1024, 2022-07-31:
The timing is a bit off compared to the 2 remotes I have here. You can try to set the tolerance to a higher level (default is 25%):

``` yaml
 remote_receiver:
  id: rcvr
  pin: 
    number: GPIO5
    inverted: True
    mode: INPUT_PULLUP
  tolerance: 55%

# Configure the IR LED: this one sends commands to other devices
remote_transmitter:
  pin: GPIO14
  carrier_duty_percent: 50%

climate:
  - platform: zhlt01
    name: "zhlt01"
    receiver_id: rcvr
```

When you change the logging level to verbose, it will log the reason for rejecting the received IR command.
I think thats done by:

``` yaml
logger:
  level: verbose
```

When I manually decode the first log, The byte containing the temperature and mode seems wrong. I get heating at 36 degree (which I think is not possible). But I could have made a decoding error ;)

marcinkowalczyk, 2022-07-31:
Hi,

Works fine with 55% tolerance. I will test later-on with other remotes I have

cfeenstra1024, 2022-07-31:
OK, nice. Please let me know the results of the other remote when you have tested that.

I rechecked the decoding I did and spotted the mistake. It is indeed cooling at 25 degree according to the raw data of your log.

ummetcivi, 2023-08-19:
This works fine with JT-01. Do you plan to merge this to master?

cfeenstra1024, 2023-08-21:
A pull request was created January 2023. I certainly hope this gets merged soon. However, there is not much I can do about this. I think the actual merges are done by Jesse Hills.

@jesserockz : Would it be possible to pick this up?

closed 2023-10-23 by jesserockz