esphome/feature-requests #1053: 'Tis the Season: ESPHome as a Gift

dlashua, 2020-12-25:
**Describe the problem you have/What new integration you would like**

I'd like ESPHome to be more suitable for Gift Giving.

**Please describe your use case for this integration and alternatives you've tried:**

ESPHome has to be compiled with various settings specific to a User's Network and Configuration. While some of these things are device specific, many of them are related to the environment the device will be in. When a device is developed to be given as a gift, it is not in it's final environment. Therefore, there are some settings that need to be configurable after the firmware has been installed. Specifically, these settings are Wifi SSID, Wifi Password, Web Password, OTA Password, API Password, MQTT settings, etc. I know that some of these settings are available in the captive_portal.

Ultimately, this feature request may really be only a request for documentation specific to setting up an ESPHome device to be used in an environment different from the one it was developed in without prior knowledge of that environment. Ideally, the documentation would include a sample ESPHome YAML for a simple device (perhaps a temperature sensor, an led, and a button) and then the additional API calls that need to be made to take the device from unconfigured in a new environment, to connected to the new network and the ability to retrieve the temperature value, the button state, and manipulate the LED.

**Additional context**

glmnet, 2020-12-28:
So you want docs to turn somebody you gift an esp8266 into a DIY Home Automation enthusiast?

You can gift an esphome which plays tones when motion is detected (runs on its own) but gift an esphome and expect him to setup, e.g. Home Assistant etc. it's a bit too much... even if he already has Home Assistant, all docs are in the getting started section

If you give an esphome with `captive_portal` then that is enough, he just joins the device on the network and can ota to it directly.

You can write a guide and we can put it on the website, no charges 😀

dlashua, 2020-12-28:
Not quite.

I'm looking for information on how to set Wifi SSID, Wifi Password, API Password, MQTT settings, etc, via API when the ESPHome device is in captive_portal mode. I don't believe setting all of these via any kind of API call is possible, though, I think at least Wifi SSID and Wifi Password are.

And the documentation for how to bring an ESPHome device developed on one Network to a different Network is lacking, if this is even possible without a recompile and reupload. I'd be happy to write the guide if I knew how to do these tasks. But I don't. That's why I'm looking for such documentation.

Thanks for the help.

glmnet, 2020-12-28:
https://esphome.io/components/captive_portal.html

you can join a lost node to a nearby network that way, then you can access the node's own web server if it is all you want, or you can OTA to it if you have ESPHome installed, e.g. in Home Assistant.

You really don't need api_password and all the other stuff for gift, if the end user really wants he can build it's own firmware. Thats what esphome is all about.

dlashua, 2020-12-28:
That doesn't really get there, but I appreciate the effort. Tasmota is more suited to gift giving at this point (which is what I've been using for devices I build for others). But I'd prefer to use esphome if it ever becomes more feasible.

multipolygon, 2021-01-05:
Making Captive Portal WiFi settings persistent between boots would be amazing for this purpose.

glmnet, 2021-01-05:
> Just making Captive Portal WiFi settings persistent between boots would be 99% "good enough" for this use case.

They are

dlashua, 2021-01-05:
@multipolygon is that a feature that exists already or are you saying ESPHome needs this feature to work for this use case? If it exists already, how do I enable it?

I would say this feature would make it 75% "good enough". It would still be desirable for the final user to be able to set the API password without having to OTA upload a whole new firmware. This way, they could be sure they are the only one who would have access to the device. Presumably, since I gave it to them as a gift, they trust me, so they are fine with me knowing the password too. But it would be better if I didn't have to.