esphome/feature-requests #1055: Integration of MAX17043 SparkFun LiPo Fuel Gauge

IsaTTeN, 2020-12-27:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

IsaTTeN, 2021-10-02:
Sorry @andyjk15 , I didn´t see your msg. I am in the same point, did you get this working??

a-marcel, 2021-12-23:
Hello, 

i'm using the [Gravity 3.7V Li Battery Fuel Gauge](https://wiki.dfrobot.com/Gravity__3.7V_Li_Battery_Fuel_Gauge_SKU__DFR0563) sensor that populates voltage and percentage data via i2c. 

I was able to collect data from that sensor (at the moment i'm only letting it run for a few hours - but the results look promising).

`MAX17043_component.h`
```
#include "esphome.h"

// inspired from:
// - https://github.com/DFRobot/DFRobot_MAX17043/blob/master/DFRobot_MAX17043.h
// - https://github.com/exxamalte/esphome-customisations/tree/master/mlx90614

#define MAX17043_ADDRESS        0x36

#define MAX17043_VCELL          0x02 // voltage
#define MAX17043_SOC            0x04 // percentage
#define MAX17043_MODE           0x06
#define MAX17043_VERSION        0x08
#define MAX17043_CONFIG         0x0c
#define MAX17043_COMMAND        0xfe

class MAX17043Sensor : public PollingComponent, public Sensor {
 public:
  Sensor *voltage_sensor = new Sensor();
  Sensor *percentage_sensor = new Sensor();

  MAX17043Sensor() : PollingComponent(10000) {}

  void setup() override {
    // Initialize the device here. Usually Wire.begin() will be called in here,
    // though that call is unnecessary if you have an 'i2c:' entry in your config
    ESP_LOGD("custom", "Starting up MAX17043 sensor");

    Wire.begin();
  }

  uint16_t read16(uint8_t reg) {
      uint16_t        temp;
      Wire.begin();
      Wire.beginTransmission(MAX17043_ADDRESS);
      Wire.write(reg);
      Wire.endTransmission();
      Wire.requestFrom(MAX17043_ADDRESS, 2);
      temp = (uint16_t)Wire.read() << 8;
      temp |= (uint16_t)Wire.read();
      Wire.endTransmission();
      return temp;
  }

  void update() override {
    float voltage = 1.25f * (float)(read16(MAX17043_VCELL) >> 4);
    voltage_sensor->publish_state(voltage);

    uint16_t percentage_tmp = read16(MAX17043_SOC);
    float percentage = (float)((percentage_tmp >> 8) + 0.003906f * (percentage_tmp & 0x00ff));

    percentage_sensor->publish_state(percentage);
  }
};
```

`ESPhome configuration`

```
esphome:
  includes:
    - MAX17043_component.h

i2c:
  sda: 21
  scl: 22
  scan: true
  id: bus_voltage

sensor:
  - platform: custom
    lambda: |-
      auto max17043_sensor = new MAX17043Sensor();
      App.register_component(max17043_sensor);
      return {max17043_sensor->voltage_sensor, max17043_sensor->percentage_sensor};
    sensors:
      - name: "Voltage"
      - name: "Percentage"
```

`Output`
```
[01:06:54][D][sensor:113]: 'Voltage': Sending state 4036.25000  with 0 decimals of accuracy
[01:06:55][D][sensor:113]: 'Percentage': Sending state 84.00781  with 0 decimals of accuracy

[01:07:54][D][sensor:113]: 'Voltage': Sending state 4036.25000  with 0 decimals of accuracy
[01:07:54][D][sensor:113]: 'Percentage': Sending state 83.84760  with 0 decimals of accuracy

[01:08:44][D][sensor:113]: 'Voltage': Sending state 4035.00000  with 0 decimals of accuracy
[01:08:44][D][sensor:113]: 'Percentage': Sending state 83.84760  with 0 decimals of accuracy
```

I'm not sure if this is helpful, but maybe this piece of code enables you to write code for your sensors. In the `MAX17043_component.h` snipped i wrote to pages that are helpful for me.

Marcel

IsaTTeN, 2022-01-23:
Hi @a-marcel, thanks for your contribution.

I used your code and it's working!!! Dinally I have the MAX17043 working en ESPHome.

Do you know how to suggest to include this device in ESPHome with your code??

closed 2022-07-01 by nagyrobi