esphome/feature-requests #1057: LED Light Component should support more steps / finer granularity

demlak, 2020-12-27:
**Describe the problem you have/What new integration you would like**

Would be nice to have finer granularity on Light Component. 256 Steps (8bit) or 100 Steps (percentage) is very low resolution. For example usage of PCA9685 allows 4096 Steps, but Light.Control seems to not reflect this.

**Please describe your use case for this integration and alternatives you've tried:**

Im using a PCA9685 12Bit PWM Driver to control an analog RGB LED Stripe and i need fine low-light stepping for a wakeup light.

I was playing with none.integer values and saw, that they seem to be rounded to integer

**Additional context**

Since i'm not a coder, i can't say how difficult this would be. But maybe it would help to have a setting that declares the stepping and/or switch percentage values from integer to finer stepping.

my current yaml for reference:
```
esphome:
  name: rgb_led_bed
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "xxx"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Rgb Led Bed Fallback Hotspot"
    password: "xxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxx"

ota:
  password: "xxx"
  
# Enable Web-Server
web_server:
  port: 80
  
time:
  - platform: homeassistant
    id: homeassistant_time
    timezone: Europe/Berlin

# Example configuration entry
i2c:
  sda: D2
  scl: D1
  scan: True
  id: bus_a

# Example configuration entry
pca9685:
  frequency: 500

#Outputs
output:
  - platform: pca9685
    id: 'pca9685_output0'
    channel: 0
    max_power: 0.80
  - platform: pca9685
    id: 'pca9685_output1'
    channel: 1
    max_power: 0.72
  - platform: pca9685
    id: 'pca9685_output2'
    channel: 2

# Example configuration entry
light:
  - platform: rgb
    name: "strip1"
    id: strip1
    restore_mode: ALWAYS_OFF
    default_transition_length: 2s
    red: pca9685_output1
    green: pca9685_output2
    blue: pca9685_output0
```

buddhra, 2021-01-16:
Seconded. I tried to set up a slow, 5-minute ramp in brightness at sunset and the big steps were pretty ugly.

AlexBMJ, 2021-08-05:
Another addition to this would be to use the phase balancer from [NachtRaveVL/PCA9685-Arduino](https://github.com/NachtRaveVL/PCA9685-Arduino) to reduce PSU noise when outputting across all channels.

`PCA9685_PhaseBalancer_Linear`
// Uses linear software-based phase balancing, with each channel being a preset 16 steps (out of the 4096/12-bit value range) away from previous channel (may cause LED flickering/skipped-cycles on PWM changes)

Voodoolmar, 2023-11-09:
Any solutions?