esphome/feature-requests #1060: Debounce on Zero Cross for AC Dimmer

rjmurph2241, 2020-12-30:
**Describe the problem you have/What new integration you would like**

Would it be possible to add de-bounce to the zero cross pin on the AC dimmer component?
**Please describe your use case for this integration and alternatives you've tried:**

I have tried to use an Robodyn AC dimmer with a NodeMCU V2 but no matter what I do I always end up with an occasional flicker.  I eventually gave up and bought a Shelly Dimmer 2 which appears to be working (so it's not my LED bulbs).  I noticed that there is a de-bounce option for zero cross on the Shelly and thought that may be one reason why the Shelly doesn't have the flicker.
**Additional context**

xanderdekok, 2021-01-02:
I had some flicker as well. It might be interference. Take a look at this bug report:
[https://github.com/esphome/issues/issues/1632](url)

rjmurph2241, 2021-01-03:
Interesting, thanks for pointing me there.  Looks like I have some experimenting to do.

micw, 2023-03-08:
The link above is wrong (link text is correct, link target is wrong). Adding the rigth one for reference:
https://github.com/esphome/issues/issues/1632