esphome/feature-requests #1061: Add Support for Shelly 1L

Veldkornet, 2020-12-30:
**Describe the problem you have/What new integration you would like**

Add support for the Shelly 1L

**Please describe your use case for this integration and alternatives you've tried:**

Shelly 1L not currently supported

**Additional context**

[Shelly 1L](https://shelly.cloud/knowledge-base/devices/shelly-1l/#tab_5fec88452cbc2)

TomHartogs, 2021-01-03:
What makes you think Shelly 1L is not supported? It is just another esp8266 powered board. I couldn't find any premade YAML files with a configuration for the Shelly 1L but I was able to get it to work by creating my own configuration just fine.

`substitutions:
  plug_name: 'plug_name'
  wifi_ssid: 'wifi_ssid'
  wifi_pw: 'wifi_password'

esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp01_1m
  esp8266_restore_from_flash: True

# WiFi connection
wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_pw}
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${plug_name}
    password: ${wifi_pw}

# Enable over-the-air updates
ota:
#  password: ${ota_pw}

# Enable Home Assistant API
api:
#  password: ${api_pw}

# Enable logging
logger:

# Enable Web server
web_server:
  port: 80

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

#==============================================================
output:
  - platform: gpio
    pin: GPIO5
    id: shelly_1l_relay

light:
  - platform: binary
    name: 'Light'
    output: shelly_1l_relay
    id: lightid

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO4
    name: 'Switch 1'
    on_state:
      then:
        - light.toggle: lightid
    id: switchid1
    filters:
      - delayed_on_off: 50ms
  - platform: gpio
    pin:
      number: GPIO14
    name: 'Switch 2'

sensor:
  - platform: adc
    pin: GPIO17
    name: 'Temperature'
    update_interval: 60s

status_led:
  pin: GPIO0
#==============================================================
`

I used the following config successfully with Home Assistant. It contains a built-in automation to switch the relay (light) when switch 1 is flipped. Switch 2 is exposed as a binary_sensor. 

The 'temperature sensor' now shows the plain ADC value. In order to get the correct temperature value, it still needs to be converted. I have not yet been able to figure out the formula that Tasmota uses for this conversion (since it worked fine there as an ADC temp).

rradar, 2021-01-08:
https://www.esphome-devices.com/devices/shelly-1l/

closed 2021-01-08 by rradar